schreibfaul1/ESP32-MiniWebRadio #179: Exception caused by settings.json file

steve6375, 2023-06-06:
Somehow the settings.json file on the SD card got set to all 0's for each parameter (except number of stations)
The result was that my project suddenly started crashing and resetting during startup.
Here is the terminal output:

```
[  6676][I][WiFiMulti.cpp:119] run(): [WIFI] 1 networks found
[  6677][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: XX:XX:XX:XX:XX:XX SSID: XXXXXXXXXX Channel: 4 (-26)
[  7928][I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
00:00:07 WiFI_info:   Connecting WiFi...
00:00:07 setup: ....  connected to xxxxxx, IP address is 192.168.1.113
00:00:07 ftpServer:   Buffers allocated: 4622 bytes
00:00:07 rtime_info:  Initializing SNTP
00:00:07 rtime_info:  Waiting for system time to be set... (1/10)
09:27:58 rtime_info:  09:27:58
09:27:58 audiotask:   is pinned to core 1
09:27:58 audiotask:   priority is 2
09:27:58 setup: ....  Number of saved stations: 17
09:27:58 setup: ....  current station number: 0
09:27:58 setup: ....  current volume: 0
09:27:58 Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008c82d  PS      : 0x00060630  A0      : 0x80184198  A1      : 0x3ffc6900  
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffc6c40  
A10     : 0x00000003  A11     : 0x00060623  A12     : 0x00060620  A13     : 0x00000000  
A14     : 0x007c16c0  A15     : 0x003fffff  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4008c82d  LEND    : 0x4008c83d  LCOUNT  : 0xffffffff

Backtrace: 0x4008c82a:0x3ffc6900 0x40184195:0x3ffc6910 0x4018ba55:0x3ffc6c20 0x4018ba9b:0x3ffc6cb0 0x400f5c52:0x3ffc6cf0 0x400da005:0x3ffc6d90 0x400f5b96:0x3ffc6e20

  #0  0x4008c82a:0x3ffc6900 in strlen at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:82       
  #1  0x40184195:0x3ffc6910 in _svfprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1528      
  #2  0x4018ba55:0x3ffc6c20 in _vsnprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vsnprintf.c:70 (discriminator 4)
  #3  0x4018ba9b:0x3ffc6cb0 in vsnprintf at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vsnprintf.c:40
  #4  0x400f5c52:0x3ffc6cf0 in Print::printf(char const*, ...) at C:/Users/Steve/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/cores/esp32/Print.cpp:55
  #5  0x400da005:0x3ffc6d90 in setup() at src/main.cpp:1263
  #6  0x400f5b96:0x3ffc6e20 in loopTask(void*) at C:/Users/Steve/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/cores/esp32/main.cpp:42

ELF file SHA256: 2b4b6a1663ed86e6

E (10066) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
E (984) psram: PSRAM ID read error: 0xffffffff
E (984) spiram: SPI RAM enabled but initialization failed. Bailing out.
E (998) esp_core_dump_flash: No core dump partition found!
E (998) esp_core_dump_flash: No core dump partition found!
ESP32 Chip: ESP32-D0WD-V3
Arduino Version: 2.0.8
ESP-IDF Version: 4.4.4
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)

```

The clue was in the lines
**09:27:58 setup: ....  current station number: 0
09:27:58 setup: ....  current volume: 0**
which led me to look at the settings.json file and I saw it was all 0's

I am not sure which of the parameters caused the exception but it looks like the lastconnectedhost line 1263 caused the issue???

main.cpp
    SerialPrintfln("setup: ....  Number of saved stations: " ANSI_ESC_CYAN "%d", _sum_stations);
    SerialPrintfln("setup: ....  current station number: " ANSI_ESC_CYAN "%d", _cur_station);
    SerialPrintfln("setup: ....  current volume: " ANSI_ESC_CYAN "%d", _cur_volume);
   SerialPrintfln("setup: ....  last connected host: " ANSI_ESC_CYAN "%s", _lastconnectedhost.c_str());

    // _alarmdays = pref.getUShort("alarm_weekday");
    // _alarmtime = pref.getUInt("alarm_time");
    // _f_timeAnnouncement = pref.getBool("timeAnnouncing");
    _state = RADIO;

My C coding is not good enough to fix the problem however!

schreibfaul1, 2023-06-06:
Thanks for the error description, I can't say why settings.json doesn't contain any non-zero values ​​yet, but `printf('%s', NULL);` causes a crash. I won't be able to take a closer look until next week.

steve6375, 2023-06-24:
[settings_BAD.zip](https://github.com/schreibfaul1/ESP32-MiniWebRadio/files/11856380/settings_BAD.zip)
Here is file which causes crash (havent tried latest version yet though).

steve6375, 2023-06-24:
```
    SerialPrintfln("setup: ....  Number of saved stations: " ANSI_ESC_CYAN "%d", _sum_stations);
    SerialPrintfln("setup: ....  current station number: " ANSI_ESC_CYAN "%d", _cur_station);
    SerialPrintfln("setup: ....  current volume: " ANSI_ESC_CYAN "%d", _cur_volume);
    if (_lastconnectedhost == "") _lastconnectedhost = "NONE";   // ZZZZZ added to prevent crash
```

schreibfaul1, 2023-06-24:
Exactly, well analyzed, _lastconnectedhost must never be NULL, but an empty string is allowed.
I couldn't find out why _lastconnectedhost became NULL, but I installed guards to avoid a crash.

closed 2023-07-04 by steve6375