schreibfaul1/ESP32-MiniWebRadio #180: Run Menuconfig

valera39, 2023-06-06:
Dear Wolle!
I'm still trying to figure out the PSRAM in ESP32S3 from Espressif.
How do I navigate through the lines in this menu? I've tried many keys but no luck.
[image]I want to find the parameter which is responsible for enabling PSRAM as written in this document
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/flash_psram_config.html
Or am I wrong?
Thanks!

steve6375, 2023-06-08:
platform.ini already contains and env for 32s3 and  -DBOARD_HAS_PSRAM   is already in that env.
So all you should have to do is select it in platformio bottom tiny systray
[image]

I have not tried a compile though...

```
[env:esp32s3]
        ;board = esp32s3-usb ;chipmodel ESP32S3, SerialUSB (e.g. Pros3)
        board = esp32-12k ;chipmodel ESP32S3, USBtoTTL  (e.g. S3-DevKit)
build_flags =
        -DCORE_DEBUG_LEVEL=3            ; 0 None, 1 Error, 2 Warn, 3 Info, 4 Debug, 5 Verbose
        -DCONFIG_ARDUHAL_LOG_COLORS
        -DBOARD_HAS_PSRAM               ; Remove this entry if the board does not have PSRAM
        -DCONFIG_IDF_TARGET_ESP32S3
build_unflags =
        -DCONFIG_IDF_TARGET_ESP32
;       -DARDUINO_USB_CDC_ON_BOOT=0     ; traditional log (board with native USB)
        -Wall
        -Wextra

```

valera39, 2023-06-08:
I used https://github.com/schreibfaul1/ESP32_Arduino_ESPIDF repository of respected Wolle to change the parameters of sdkconfig file, where according to the documentation
from Espessif https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/flash_psram_config.html, the PSRAM initialization is prescribed.

closed 2023-06-10 by valera39