schreibfaul1/ESP32-MiniWebRadio #181: Improvement for MinWebWeb browser GUI

steve6375, 2023-06-06:
The columns labelled X and Cy are not very clear as to what they do or mean.
The column titles can be  easily changed by adding a title entry for each one. e,g,

    fields: [{
      name: 'X',
      **title: 'Hide',**
      type: 'text',
      width: 20,
      align: 'center'
    },

This means we can have the table in a different language if required.
Perhaps the text for these four columns could be put as defined strings in common.h

[image]

steve6375, 2023-06-06:
Also, to just list .xlsx files instead of all files, I used  input  line as below:
```
      <button class="button buttonblue"
              onclick="javascript:document.getElementById('file').click();"
              title="Load from PC">load xlsx
      </button>
      <input id="file" type="file" accept="application/vnd.openxmlformats-officedocument.spreadsheetml.sheet" style="visibility: hidden; width: 0px;"  name="img"
          onchange="loadDataExcel(this.files);"/>
      <br>
      </center>
  </div>
```

approx line 1705 in index.h

steve6375, 2023-06-08:
Increase height of screen 1 a bit - use 550px instead of 540px
[image]

```
#content {
            min-height : 550px;
            min-width : 725px;
            overflow : hidden;
            background-color : lightskyblue;
            margin : 0;
            padding : 5px;
        } 
```

schreibfaul1, 2023-06-09:
> The columns labelled X and Cy are not very clear as to what they do or mean. The column titles can be easily changed by adding a title entry for each one in index.h file . e,g, lines 1056
> 
> ```
> fields: [{
>   name: 'X',
>   **title: 'Hide',**
>   type: 'text',
>   width: 20,
>   align: 'center'
> },
> ```
> 
> This means we can have the table in a different language if required. Perhaps the text for these four columns could be put as defined strings in common.h for convenience?
> 
> [image]

[image]
Replaced free text with a checkbox and renamed the column

steve6375, 2023-06-09:
I think Tag would be a better heading than Cy because by using letters or numbers in the Tag field the entries can be ordered easily by the user by using sort on Tag column. User can use country if they like or numbers or both, like UK1 and UK2 and they will be ordered correctly by sort.

closed 2023-06-18 by steve6375