schreibfaul1/ESP32-MiniWebRadio #183: Small display bug - RSSI icon is not updated after Alarm screen has been used

steve6375, 2023-06-08:
The Alarm screen is the only screen that overwrites the footer area.
The ShowFooterRSSI() function does not update the signal bar icon if the RSSI value is the same as before.
So this means the graphics from the Alarm screen is still displayed in the footer instead of the bars icon.
To reproduce:

1. Go to Alarm Screen
2. Return back to main Radio screen

Easy fix is to always update the icon in  ShowFooterRSSI() function ?

steve6375, 2023-06-08:
P.S. I also changed display to use # instead of STA  as I think it is more obvious (STA = status?)
Also Slp instead of S:

[image]

Hhaltt, 2023-06-09:
> P.S. I also changed display to use # instead of STA as I think it is more obvious (STA = status?)

STA = Station i think

schreibfaul1, 2023-06-10:
I took it over to the 'small buttons' branch. There the layout is different and the footer is never overwritten. 
If you want to take over the branch, please refresh the content of the SD card from the repository because the icons are smaller.

steve6375, 2023-06-10:
Great. Thanks 
P s. There is another small issue with the clock time adjust screen. If you adjust the volume up or down then clock numerals are destroyed by the volume bar.  I'm not sure why you have the volume controls on this screen anyway?

steve6375, 2023-06-10:
@schreibfaul1
Hi, I have tested the new small icons build and it seems to be OK and I can see you have done a LOT of work!
Many thanks for that.
However, my touch panel on the ILI9341 is now **very unresponsive** and is missing touches a lot (and even seems to activate the wrong button sometimes). Does anyone else see that behaviour now with the new build? I did change the audiotask.cpp as well as update the main.cpp and the new zip file SD contents were added.

My common.h is unchanged from how I had it before
```
#define _SSID               "yourSSID"                        // Your WiFi credentials here
#define _PW                 "yourrouterpwd"
#define TZName              "GMT0BST,M3.5.0/1,M10.5.0" // "CET-1CEST,M3.5.0,M10.5.0/3"    // Timezone (more TZNames in "rtime.cpp")
#define DECODER             0                               // (0)VS1053 , (1)MAX98357A PCM5102A... (2)AC101 (3)ES8388 (4)WM8978
#define TFT_CONTROLLER      0                               // (0)ILI9341, (1)HX8347D, (2)ILI9486a, (3)ILI9486b, (4)ILI9488
#define DISPLAY_INVERSION   0                               // (0) off (1) on
#define TFT_FREQUENCY       40000000                        // 27000000, 40000000, 80000000
#define TFT_ROTATION        1                               // 1 or 3 (landscape)
#define TP_VERSION          0                               // (0)ILI9341, (1)ILI9341RPI, (2)HX8347D, (3)ILI9486, (4)ILI9488
#define TP_ROTATION         3                              // 1 or 3 (landscape)
#define AUDIOTASK_CORE      1                               // 0 or 1
#define AUDIOTASK_PRIO      2                               // 0 ... 24  Priority of the Task (0...configMAX_PRIORITIES -1)
#define SDMMC_FREQUENCY     40000000                        // 40000000, 2000000, 10000000, not every SD Card will run at 40MHz
#define FTP_USERNAME        "esp32"                         // user and pw in FTP Client
#define FTP_PASSWORD        "esp32"
#define CONN_TIMEOUT        250                             // unencrypted connection timeout in ms (http://...)
#define CONN_TIMEOUT_SSL    2000                            // encrypted connection timeout in ms (https://...)

```

steve6375, 2023-06-11:
I downloaded the whole  small icon branch again and now touchpad seems OK 👍

schreibfaul1, 2023-06-11:
##### I downloaded the whole small icon branch again and now touchpad seems OK 👍
Good to know, I actually changed something. The TP controller is basically an AD converter and returns the voltage from the TP in mV. Unfortunately, the tolerances are sometimes quite large. I determine the mean value from three measured values. What is new is that if the measured values ​​deviate by 5px, the entire measurement is repeated. If you still have problems, please remove these lines, maybe they are not necessary.
[image]

steve6375, 2023-06-11:
sorry, I actually downloaded the main build by mistake which is why the previous comment is now deleted! A fresh download of small buttons branch showed same problem, i removed those 4 lines and now better, but as you say, TP result seems to be very erratic and inaccurate so it is difficult to make small buttons work.

steve6375, 2023-06-11:
I used this code to take 5 readings

```

bool TP::read_TP(uint16_t& x, uint16_t& y) {
    uint16_t _y[5];
    uint16_t _x[5];
    uint16_t tmpxy;
    uint8_t i;
    if (digitalRead(TP_IRQ)) return false;
    x = TP_Send(0xD0);  // discard first read
    for (i = 0; i < 5; i++) {
        x = TP_Send(0xD0);  // x
        if ((x < Xmin) || (x > Xmax)) return false;  // outside the display
        x = Xmax - x;
        _x[i] = x / xFaktor;
        y = TP_Send(0x90);  // y
        if ((y < Ymin) || (y > Ymax)) return false;  // outside the display
        y = Ymax - y;
        _y[i] = y / yFaktor;
    }
    x = (_x[0] + _x[1] + _x[2] + _x[3] + _x[4]) / 5;  // take the mean
    y = (_y[0] + _y[1] + _y[2] + _y[3] + _y[4]) / 5;
    //-------------------------------------------------------------
```

and removed this line:

void TP::loop() {
    if (!digitalRead(TP_IRQ)) {
        if (f_loop) {
            **// read_TP(x, y);  // skip first measurement**
            if (read_TP(x, y)) {
                f_loop = false;
                log_i("tp_pressed");
                if (tp_pressed) tp_pressed(x, y);
            }  // read second measurement
        }
    } else {
        if (f_loop == false) {
            log_i("tp_released");
            if (tp_released) tp_released();
            f_loop = true;
        }
    }
}

Now accuracy seems very good :-)

[404461][I][tft.cpp:4636] read_TP(): x=42 y=161
[404461][I][tft.cpp:4636] read_TP(): x=42 y=163
[404461][I][tft.cpp:4636] read_TP(): x=40 y=168
[404465][I][tft.cpp:4636] read_TP(): x=39 y=159
[404470][I][tft.cpp:4636] read_TP(): x=51 y=157
[404476][I][tft.cpp:4538] loop(): tp_pressed
14:10:12 Touchpoint  x=161, y=42
[404547][I][tft.cpp:4544] loop(): tp_released
[405108][I][tft.cpp:4636] read_TP(): x=30 y=217
[405108][I][tft.cpp:4636] read_TP(): x=41 y=227
[405108][I][tft.cpp:4636] read_TP(): x=44 y=223
[405113][I][tft.cpp:4636] read_TP(): x=51 y=218
[405118][I][tft.cpp:4636] read_TP(): x=51 y=215
[405123][I][tft.cpp:4538] loop(): tp_pressed
14:10:13 Touchpoint  x=220, y=43
[405146][I][tft.cpp:4544] loop(): tp_released
[561686][I][tft.cpp:4636] read_TP(): x=215 y=123
[561686][I][tft.cpp:4636] read_TP(): x=208 y=115
[561686][I][tft.cpp:4636] read_TP(): x=205 y=107
[561691][I][tft.cpp:4636] read_TP(): x=208 y=108
[561696][I][tft.cpp:4636] read_TP(): x=206 y=114
[561701][I][tft.cpp:4538] loop(): tp_pressed
14:12:49 Touchpoint  x=113, y=208
[561928][I][tft.cpp:4544] loop(): tp_released
[562845][I][tft.cpp:4636] read_TP(): x=207 y=98
[562845][I][tft.cpp:4636] read_TP(): x=205 y=116
[562845][I][tft.cpp:4636] read_TP(): x=205 y=110
[562849][I][tft.cpp:4636] read_TP(): x=210 y=102
[562855][I][tft.cpp:4636] read_TP(): x=207 y=119
[562860][I][tft.cpp:4538] loop(): tp_pressed
14:12:50 Touchpoint  x=109, y=206
[563084][I][tft.cpp:4544] loop(): tp_released

steve6375, 2023-06-13:
The small icons just don't work: (1) the TP ADC is too inaccurate and (2) the icons are too small for a large finger.
I have spent many hours trying to improve the touchpad accuracy but I am afraid it is hopeless as the results are just too erratic.
I think it would be better to revert back to the large buttons which require less accurate results. The old version worked well and seemed to be much faster too,

steve6375, 2023-06-13:
I modified the code in tft for my ILI9341 touch screen and improved the x.y. code.

```
    if (TP_vers == 0) {  // ILI9341 display
        Xmax = 1825;  // was 1913;     // Values Calibration
        Xmin = 100; // was 150;
        Ymax = 1860;  // was 1944;
        Ymin = 130; // was 220;
        xFaktor = float(Xmax - Xmin) / ILI9341_WIDTH;
        yFaktor = float(Ymax - Ymin) / ILI9341_HEIGHT;
    }
```
   and read pos code with debug output...
```

bool TP::read_TP(uint16_t& x, uint16_t& y) {
    uint16_t _y[6];
    uint16_t _x[6];
    uint16_t xx;
    uint16_t yy;
    uint16_t tmpxy;
    uint8_t i;
    if (digitalRead(TP_IRQ)) return false;

    for (i = 0; i < 6; i++) {
        if ( TP_Send(0xB0) > 2000) return false;  // reject if bad pressure
        if ( TP_Send(0xC0) > 2000) return false;  // reject if bad pressure
        x = TP_Send(0xD0);  // x
        if (x < Xmin) return false;  // x = 1;  // outside the display
        if (x > Xmax) return false;  // x = Xmax - 1;  // outside the display
        _x[i] = x;
        y = TP_Send(0x90);  // y
        if (y < Ymin) return false; // y = 1;  // outside the display
        if (y > Ymax) return false; // y = Ymax - 1;  // outside the display
        _y[i] = y;
    }

    for (i = 0; i < 6; i++) {
    log_i("TP     x=%i, y=%i (Xmax=%i,Ymax=%i,Xmin=%i,Ymin=%i)", _x[i], _y[i], Xmax, Ymax, Xmin, Ymin);
    }

    x = besttwoavg(_x[0], _x[1], _x[2]);
    xx = besttwoavg(_x[3], _x[4], _x[5]);
    y = besttwoavg(_y[0] , _y[1], _y[2]);
    yy = besttwoavg(_y[3], _y[4], _y[5]);
    x = (x + xx) / 2;
    y = (y + yy) / 2;
    log_i("TPbest x=%i, y=%i", x, y);
    x = Xmax - x;
    y = Ymax - y;
    log_i("TPconv x=%i, y=%i", x, y);
    x = (x + 1) / xFaktor;      // avoid divide by 0 errors
    y = (y + 1) / yFaktor;
    log_i("TPFakt X=%i, Y=%i (xFak=%f,yFak=%f)", x, y,xFaktor,yFaktor);

    //-------------------------------------------------------------
```

The problem is that the calibration will vary with different from screen to screen, so best to have large buttons?

closed 2023-06-18 by steve6375