schreibfaul1/ESP32-MiniWebRadio #184: And again about PSRAM

valera39, 2023-06-09:
Dear Wolle, correct me if I am wrong.
The ESP32S3 from Al Thinker uses PSRAM Quad SPI memory and the ESP32S3 from Espressif (which I have) uses PSRAM Octal SPI memory.
Is it enough to change CONFIG_SPIRAM_MODE_QUAD=y to CONFIG_SPIRAM_MODE_OCT=y in sdkconfig.esp32s3 file, in the "SPI RAM config" section, 
and also change CONFIG_SPIRAM_CLK_IO=30 and CONFIG_SPIRAM_CS_IO=26 to the appropriate gpio according to https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html documentation.
Quote from the documentation: 
"For boards with Octal SPI flash/PSRAM memory embedded ESP32-S3-WROOM-1/1U modules, and boards with ESP32-S3-WROOM-2 modules, the pins GPIO35, GPIO36 and GPIO37 are used for the internal communication between ESP32-S3 and SPI flash/PSRAM memory, thus not available for external use."
Looking forward to hearing from you!

schreibfaul1, 2023-06-10:
Congratulations if you own an OCTAL PSRAM and it works now. If I set OCT in menuconfig, the ESP32 crashes, probably because no OCTAL-PSRAM is found.
[image]
Build flags have no effect, the only way is to edit menuconfig or sdkconfig directly.

valera39, 2023-06-10:
Thanks for the quick response!
I got to point 4 in your image, but I could not understand how to navigate in this menu to get to the memory type configuration 
[image]

Maybe I didn't translate your message correctly, but I understood the following: you can't change the sdkconfig.esp32s3 file in a text editor, and you have to edit it in menuconfig?
And maybe in the coming radio updates add configuration for ESP32S3 with PSRAM Octal SPI memory?
Thanks!

schreibfaul1, 2023-06-10:
The idea is to only use the menuconfig. Make sure that esp32s3 is selected as the environment.
In nenuconfig these are the points:

Component config --->
ESP32S3-Specific --->
[*]Support for external, SPI-connected RAM
        SPI RAM config --->
Mode (QUAD/OCT) of SPI RAM chip in use (Quad Mode RSRAM) --->

If you then go back with ESC and save with 'Y', a new sdkconfig.esp32s3 will be created and the previous one marked as '.old'

Dudinu, 2023-06-10:
Hi, valera39
Concerning "I could not understand how to navigate in this menu "  I find here 
https://github.com/platformio/platform-espressif32/issues/423 
an explanation :   https://community.platformio.org/t/menuconfig-navigation-does-not-work-under-windows-10/20881
and a solution :

[image]

valera39, 2023-06-10:
Dear Wolle  and [Dudinu](https://github.com/Dudinu)!
Thank you very much, thanks to your tips I was able to change the type of memory and enable PSRAM But another problem arose, no connection to the network.
Here is the log:
--- Terminal on COM5 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
[   874][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
E (875) esp_core_dump_flash: No core dump partition found!
E (875) esp_core_dump_flash: No coESP32 Chip: ESP32-S3
Arduino Version: 2.0.8
ESP-IDF Version: 4.4.4
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)

00:00:00
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V2     *    
00:00:00        ***************************    
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
[   920][E][Preferences.cpp:50] begin(): nvs_open failed: NOT_INITIALIZED
00:00:00 tft_info: .  init ILI9488
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found
E (254) ledc: ledc_get_duty(739): LEDC is not initialized
00:00:01 setup: ....  seek for stations.csv
00:00:01 setup: ....  stations.csv found
00:00:01 setup: ....  seek for WiFi networks
[  2000][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: ASUS_AC66U
[  2012][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_01
[  2013][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_02
[  2021][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_03
[  2029][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_04
[  2037][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_04
[  2051][E][WiFiGeneric.cpp:680] wifiLowLevelInit(): esp_wifi_init 4353
[  2052][E][WiFiGeneric.cpp:680] wifiLowLevelInit(): esp_wifi_init 4353
00:00:01 WiFi credentials are not correct
00:00:02 WiFi credentials are not correct

Is it possible that the changed sdkconfig.esp32s3 is the cause?

schreibfaul1, 2023-06-10:
I don't know where this comes from, but this is really bad:
`[ 920][E][Preferences.cpp:50] begin(): nvs_open failed: NOT_INITIALIZED`

nvs is used to store the SSID and WiFi password. In addition, nvs is required for the channel list.

Try to load the original sdkconfig.esp32s3 and then only set OCTAL-PSRAM in menuconfig.

valera39, 2023-06-10:
It worked!
20:37:09 AUDIO_info:  PSRAM found, inputBufferSize: 295903 bytes
If the Support for external, SPI-connected RAM and OCTAL Mode PSRAM parameters are enabled, PSRAM is detected.
Thank you so much for your friendly communication and tips!
Always looking forward to new additions and improvements to the best internet radio.
P.S. Maybe in the future add this to the manual for those who use ESP32S3 from Espressif with OCTAL PSRAM

schreibfaul1, 2023-06-10:
Very good! The settings in the menuconfig are then the 'workaround' for your 8MB PSRAM board. The selectable boards don't seem to work, maybe I can create a new *.json file myself, which is then selectable in platformio.ini.

closed 2023-06-17 by valera39