schreibfaul1/ESP32-MiniWebRadio #185: Enhancement: Add ability to set SSID and password using WiFi + browser

steve6375, 2023-06-10:
See https://youtu.be/GOAEjILwqPE?list=PLtK9sSeO6bIqfwJ_q04FFyJPwICcbtf-1

https://www.youtube.com/redirect?event=video_description&redir_token=QUFFLUhqa3VXVzBRekpnQjV1dTN5dGRjN3htXzRvVFQ4QXxBQ3Jtc0ttU2V2dnhsUGRjcjd1NGpaNzJxVXdoYzJGRnVrWTN5WG1NcHp4Sk5qdzlpYmdQRUk2MG5ycUU2MWJ5SDZuM2tHZWZvRm9JbmJQc0lpbjdmRlI1SnN3MldYczFiNkVoZ1A5alN1bUx1RG9SamZBSkVLcw&q=https%3A%2F%2Fgithub.com%2FMakerfabs%2FMakerfabs_FAQ%2Ftree%2Fmaster%2FArduino_ESP32%2Farduino_example%2Fwifi_set_demo&v=GOAEjILwqPE

Is this possible?

schreibfaul1, 2023-06-12:
Das ist eine gute Idee. Wenn die Credentials nicht korrekt sind, könnte ein Accesspoint geöffnte werden um zu einem verfügbaren WiFi Netzwerk ein PW einzugeben. Ich gucke mir den Democode an um zu sehen, was davon verwendbar ist.

Hhaltt, 2023-06-14:
And a super feature would be a re-search of the network after 20-30 seconds. Every day I come to work and turn on the power of the workshop. The radio boots up before the router. Accordingly, I have to restart the radio after the router boots up

schreibfaul1, 2023-06-15:
Hello @Hhaltt,
if no connection to a WiFi network can be established at the start, an access point should be opened to give the user the opportunity to enter the access data via a smartphone/tablet. 
Then a restart after 20..30s would not make sense.
@steve6375 This is the first test, I will certainly improve it a bit before the 'WiFi Manager' is taken over in the master.

steve6375, 2023-06-17:
Did you add instructions to code and Read.Me?

closed 2024-02-11 by github-actions[bot]