schreibfaul1/ESP32-MiniWebRadio #186: Brightness on ILI9341 increases greatly past 95%

steve6375, 2023-06-13:
The current code sets led max as 255 (I think ledc was 254 before due to floating point maths -  SetBrightness now adds 0.5)
Also on my TFT, the change between about 97% and 100% is very big.
Going from 95 to 100 is a big change in brightness.
The following code is my very clumsy attempt to have a fine adjustment between 95 and 100.
I am sure you could write better code but at least it seems to work!
ZZZ line are for my debug

```
/***********************************************************************************************************************
*                                        T F T   B R I G H T N E S S                                                   *
***********************************************************************************************************************/
void setTFTbrightness(uint8_t duty){ //duty 0...100 (min...max)
    if(TFT_BL == -1) return;
    uint32_t d;
    ledcAttachPin(TFT_BL, 1);        //Configure variable led, TFT_BL pin to channel 1
    ledcSetup(1, 12000, 8);          // 12 kHz PWM and 8 bit resolution
    d = duty * 2.55 + 0.5;
    ledcWrite(1, d);
    SerialPrintfln("ledcwrite=%i",d);   // ZZZZZZZZ
}
inline uint32_t getTFTbrightness(){
    return ledcRead(1);
}
inline uint8_t downBrightness(){
    if(_brightness > 5) {
       if (_brightness > 5 && _brightness < 96) 
        {
         _brightness -= _brightness % 5;
         _brightness -= 5;
        }
        else
        {
        _brightness -= 1;
        if (_brightness >= 100) _brightness = 99;
        }
        setTFTbrightness(_brightness);
        showBrightnessBar();
    SerialPrintfln("Bright=%d",_brightness);    // ZZZZZZZZZZ
    } return _brightness;
}
inline uint8_t upBrightness(){
    if(_brightness < 100){
       if (_brightness >= 95) _brightness += 1;
        else if (_brightness < 95)
        {
         _brightness -= _brightness % 5;
         _brightness += 5;
        }
        setTFTbrightness(_brightness);
        showBrightnessBar();
    SerialPrintfln("Bright=%d",_brightness);   // ZZZZZZZZZZZZZ
    }
    return _brightness;
}
inline uint8_t getBrightness(){
    return _brightness;
}

```

Debug o/p
```
20:31:02 Bright=95
20:31:03 ledcwrite=245
20:31:03 Bright=96
20:31:04 ledcwrite=247
20:31:04 Bright=97
20:31:05 ledcwrite=250
20:31:05 Bright=98
20:31:06 ledcwrite=252
20:31:06 Bright=99
20:31:06 ledcwrite=255
20:31:06 Bright=100
20:31:10 ledcwrite=252
20:31:10 Bright=99
20:31:10 ledcwrite=250
20:31:10 Bright=98
20:31:11 ledcwrite=247
20:31:11 Bright=97
20:31:11 ledcwrite=245
20:31:11 Bright=96
20:31:12 ledcwrite=242
20:31:12 Bright=95
20:31:12 ledcwrite=229
20:31:12 Bright=90
20:31:13 ledcwrite=217
20:31:13 Bright=85
20:31:14 ledcwrite=229
20:31:14 Bright=90
20:31:15 ledcwrite=242
20:31:15 Bright=95
20:31:15 ledcwrite=245
20:31:15 Bright=96
20:31:15 ledcwrite=247
20:31:15 Bright=97
20:31:16 ledcwrite=250
20:31:16 Bright=98
20:31:16 ledcwrite=252
20:31:16 Bright=99
20:31:17 ledcwrite=255
20:31:17 Bright=100
20:31:19 ledcwrite=252
20:31:19 Bright=99
20:31:20 ledcwrite=250
20:31:20 Bright=98
20:31:20 ledcwrite=247
20:31:20 Bright=97

```

schreibfaul1, 2023-06-14:
Das werde ich demnächst anpassen. Dabei sit mir aufgefallen, dass 12KHz zu groß ist. Bei manchen Displays (nicht bei allen) gibt es Ungenauigkeiten mit dem TP Controller wenn die Helligkeit nicht 100% ist. Ich kenne die Ursache nicht, aber mit 1.2KHz gibt es die Probleme nicht.

schreibfaul1, 2023-06-15:
(uint)100 * 2.55 is not 255, I corrected that

steve6375, 2023-06-23:
I noticed that for Raspberry Pi a 56 Ohm resistor was wired in series with the LED pin. I am not sure why it was done.
https://youtu.be/3vuK5w8o8dg?t=19
I tried this with a 47 Ohm resistor as I did not have a 56R - it does seem to be better, what do you think?

schreibfaul1, 2023-06-23:
It certainly depends on the display. My displays (the red ones) use a transistor Q1 to switch the backlight. The circuit: http://www.lcdwiki.com/res/MSP4021/4.0inch_SPI_Schematic.pdf is always similar. If I increase the base resistance from 1000 ohms to 1056 it won't change much.

steve6375, 2023-06-23:
My TFT display also has 1k in series too so it shouldn't make much difference.

schreibfaul1, 2023-06-23:
The brightness will not be linear to the duty cycles of the PWM. Maybe a correction would make sense?

closed 2023-07-05 by steve6375