schreibfaul1/ESP32-MiniWebRadio #187: small_buttons branch: Hidden stations are lost from stations.csv

steve6375, 2023-06-14:
Due to latest code change in main.cpp, the hidden status of stations.csv is lost on load/save (X <> Hide mixup)
I also have modified index.h so only unbroken stations are listed using, e.g.
https://de1.api.radio-browser.info/json/countries?hidebroken=true

Plus some other changes in index.h like retry in SD save and Load and many tooltips.

See attached files.
[small_buttons_13_06_2023_index_and_main.zip](https://github.com/schreibfaul1/ESP32-MiniWebRadio/files/11749093/small_buttons_13_06_2023_index_and_main.zip)

steve6375, 2023-06-26:
Do you want to modify index.h so that only working stations are listed by adding **?hidebroken=true**

e.g. `https://de1.api.radio-browser.info/json/countries?hidebroken=true`

in index.h below - this will then only list station URLs which work. ???

```
function selectcategory (presctrl) { // tab Search: preset, select a category
  if(presctrl.value == "bycountry")  {loadJSON('https://de1.api.radio-browser.info/json/countries', gotItems, 'jsonp'); category="country"}
  if(presctrl.value == "bylanguage") {loadJSON('https://de1.api.radio-browser.info/json/languages', gotItems, 'jsonp'); category="language"}
  if(presctrl.value == "bytag")      {loadJSON('https://de1.api.radio-browser.info/json/tags',      gotItems, 'jsonp'); category="tag"}
}

function selectitem (presctrl) { // tab Search: preset, select a station
  if(category == "country")  loadJSON('https://de1.api.radio-browser.info/json/stations/bycountry/'  + presctrl.value, gotStations, 'jsonp')
  if(category == "language") loadJSON('https://de1.api.radio-browser.info/json/stations/bylanguage/' + presctrl.value, gotStations, 'jsonp')
  if(category == "tag")      loadJSON('https://de1.api.radio-browser.info/json/stations/bytag/'      + presctrl.value, gotStations, 'jsonp')
}

```

schreibfaul1, 2023-06-27:
good idea

closed 2023-07-05 by steve6375