schreibfaul1/ESP32-MiniWebRadio #188: All BBC Radio streaming no longer works

steve6375, 2023-06-16:
around midday yesterday 15 June 2023, all BBC streaming URLs no longer work.
AFAIK the BBC have stopped this service (shoutcast).
https://www.bbc.co.uk/sounds/help/questions/recent-changes-to-bbc-sounds/shoutcast-closure
The BBC World Service URL still works.
It would be nice to get the new HLS or DASH working, but I am not sure if it is possible?

schreibfaul1, 2023-06-16:
MPEG-DASH won't work, but HLS will
I have no idea how current this is:
https://github.com/mattconsto/bbc-radio-playlists/blob/master/BBC%20Radio%20128kbps.m3u8
The streams are not accessible in Germany, "Access Denied"

steve6375, 2023-06-16:
https://garfnet.org.uk/cms/tables/radio-frequencies/internet-radio-player/

I found HLS streams here but they dont seem to work in MiniWebRadio but do in a VLS player for instance, e.g.
https://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/uk/sbr_high/ak/bbc_radio_fourfm.m3u8
However, if I download that as a file and open it up. I can use the URL inside it
http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=128000.norewind.m3u8

shouldn't the first url work in MiniWebRadio?

schreibfaul1, 2023-06-16:
I have no idea, these URLs are not accessible to me without a VPN. The only stream from the list https://gist.github.com/bpsib/67089b959e4fa898af69fea59ad74bc3?permalink_comment_id=4395939 that works is BBC World Service. 
M3U8 is also used for IPTV. Videos don't work.

schreibfaul1, 2023-06-16:
These are examples of stations (m3u8 HLS streams) that can be received worldwide and work with MiniWebRadio
http://stream.revma.ihrhls.com/zc4882/hls.m3u8 // USA NAT, iHeartCountry Radio Top 20
http://stream.revma.ihrhls.com/zc1401/hls.m3u8 // USA Albuquerque NM, iHeartPodcast AM 1350
http://stream.revma.ihrhls.com/zc3348/hls.m3u8 // USA Allentown PA, iHeartPodcast AM 1470
http://stream.revma.ihrhls.com/zc6892/hls.m3u8 // USA NAT, iHeartRadio Bluegrass 
http://stream.revma.ihrhls.com/zc7368/hls.m3u8 // USA NAT, iHeartRadio Freestyle
http://stream.revma.ihrhls.com/zc5158/hls.m3u8 // USA NAT, iHeartRadio Music Festival
http://arcast.com.ar:1935/radio/radionar.stream/playlist.m3u8
http://lampsifmlive.mdc.akamaized.net/strmLampsi/userLampsi/playlist.m3u8
http://6182bfa7cd933.streamlock.net:1935/Radiokriti/Radiokriti/playlist.m3u8
https://cdnlive.codev8.net/canalextremaduralive/smil:channel2.smil/playlist.m3u8
http://eldoradolive02.akamaized.net/hls/live/2043453/eldorado/master.m3u8
https://air.pc.cdn.bitgravity.com/air/live/pbaudio001/playlist.m3u8
http://18683.live.streamtheworld.com:3690/XHPIFMAAC/HLSTS/playlist.m3u8
http://26373.live.streamtheworld.com:3690/XHQQ_FMAAC/HLSTS/playlist.m3u8
http://20283.live.streamtheworld.com/XHQQ_FMAAC/HLSTS/playlist.m3u8
http://19003.live.streamtheworld.com/BLUE_FM_100_7AAC/HLSTS/playlist.m3u8
http://mdstrm.com/audio/6237761f64b72676f05cf6ab/live.m3u8
https://live7digi.antenaplay.ro/radiozu/radiozu.m3u8
http://us-b4-p-e-zs14-audio.cdn.mdstrm.com/live-audio-aw-bkp/6237761f64b72676f05cf6ab/playlist.m3u8?listeningSessionID=62a10f0200a9bf00_2098051_cKYJo206__0000001WWlO
https://hls-01-regions.emgsound.ru/11_msk/playlist.m3u8      EuropaPlus
https://hls-01-europaplus-top40.emgsound.ru/21/playlist.m3u8 Top40
https://hls-01-europaplus-new.emgsound.ru/27/playlist.m3u8   NEW
https://hls-01-europaplus-party.emgsound.ru/22/playlist.m3u8 PARTY
https://hls-01-europaplus-light.emgsound.ru/23/playlist.m3u8 LIGHT
https://digitacdn.akamaized.net/hls/live/629243/radiosuomipop/master.m3u8
https://ndtvprofitelemarchana.akamaized.net/hls/live/2003680/ndtvprofit/ndtvprofitmaster.m3u8
http://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/nonuk/sbr_low/ak/bbc_radio_fourfm.m3u8 
https://live-cdn.sr.se/pool2/p2musik/p2musik.isml/p2musik-audio=192000.m3u8
https://bss1.neterra.tv/magicfm/magicfm.m3u8

steve6375, 2023-06-16:
OK - in UK
Console output in PlatformIO shows
```
14:40:56 AUDIO_info:  new request: "http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d320000.norewind.m3u8"
14:40:56 StreamTitle: HTTP/1.0 400 Bad Request
```

so I think it should convert %3d to =  which is why 
https://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/uk/sbr_high/ak/bbc_radio_fourfm.m3u8
fails

schreibfaul1, 2023-06-16:
You don't need to convert the %3.
http://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/nonuk/sbr_low/ak/bbc_world_service.m3u8 is free to air and runs without interruptions
m3u8 is a playlist containing short segments. The m3u8 content is constantly changing.

steve6375, 2023-06-16:
http://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/nonuk/sbr_low/ak/bbc_world_service.m3u8 
but does not work
because it contains a URL with a %3d inside it.
vs1053_ext.cpp bool VS1053::httpPrint  does not change %3d to = and so the new request does not play.

steve6375, 2023-06-16:
After much confusion, there seems to be two issues.
(1) First, clicking on the sendstation url does not work if there is a %3d in the url
[image]
It works if the URL is added into the stations list and you select that station number, but it doesn't work if you click on the button in the web app to send the URL to the esp32.
It seems to get converted into a / when it should be %3d
```
21:03:43 setStationViaURL http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio/128000.norewind.m3u8
21:03:43 IN connecttohost http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio/128000.norewind.m3u8
21:03:43 IN connecttohost nopipe http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio/128000.norewind.m3u8
21:03:43 AUDIO_info:  Connect to new host: "http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio/128000.norewind.m3u8"
21:03:43 AUDIO_info:  Song stopped correctly after 0 msec
```

(2) The URL official url https://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/uk/sbr_high/ak/bbc_radio_fourfm.m3u8
doesn't work for me in UK but the URL inside the m3u8 file does work

Here is the bbci url which did not work - it seems to be requesting 320000 data rate ???
```
21:17:04 in setStation 25
21:17:04 content BBC Radio 4  hls#https://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/uk/sbr_high/ak/bbc_radio_fourfm.m3u8
21:17:04 action: ...  switch to station 25
21:17:04 IN connecttohost https://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/uk/sbr_high/ak/bbc_radio_fourfm.m3u8
21:17:04 IN connecttohost nopipe https://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/uk/sbr_high/ak/bbc_radio_fourfm.m3u8
21:17:04 AUDIO_info:  Connect to new host: "https://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/uk/sbr_high/ak/bbc_radio_fourfm.m3u8"
21:17:04 AUDIO_info:  Song stopped correctly after 20 msec
21:17:04 AUDIO_info:  SSL has been established in 544 ms, free Heap: 62120 bytes
21:17:04 file "/logo/s/BBC Radio 4  hls.jpg" not found
[3603863][E][vs1053_ext.cpp:1189] readPlayListData(): m3u8 playlists requires PSRAM enabled!
21:17:05 AUDIO_info:  new request: "s-hls-uk-live.akamaized.net" /pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d320000.norewind.m3u8
21:17:05 StreamTitle: HTTP/1.0 400 Bad Request
21:17:05 AUDIO_info:  Song stopped correctly after 0 msec
21:17:05 Audiocodec:  AAC
```

and here is the URL inside of the file when used directly. It seems to request 128000 data rate. Maybe this is why it works?

http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d128000.norewind.m3u8

```
21:19:49 AUDIO_info:  new request: "as-hls-uk-live.akamaized.net" /pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=128000-263584874.ts
21:19:49 in setStation 24
21:19:49 content BBC Radio 4 percent#http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d128000.norewind.m3u8
21:19:49 action: ...  switch to station 24
21:19:49 IN connecttohost http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d128000.norewind.m3u8
21:19:49 IN connecttohost nopipe http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d128000.norewind.m3u8
21:19:49 AUDIO_info:  Connect to new host: "http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d128000.norewind.m3u8"
21:19:49 AUDIO_info:  Song stopped correctly after 20 msec
21:19:49 AUDIO_info:  Connection has been established in 12 ms, free Heap: 107324 bytes
[3768828][E][vs1053_ext.cpp:1189] readPlayListData(): m3u8 playlists requires PSRAM enabled!
21:19:49 AUDIO_info:  new request: "as-hls-uk-live.akamaized.net" /pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=128000-263584878.ts
21:19:50 file "/logo/s/BBC Radio 4 percent.jpg" not found
21:19:50 Audiocodec:  AAC
21:19:55 AUDIO_info:  new request: "as-hls-uk-live.akamaized.net" /pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d128000.norewind.m3u8
[3774906][E][vs1053_ext.cpp:1189] readPlayListData(): m3u8 playlists requires PSRAM enabled!
21:19:56 AUDIO_info:  new request: "as-hls-uk-live.akamaized.net" /pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=128000-263584879.ts
```

Arne65, 2023-06-16:
Hello Steve
this stream runs with a DAC have tested it

steve6375, 2023-06-16:
> 

Which URL and are you in UK?

Arne65, 2023-06-16:
http://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/nonuk/sbr_low/ak/bbc_world_service.m3u8

Arne65, 2023-06-16:
In germany

schreibfaul1, 2023-06-16:
I also only tested it with an external DAC, not with VS1053. There was indeed an error when extracting the m3u8 playlist. I updated the vs0153_ext library, hope it works now. The library must be reloaded. 
[image]

With the exception of the 'World Service', BBC Radio cannot be received outside of the UK.

steve6375, 2023-06-16:
I did a full rebuild but still an issue - it is losing the first letter of the domain name

[image]

Should start with as.hls   not s.hls

steve6375, 2023-06-16:
This is working though.
http://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/nonuk/sbr_low/ak/bbc_world_service.m3u8
```
23:22:59 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_world_service/bbc_world_service.isml/bbc_world_service-audio=96000-263586028.ts"
23:22:59 setStationViaURL http://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/nonuk/sbr_low/ak/bbc_world_service.m3u8
23:22:59 IN connecttohost http://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/nonuk/sbr_low/ak/bbc_world_service.m3u8
23:22:59 IN connecttohost nopipe http://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/nonuk/sbr_low/ak/bbc_world_service.m3u8
23:22:59 AUDIO_info:  Connect to new host: "http://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/nonuk/sbr_low/ak/bbc_world_service.m3u8"
23:22:59 AUDIO_info:  Song stopped correctly after 0 msec
23:22:59 AUDIO_info:  Connection has been established in 23 ms, free Heap: 102440 bytes
[1633079][E][vs1053_ext.cpp:1189] readPlayListData(): m3u8 playlists requires PSRAM enabled!
23:22:59 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_world_service/bbc_world_service.isml/bbc_world_service-audio%3d96000.norewind.m3u8"
23:22:59 AUDIO_info:  The host has disconnected, reconnecting
23:22:59 file "/logo/s/.jpg" not found
[1633184][E][vs1053_ext.cpp:1189] readPlayListData(): m3u8 playlists requires PSRAM enabled!
23:22:59 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_world_service/bbc_world_service.isml/bbc_world_service-audio=96000-263586030.ts"
23:23:00 Audiocodec:  AAC
23:23:05 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_world_service/bbc_world_service.isml/bbc_world_service-audio=96000-263586031.ts"
23:23:11 AUDIO_info:  new request: "http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_world_service/bbc_world_service.isml/bbc_world_service-audio=96000-263586032.ts"

```

steve6375, 2023-06-16:
Losing first letter on all? UK URLs
[image]

Here are contents of the two files - uk one has an extra line??

```
#EXTM3U
#EXT-X-VERSION:3
#EXT-X-STREAM-INF:PROGRAM-ID=1,BANDWIDTH=101760,CODECS="mp4a.40.5"
http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d96000.norewind.m3u8
https://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/uk/sbr_low/ak/bbc_radio_fourfm.m3u8
```

```
#EXTM3U
#EXT-X-VERSION:3
#EXT-X-STREAM-INF:PROGRAM-ID=1,BANDWIDTH=101760,CODECS="mp4a.40.5"
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_world_service/bbc_world_service.isml/bbc_world_service-audio%3d96000.norewind.m3u8

```

schreibfaul1, 2023-06-17:
this is strange
https://a.files....network.m3u8
contains as a destination
http://as-hls-uk.....m3u8

A secure connection (https) is forwarded to an insecure one (http).
That would explain why a character is missing, because everything after 'https://' is read out.
I added another fix, but I can't test it.

The relevant code is in `VS1053::httpPrint(const char* host)`

old:
````c++
	if(m_f_ssl) h_host = strdup(host + 8);
	else
		h_host = strdup(host + 7);
````

new:

````c++
    if(startsWith(host, "https")){
        h_host = strdup(host + 8);
        m_f_ssl = true;
    }
    else{
        h_host = strdup(host + 7);
        m_f_ssl = false;
    }
````

steve6375, 2023-06-17:
I found the 2nd bug!
m_f_ssl is incorrect because it shows if https or http was used for the last line in the m3u8 list - not the actual line to be played.
I fixed it by adding lines to set m_f_ssl to actual url being used and now it works!
```
bool VS1053::httpPrint(const char* host) {
	// user and pwd for authentification only, can be empty

	if(host == NULL) {
		AUDIO_INFO("Hostaddress is empty");
		return false;
	}

	char* h_host = NULL;  // pointer of l_host without http:// or https://

    	if(startsWith(host, "https")) m_f_ssl = true;   // NEW CODE
	else
		m_f_ssl = false;

	if(m_f_ssl) h_host = strdup(host + 8);
	else
		h_host = strdup(host + 7);

```

schreibfaul1, 2023-06-17:
With this change, does BBC Radio now work? If yes, I adopt that in VS1053_ext and audioI2S. Both libraries often have the same routines.

steve6375, 2023-06-17:
yes, BBC Radio works now. 👍

closed 2023-06-18 by steve6375

jonesy2, 2023-07-21:
For info, quite a few BBC streams are available outside the UK (no VPN required)
follow the instructions here
https://gist.github.com/bpsib/67089b959e4fa898af69fea59ad74bc3#file-bbc-radio-hls-uk-m3u-L131
The bit rates are lower, but still 96k
For example Radio 2 is here
http://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/nonuk/sbr_low/ak/bbc_radio_two.m3u8