schreibfaul1/ESP32-MiniWebRadio #189: Sending URL with %3d in string from web UI does not work

steve6375, 2023-06-17:
e.g. http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d128000.norewind.m3u8

The esp32 receives
http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio/128000.norewind.m3u8

If I send http://x%3dy.m3u8, i get on esp32:
[image]

```
10:42:06 WS_onCmd:    cmd="stationURL", params="http://x/y.m3u8", arg="version=0.5546679962762022"
10:42:06 setStationViaURL http://x/y.m3u8
10:42:06 IN connecttohost http://x/y.m3u8
10:42:06 IN connecttohost nopipe http://x/y.m3u8     
10:42:06 AUDIO_info:  Connect to new host: "http://x/y.m3u8"
10:42:06 AUDIO_info:  Song stopped correctly after 0 msec
10:42:06 AUDIO_info:  Connect to "x/y.m3u8" on port 80, extension "/"
[1086111][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for x/y.m3u8
10:42:06 AUDIO_info:  Request http://x/y.m3u8 failed!
10:42:06 StreamTitle:
10:42:07 file "/logo/s/.jpg" not found
10:42:07 WS_onCmd:    cmd="SD/logo/.jpg", params="t=1686991335258", arg=""
10:42:07 file "/logo/s/.jpg" not found
10:42:23 WS_onCmd:    cmd="ping", params="", arg=""

```

steve6375, 2023-06-17:
Same issue with this button:
[image]

I fixed by adding code in index.h in two places:

```
function setstation () { // Radio: button play - Enter a streamURL here....
  var theUrl = '/stationURL?' + station.value + '&version=' + Math.random()
  theUrl = theUrl.replace(/%3d/g, '=') // %3d convert to =
  theUrl = theUrl.replace(/%21/g, '!') // 
  theUrl = theUrl.replace(/%22/g, '"') // 
  theUrl = theUrl.replace(/%23/g, '#') // 
  theUrl = theUrl.replace(/%3f/g, '?') // 
  theUrl = theUrl.replace(/%40/g, '@') // 
```

```
function teststreamurl () { // Search: button play - enter a url to play from
  var theUrl = '/stationURL?' + streamurl.value + '&version=' + Math.random()
  theUrl = theUrl.replace(/%3d/g, '=') // %3d convert to =
  theUrl = theUrl.replace(/%21/g, '!') // 
  theUrl = theUrl.replace(/%22/g, '"') // 
  theUrl = theUrl.replace(/%23/g, '#') // 
  theUrl = theUrl.replace(/%3f/g, '?') // 
  theUrl = theUrl.replace(/%40/g, '@') // 
```

schreibfaul1, 2023-06-17:
Added your improvements and merged them into the master. Thanks a lot for this

steve6375, 2023-06-17:
Thanks for a great project.
Did you see my better TP code which is even better than the one you just published?
Even so, because each TP is different, I think users will try new release and complain about wrong button being activated.
Also I have improved index.h to retry on a failed POST or SEND and it makes web buttons MUCH more reliable.
\
[src.zip](https://github.com/schreibfaul1/ESP32-MiniWebRadio/files/11779584/src.zip)
[tft.zip](https://github.com/schreibfaul1/ESP32-MiniWebRadio/files/11779590/tft.zip)

closed 2023-06-18 by steve6375