schreibfaul1/ESP32-MiniWebRadio #190: garbled noise when a new station is selected

steve6375, 2023-06-17:
When a new station starts to play there is a sometimes quite loud garbled noise for approx 1/2 second as new station starts to play.
Would it be possible to 
mute
change station
delay 500 milliseconds
unmute

to avoid nasty noise?

steve6375, 2023-06-17:
P.S. seems to be worse when new station uses a different data rate

schreibfaul1, 2023-06-17:
This could be leftovers in the internal memory of the VS1053. This is not the case with an external DAC. Maybe there is a better solution than a delay(500)

schreibfaul1, 2023-06-17:
I have 'garbled noise' even after a SW reset of the VS1053. Now I added 200ms mute in processWebStream()

steve6375, 2023-06-17:
I have tested with various streams and different delays
try changing between
https://stream.live.vc.bbcmedia.co.uk/bbc_world_service
and
http://wsdownload.bbc.co.uk/worldservice/meta/live/shoutcast/mp3/eieuk.pls

This seems to be worse case! Quite a long delay is needed > 1 second, to avoid bad noise

Data sheet says to send a cancel command and wait for stream to stop...

steve6375, 2023-06-17:
10.5.2 Cancelling Playback
Cancelling playback of a song is a normal operation when the user wants to jump to another
song while doing playback.
1. Send a portion of an audio file to VS1053b.
2. Set SCI_MODE bit SM_CANCEL.
3. Continue sending audio file, but check SM_CANCEL after every 32 bytes of data. If it
is still set, goto 3. If SM_CANCEL doesn’t clear after 2048 bytes or one second, do a
software reset (this should be extremely rare).
4. When SM_CANCEL has cleared, read extra parameter value endFillByte (Chapter 10.11).
5. Send 2052 bytes of endFillByte[7:0].
6. HDAT0 and HDAT1 should now both contain 0 to indicate that no format is being decoded.
You can now send the next audio file.

schreibfaul1, 2023-06-17:
Okay, those are the special cases, the BBC streams don't go through processWebStream().
For that there is `processWebStreamHLS() ` for aac segments and
`processWebStreamTS() ` for TS packages

steve6375, 2023-06-17:
I used main.cpp to add delay. Could be optional only if decoder used?

```
void setStation(uint16_t sta){
    SerialPrintfln("in setStation %d", sta);
    //SerialPrintfln("sta %d, _cur_station %d", sta, _cur_station );
    if(sta == 0) return;
    if(sta > _sum_stations) sta = _cur_station;
    sprintf (_chbuf, "station_%03d", sta);
    String content = stations.getString(_chbuf);
    SerialPrintfln("content %s", content.c_str());
    _stationName_nvs = content.substring(0, content.indexOf("#")); //get stationname
    content = content.substring(content.indexOf("#") + 1, content.length()); //get URL
    content.trim();
    free(_stationURL);
    _stationURL = strdup(content.c_str());
    _homepage = "";
    if(_state == RADIO) clearStreamTitle();

    SerialPrintfln("action: ...  switch to station " ANSI_ESC_CYAN "%d", sta);

    if(_f_isWebConnected && sta == _cur_station && _state == RADIO){ // Station is already selected
        _f_newStreamTitle = true;
    }
    else{
        audioSetVolume(0);
        _streamTitle[0] = '\0';
        _icyDescription[0] = '\0';
        _f_newStreamTitle = true;
        _f_newIcyDescription = true;
        connecttohost(_stationURL);
        delay(1200);            // 2000 is ok, 1500 is ok, 250,400,600 is NOT OK, 
        audioSetVolume(_cur_volume);
    }
    _cur_station = sta;
    StationsItems();
    if(_state == RADIO || _state == RADIOico) showLogoAndStationName();
    showFooterStaNr();
}
```

schreibfaul1, 2023-06-17:
I read what is in the data sheet. As described, HDAT0 and HDAT1 go to '0'. I don't think there is anything that can be improved here.

schreibfaul1, 2023-06-18:
It looks like as long as the noise comes until the VS1053 knows the correct sampleRate or codec? I have now muted this for a short time via a register. is it better now?

steve6375, 2023-06-18:
Where is reg change code?

I have reduced volume, stopped play, start new connection, delay. restore volume in setstation code in main.
This seems OK and should be better as no click when vol(0) is set.

```
void setStation(uint16_t sta){
    SerialPrintfln("in setStation %d", sta);
    //SerialPrintfln("sta %d, _cur_station %d", sta, _cur_station );
    if(sta == 0) return;
    if(sta > _sum_stations) sta = _cur_station;
    sprintf (_chbuf, "station_%03d", sta);
    String content = stations.getString(_chbuf);
    SerialPrintfln("content %s", content.c_str());
    _stationName_nvs = content.substring(0, content.indexOf("#")); //get stationname
    content = content.substring(content.indexOf("#") + 1, content.length()); //get URL
    content.trim();
    free(_stationURL);
    _stationURL = strdup(content.c_str());
    _homepage = "";
    if(_state == RADIO) clearStreamTitle();

    SerialPrintfln("action: ...  switch to station " ANSI_ESC_CYAN "%d", sta);

    if(_f_isWebConnected && sta == _cur_station && _state == RADIO){ // Station is already selected
        _f_newStreamTitle = true;
    }
    else{
        int v = _cur_volume;
        while ( v > 0) audioSetVolume(v--);
        if (audioIsRunning()) audioStopSong();  // crashes if stop when no audio is running
        _streamTitle[0] = '\0';
        _icyDescription[0] = '\0';
        _f_newStreamTitle = true;
        _f_newIcyDescription = true;
        connecttohost(_stationURL);
        if(DECODER == 0) delay(420);            // 1000, 500 400(just) OK      300,250 NOT OK
        while ( v < _cur_volume) audioSetVolume(v++);
    }
    _cur_station = sta;
    StationsItems();
    if(_state == RADIO || _state == RADIOico) showLogoAndStationName();
    showFooterStaNr();
}

```

schreibfaul1, 2023-06-18:
you can also do it like this, delay() stops the whole core, maybe vTaskDelay() is better?

steve6375, 2023-07-04:
seems to be much better in latest release. Thanks!

closed 2023-07-04 by steve6375