schreibfaul1/ESP32-MiniWebRadio #192: compile error for esp-s3-12k

clear-sky, 2023-06-17:
Hi dear wolle
i was looking forward for you!
in this new version i get compile error 
please see my terminal output:
```

Processing esp32 (board: esp32dev; platform: https://github.com/platformio/platform-espressif32.git#v6.2.0; framework: arduino, espidf)
-----------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.2.0+sha.1d6e199) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 2.0.8+sha.d3254f7 
 - framework-espidf @ 3.40404.0 (4.4.4) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
/home/conrad/.platformio/penv/.espidf-4.4.4/bin/python: No module named pip
CalledProcessError: Command '['/home/conrad/.platformio/penv/.espidf-4.4.4/bin/python', '-m', 'pip', 'list', '--format=json', '--disable-pip-version-check']' returned non-zero exit status 1.:
  File "/home/conrad/.local/lib/python3.11/site-packages/platformio/builder/main.py", line 174:
    env.SConscript("$BUILD_SCRIPT")
  File "/home/conrad/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/conrad/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/conrad/.platformio/platforms/espressif32@src-71ae141cfc7678fbfef85a32979c43f9/builder/main.py", line 312:
    target_elf = env.BuildProgram()
  File "/home/conrad/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Util/envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "/home/conrad/.local/lib/python3.11/site-packages/platformio/builder/tools/piobuild.py", line 60:
    env.ProcessProgramDeps()
  File "/home/conrad/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Util/envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "/home/conrad/.local/lib/python3.11/site-packages/platformio/builder/tools/piobuild.py", line 120:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "/home/conrad/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Util/envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "/home/conrad/.local/lib/python3.11/site-packages/platformio/builder/tools/piobuild.py", line 331:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "/home/conrad/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 662:
    return method(*args, **kw)
  File "/home/conrad/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/conrad/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/conrad/.platformio/platforms/espressif32@src-71ae141cfc7678fbfef85a32979c43f9/builder/frameworks/espidf.py", line 1192:
    install_python_deps()
  File "/home/conrad/.platformio/platforms/espressif32@src-71ae141cfc7678fbfef85a32979c43f9/builder/frameworks/espidf.py", line 1101:
    installed_packages = _get_installed_pip_packages(python_exe_path)
  File "/home/conrad/.platformio/platforms/espressif32@src-71ae141cfc7678fbfef85a32979c43f9/builder/frameworks/espidf.py", line 1067:
    pip_output = subprocess.check_output(
  File "/usr/lib/python3.11/subprocess.py", line 466:
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.11/subprocess.py", line 571:
    raise CalledProcessError(retcode, process.args,
============================== [FAILED] Took 0.25 seconds ==============================

Processing esp32s3 (board: esp32-12k; platform: https://github.com/platformio/platform-espressif32.git#v6.2.0; framework: arduino, espidf)
-----------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-12k.html
PLATFORM: Espressif 32 (6.2.0+sha.1d6e199) > AI Thinker
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 2.0.8+sha.d3254f7 
 - framework-espidf @ 3.40404.0 (4.4.4) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
/home/conrad/.platformio/penv/.espidf-4.4.4/bin/python: No module named pip
CalledProcessError: Command '['/home/conrad/.platformio/penv/.espidf-4.4.4/bin/python', '-m', 'pip', 'list', '--format=json', '--disable-pip-version-check']' returned non-zero exit status 1.:
  File "/home/conrad/.local/lib/python3.11/site-packages/platformio/builder/main.py", line 174:
    env.SConscript("$BUILD_SCRIPT")
  File "/home/conrad/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/conrad/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/conrad/.platformio/platforms/espressif32@src-71ae141cfc7678fbfef85a32979c43f9/builder/main.py", line 312:
    target_elf = env.BuildProgram()
  File "/home/conrad/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Util/envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "/home/conrad/.local/lib/python3.11/site-packages/platformio/builder/tools/piobuild.py", line 60:
    env.ProcessProgramDeps()
  File "/home/conrad/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Util/envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "/home/conrad/.local/lib/python3.11/site-packages/platformio/builder/tools/piobuild.py", line 120:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "/home/conrad/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Util/envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "/home/conrad/.local/lib/python3.11/site-packages/platformio/builder/tools/piobuild.py", line 331:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "/home/conrad/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 662:
    return method(*args, **kw)
  File "/home/conrad/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/conrad/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/conrad/.platformio/platforms/espressif32@src-71ae141cfc7678fbfef85a32979c43f9/builder/frameworks/espidf.py", line 1192:
    install_python_deps()
  File "/home/conrad/.platformio/platforms/espressif32@src-71ae141cfc7678fbfef85a32979c43f9/builder/frameworks/espidf.py", line 1101:
    installed_packages = _get_installed_pip_packages(python_exe_path)
  File "/home/conrad/.platformio/platforms/espressif32@src-71ae141cfc7678fbfef85a32979c43f9/builder/frameworks/espidf.py", line 1067:
    pip_output = subprocess.check_output(
  File "/usr/lib/python3.11/subprocess.py", line 466:
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.11/subprocess.py", line 571:
    raise CalledProcessError(retcode, process.args,
============================== [FAILED] Took 0.25 seconds ==============================

Environment    Status    Duration
-------------  --------  ------------
esp32          FAILED    00:00:00.251
esp32s3        FAILED    00:00:00.254
========================= 2 failed, 0 succeeded in 00:00:00.505 =========================
```

schreibfaul1, 2023-06-17:
I suspect the Python module pip is missing.
If you call pip in the VS Code terminal it should look like this:
[image]
Maybe `pip install pip` will help
[image]

clear-sky, 2023-06-18:
hi dear wolle
well well well
problem is not pip
deleted .platformio in my home directory and now problem can be seen better
if i use default partition table everything goes well
but if use my custom partition table problem happens
this is error message

`Partitions tables occupies 8.0MB of flash (8388608 bytes) which does not fit in configured flash size 4MB. Change the flash size in menuconfig under the 'Serial Flasher Config' menu.`

run pio run -t menuconfig in terminal and changed flash size to 8MB

this is the path
main menu>serial flasher config>flash size

in this condition error message does not disappear
but if i use default partition table and change flash size to 8MB error message disappears
do you have any idea?
this is my custom partition table (for esp-s3-12k)
```
# Name,     Type,   SubType,   Offset,   Size,     Flags
  phy_init, data,   phy,       0x9000,   0x7000,
  factory,  app,    factory,   0x10000,  0x7B0000,
  nvs,      data,   nvs,       0x7C0000, 0x40000,
```

schreibfaul1, 2023-06-18:
solved very well! i have ordered a S3 board, after receipt i will add new board definitions in platformio.ini so you can select boards with different flash.

clear-sky, 2023-06-18:
i think problem has not been solved!
my reason is partition size is used from default partition table !
i wish i were german! if i were it will be my honor to buy it and  bring it to your home
_thanks with respect_

closed 2023-06-20 by clear-sky

clear-sky, 2023-06-20:
for some people that want to change partition size with s3 module:
1-edit miniwebradio.csv (example can be seen above)
2-open platformio.ini and delete [env:esp32] section
3-edit platformio.ini (every line you see that points to 4mB size
4-run pio -t menuconfig and change partition size in this path:
main menu>serial flasher config>flash size