schreibfaul1/ESP32-MiniWebRadio #194: v 2.8 doesn't work

valera39, 2023-06-17:
Hello Wolle!
Configuration ESP32S3 N16R8, VS1053. In platformio.ini I have recommended the line board = ESP32-S3-DevKitC-1-N16R8 ;chipmodel ESP32S3, USBtoTTL (S3-DevKitC-1)
But PSRAM is not initialized. Do I need to edit sdkconfig.esp32s3 again ? Stations are not playing and logos are not being loaded. Here is the log:
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x19 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
E (502) psram: PSRAM ID read error: 0x00ffffff
[   502][W][esp32-hal-psram.c:71] psramInit(): PSRAM init failed!
E (504) esp_core_dump_flash: No core dump partition found!
E (507) esp_core_dump_flash: No core dump partition found!
ESP32 Chip: ESP32-S3
Arduino Version: 2.0.8
ESP-IDF Version: 4.4.4
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)

00:00:00
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V2     *    
00:00:00        ***************************    
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9488
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found
E (319) ledc: ledc_get_duty(739): LEDC is not initialized
00:00:01 setup: ....  seek for stations.csv
00:00:01 setup: ....  stations.csv found
00:00:01 setup: ....  seek for WiFi networks
[  1711][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: ASUS_AC66U
[  1722][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_01
[  1723][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_02
[  1728][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_03
[  1736][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_04
[  1744][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_04
[  8221][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[  8222][I][WiFiMulti.cpp:119] run(): [WIFI] 5 networks found
[  8223][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 2C:FD:A1:C2:44:F0 SSID: ASUS_AC66U Channel: 1 (-66)
[  8292][I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
00:00:07 WiFI_info:   Connecting WiFi...
00:00:07 setup: ....  connected to ASUS_AC66U, IP address is 192.168.1.55
00:00:07 ftpServer:   Buffers allocated: 4622 bytes
00:00:07 rtime_info:  Initializing SNTP
00:00:07 rtime_info:  Waiting for system time to be set... (1/10)
00:00:09 rtime_info:  Waiting for system time to be set... (2/10)
23:23:02 rtime_info:  23:23:02
23:23:02 audiotask:   is pinned to core 1
23:23:02 audiotask:   priority is 2
23:23:02 VS1053 chipID = -1350368918, version = 3
23:23:02 setup: ....  Number of saved stations: 88
23:23:02 setup: ....  current station number: 2
23:23:02 setup: ....  current volume: 2
23:23:02 setup: ....  last connected host: http://play.global.audio/bgradio_low.ogg
23:23:02 setup: ....  connection timeout: 250 ms
23:23:02 setup: ....  connection timeout SSL: 2000 ms
23:23:03 action: ...  current volume is 2
[ 13316][E][Preferences.cpp:483] getString(): nvs_get_str len fail: station_002 NOT_FOUND
23:23:03 action: ...  switch to station 2
23:23:03 AUDIO_info:  Connect to new host: "http://"
23:23:03 AUDIO_info:  Song stopped correctly after 0 msec
23:23:03 AUDIO_info:  PSRAM not found, inputBufferSize: 14399 bytes
[ 13598][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for 
23:23:03 AUDIO_info:  Request http:// failed!
23:23:03 StreamTitle:
23:23:04 file "/logo/m/.jpg" not found
23:23:04 WiFI_info:   RSSI is -65 dB
23:23:04 DLNA_info:   SSDP search for media servers started
23:23:08 DLNA_info:   SSDP query discovered 1 media servers
23:23:08 DLNA_info:   connected successfully to 192.168.1.4:6000
23:23:08 ftpServer:   Ftp server waiting for connection on port 21
23:23:25 WiFI_info:   RSSI is -62 dB
23:23:31 WiFI_info:   RSSI is -68 dB
23:23:31 Webpage:     index.html
23:23:33 WiFI_info:   RSSI is -65 dB
[ 48843][E][Preferences.cpp:483] getString(): nvs_get_str len fail: station_003 NOT_FOUND
23:23:39 action: ...  switch to station 3
23:23:39 AUDIO_info:  Connect to new host: "http://"
23:23:39 AUDIO_info:  Song stopped correctly after 0 msec
[ 49115][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for 
23:23:39 AUDIO_info:  Request http:// failed!
23:23:39 StreamTitle:
23:23:39 file "/logo/m/.jpg" not found
23:23:51 WiFI_info:   RSSI is -80 dB
23:23:52 WiFI_info:   RSSI is -69 dB
[ 63465][E][Preferences.cpp:483] getString(): nvs_get_str len fail: station_004 NOT_FOUND
23:23:53 action: ...  switch to station 4
23:23:53 AUDIO_info:  Connect to new host: "http://"
23:23:53 AUDIO_info:  Song stopped correctly after 0 msec
[ 63738][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for 
23:23:53 AUDIO_info:  Request http:// failed!
23:23:53 StreamTitle:
23:23:54 file "/logo/m/.jpg" not found
23:23:54 WiFI_info:   RSSI is -59 dB
23:23:57 WiFI_info:   RSSI is -76 dB
23:23:59 WiFI_info:   RSSI is -63 dB

schreibfaul1, 2023-06-18:
hello valera39,
i added the additional boards in the platform.ini without knowing if they work at all. On June 22nd I will probably get an N16R8 board and can then test the settings and adjust them if necessary.

schreibfaul1, 2023-06-18:
I will also have to adjust the miniwebradio.csv, with the one I'm using now the nvs area is in the wrong place.

valera39, 2023-06-18:
Okay, I'll look forward to receiving your boards.

closed 2023-07-04 by valera39