schreibfaul1/ESP32-MiniWebRadio #197: VS1053 flac and m4a playback not working

steve6375, 2023-06-18:
Should these work without 'plugin' on esp32   no PSRAM + VS1053?

What is the 'plugin' for FLAC and how do we add it?
Should m4a work without any plugin?

steve6375, 2023-06-18:
OK , I  found line 332 in vs1053_ext.cpp needs uncommenting to play FLAC and can now play FLAC file, but m4a still not working?

11:35:46 AUDIO_info:  Song stopped correctly after 0 msec
11:35:46 AUDIO_info:  Reading file: "/audiofiles/sample.m4a"
11:35:46 AUDIO_info:  stream ready
11:35:46 Audiocodec:  M4A
11:35:47 AUDIO_info:  Song stopped correctly after 0 msec
11:35:47 AUDIO_info:  End of file "sample.m4a"
11:35:47 end of file: sample.m4a

steve6375, 2023-06-18:
Very strange!
The sample m4a file you provide would not play.
So I downloaded Symphony 6 and put it in SSD using ftp
https://filesamples.com/formats/m4a
I clicked on the file and it plays!
then I clicked on sample.m4a and now that plays too!
So I reset esp32 - and same thing - sample.m4a will not play until I play symphony6 m4a file.

steve6375, 2023-06-18:
none of the sample1.m4a to sample4.m4a files on https://filesamples.com/formats/m4a work
until symphony 6 m4a is played.

steve6375, 2023-06-18:
hmmm. After power-off and power-on, now I can't play any m4a file!

schreibfaul1, 2023-06-18:
I didn't activate the plugin by default because it caused a crash on one of my VS1053 boards.
m4a is not reliable, depends on the internal structure of the files, mostly m4a works on the 2nd try.

steve6375, 2023-06-19:
FLAC, wav, aac does not work until you **first** play mp3 file.

1. POWER OFF - POWER ON
2. Play FLAC file --> no work
3. Play sample.mp3 - works
4. Play FLAC file --> now plays

Playing aac or wav does not make FLAC play, only mp3

Also
1. POWER OFF - POWER ON
2. Play sample.wav --> works for about 1/2 second then stops (also sample.aac)
3. Play sample.mp3 - works
4. Play sample.wav file or aac file --> now plays

So it seems that an mp3 file must be played first? 

Still cannot get any m4a to play.

closed 2024-02-10 by github-actions[bot]