schreibfaul1/ESP32-MiniWebRadio #199: Assembly error

valera39, 2023-06-21:
Hello Wolle!
PlatformIO gives an error when building with parameters in platformio.ini:
;board_build.partitions = miniwebradio.csv      ;  4MB Flash
        ;board_build.partitions = miniwebradio8MB.csv    ;  8MB Flash
        board_build.partitions = miniwebradio16MB.csv  ; 16MB Flash

        ;board = esp32-12k                      ; quad  4MB Flash, 8MB quad PSRAM
        ;board = ESP32-S3-DevKitC-1-N8R2        ; quad  8MB Flash, 2MB quad PSRAM
        ;board = ESP32-S3-DevKitC-1-N8R8        ; quad  8MB Flash, 8MB octo PSRAM  *1)
        board = ESP32-S3-DevKitC-1-N16R8       ; octo 16MB Flash, 8MB octo PSRAM  *1)
        ;board = ESP32-S3-DevKitC-1-N16R8b      ; quad 16MB Flash, 8MB octo PSRAM  *1)
        
Building .pio\build\esp32s3\bootloader.bin
usage: esptool elf2image [-h] [--output OUTPUT] [--version {1,2,3}]
                         [--min-rev-full {0, ... 65535}]
                         [--max-rev-full {0, ... 65535}] [--secure-pad]
                         [--secure-pad-v2]
                         [--elf-sha256-offset ELF_SHA256_OFFSET]
                         [--dont-append-digest] [--use_segments]
                         [--flash-mmu-page-size {64KB,32KB,16KB,8KB}]
                         [--pad-to-size PAD_TO_SIZE]
                         [--flash_freq {80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                         [--flash_mode {qio,qout,dio,dout}]
                         [--flash_size {256KB,512KB,1MB,2MB,2MB-c1,4MB,4MB-c1,8MB,16MB,32MB,64MB,128MB}]
                         [--spi-connection SPI_CONNECTION]
                         input
esptool elf2image: error: argument --flash_mode/-fm: invalid choice: 'opi' (choose from 'qio', 'qout', 'dio', 'dout')
*** [.pio\build\esp32s3\bootloader.bin] Error 2

schreibfaul1, 2023-06-21:
Too bad, could have worked. I copied the board definition from a 'branch' of PlatformIO Git and changed 'memory_type' and 'flash_mode'. You can now only try it yourself. Maybe 'flash_mode' has to be qio?
[image]
That's just a guess.

valera39, 2023-06-21:
Should this be changed in ESP32-S3-DevKitC-1-N16R8.json ?
That's it. "flash_mode": "qio" ,change?

schreibfaul1, 2023-06-21:
Yes, in the json file

valera39, 2023-06-21:
Wait. It's not like I didn't read it.*1)   in menuconfig the type of PSRAM must be changed from QUAD to OCTAL!
;       component config ---> ESP32S3-Specific ---> SPI RAM config ---> MODE (QUAD/OCT)... ---> (x) Octal Mode PSRAM.
I didn't do that. I'll try to see if it's my inattention.

valera39, 2023-06-21:
It worked on N8R8! I should read more carefully. But now another problem.
[image]
ESP32 Chip: ESP32-S3
Arduino Version: 2.0.9
ESP-IDF Version: 4.4.4
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
PSRAM total size: 8388608 bytes

00:00:00
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V2     *    
00:00:00        ***************************    
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9488
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found
E (248) ledc: ledc_get_duty(739): LEDC is not initialized
00:00:01 setup: ....  seek for stations.csv
00:00:01 setup: ....  stations.csv found
00:00:01 setup: ....  seek for WiFi networks
[  2057][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: ASUS_AC66U
[  2068][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: ASUS_AC66U
[  2071][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_01
[  2078][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_02
[  2086][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_03
[  2093][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_04
[  2101][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_04
[  8168][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[  8168][I][WiFiMulti.cpp:119] run(): [WIFI] 4 networks found
[  8169][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 2C:FD:A1:C2:44:F0 SSID: ASUS_AC66U Channel: 12 (-83)
[  8256][I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
00:00:07 WiFI_info:   Connecting WiFi...
00:00:07 setup: ....  connected to ASUS_AC66U, IP address is 192.168.1.130
00:00:07 ftpServer:   Buffers allocated: 4622 bytes
00:00:07 rtime_info:  Initializing SNTP
00:00:07 rtime_info:  Waiting for system time to be set... (1/10)
00:00:09 rtime_info:  Waiting for system time to be set... (2/10)
21:41:35 rtime_info:  21:41:35
21:41:35 audiotask:   is pinned to core 1
21:41:35 audiotask:   priority is 2
21:41:36 VS1053 chipID = -1350368918, version = 3
21:41:36 setup: ....  Number of saved stations: 89
21:41:36 setup: ....  current station number: 1
21:41:36 setup: ....  current volume: 3
21:41:36 setup: ....  last connected host: https://listen4.myradio24.com/alexx777ru
21:41:36 setup: ....  connection timeout: 250 ms
21:41:36 setup: ....  connection timeout SSL: 2000 ms
21:41:36 action: ...  current volume is 3
[ 13430][E][Preferences.cpp:483] getString(): nvs_get_str len fail: station_001 NOT_FOUND
21:41:37 action: ...  switch to station 1
21:41:37 AUDIO_info:  Connect to new host: "http://"
21:41:37 AUDIO_info:  Song stopped correctly after 0 msec
21:41:37 AUDIO_info:  PSRAM found, inputBufferSize: 295903 bytes
[ 13691][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for 
21:41:37 AUDIO_info:  Request http:// failed!
21:41:37 StreamTitle:
21:41:37 WiFI_info:   RSSI is -71 dB
21:41:37 DLNA_info:   SSDP search for media servers started
21:41:41 DLNA_info:   SSDP query discovered 1 media servers
21:41:41 DLNA_info:   connected successfully to 192.168.1.4:6000
21:41:41 ftpServer:   Ftp server waiting for connection on port 21
21:41:46 WiFI_info:   RSSI is -76 dB
21:41:55 WiFI_info:   RSSI is -73 dB
21:42:01 WiFI_info:   RSSI is -69 dB
21:42:05 WiFI_info:   RSSI is -77 dB
21:42:09 WiFI_info:   RSSI is -80 dB
21:42:10 WiFI_info:   RSSI is -75 dB
21:42:11 WiFI_info:   RSSI is -79 dB
21:42:12 WiFI_info:   RSSI is -75 dB
21:42:20 WiFI_info:   RSSI is -73 dB
21:42:23 WiFI_info:   RSSI is -83 dB
21:42:24 WiFI_info:   RSSI is -68 dB
21:43:17 WiFI_info:   RSSI is -72 dB
21:43:28 WiFI_info:   RSSI is -65 dB
21:43:46 WiFI_info:   RSSI is -76 dB
21:43:52 WiFI_info:   RSSI is -79 dB
21:43:53 WiFI_info:   RSSI is -62 dB
21:43:54 WiFI_info:   RSSI is -67 dB
21:43:59 WiFI_info:   RSSI is -71 dB
21:44:01 WiFI_info:   RSSI is -58 dB
21:44:07 WiFI_info:   RSSI is -69 dB
21:44:08 WiFI_info:   RSSI is -61 dB
21:44:09 WiFI_info:   RSSI is -67 dB
21:44:52 WiFI_info:   RSSI is -78 dB
21:44:53 WiFI_info:   RSSI is -72 dB
21:44:55 WiFI_info:   RSSI is -80 dB
21:44:56 WiFI_info:   RSSI is -74 dB
21:45:02 WiFI_info:   RSSI is -62 dB
21:45:06 WiFI_info:   RSSI is -72 dB
21:45:15 WiFI_info:   RSSI is -64 dB
21:46:15 WiFI_info:   RSSI is -76 dB
21:46:19 WiFI_info:   RSSI is -78 dB
21:46:21 WiFI_info:   RSSI is -73 dB
21:46:22 WiFI_info:   RSSI is -77 dB
21:46:24 WiFI_info:   RSSI is -73 dB
21:46:27 WiFI_info:   RSSI is -77 dB
21:46:28 WiFI_info:   RSSI is -71 dB
[305853][E][Preferences.cpp:483] getString(): nvs_get_str len fail: station_002 NOT_FOUND
21:46:29 action: ...  switch to station 2
21:46:29 AUDIO_info:  Connect to new host: "http://"
21:46:29 AUDIO_info:  Song stopped correctly after 0 msec
[306106][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for 
21:46:29 AUDIO_info:  Request http:// failed!
21:46:29 StreamTitle:
21:46:31 WiFI_info:   RSSI is -81 dB
[308861][E][Preferences.cpp:483] getString(): nvs_get_str len fail: station_003 NOT_FOUND
21:46:32 action: ...  switch to station 3
21:46:32 AUDIO_info:  Connect to new host: "http://"
21:46:32 AUDIO_info:  Song stopped correctly after 0 msec
[309114][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for 
21:46:32 AUDIO_info:  Request http:// failed!

schreibfaul1, 2023-06-21:
I'm glad to read that. I got the N8R8 with octal PSRAM today and it works as described. The 16MB flash isn't really used either, currently around 3.7MB.

You have overwritten the contents of the nvs with the new partition table. I should have told you that, the nvs partition is no longer in the same place. If you now always use the 8MB partition table, this won't happen again.
So now all channels are gone. But the settings.json on the SD card still contains values. You can delete the settings.json, it will be rewritten again.

valera39, 2023-06-21:
Thank you so much for the work you have done! VU metr is awesome! Thanks to your advice I was able to run 
my N16R8 and N8R8 boards. A few more questions:
1) on VS1053 when switching stations in the speakers for 1/2 second first a small squeak and then a click can be heard.
On the oscilloscope, which is connected to the output of VS1053, you can see a DC voltage spike at the moment of switching. This started around firmware 2.7.3f. Before that the switching was completely silent. I tried the changes suggested by steve6375 in #190, but they didn't help at all.
https://files.fm/u/9gekx6f4w
2) Very often the touchscreen does not initialize when booting up, it helps to briefly disconnect the TP_IRQ wire from gpio 39.
3) you can adjust the speed of the VU meter segments and can it be placed horizontally?
4) it is not always possible to save an updated station list in the web browser, most often it helps to restart the radio

[image]
Thank you!

steve6375, 2023-06-22:
when using web gui, press F12 in browser and click on Console. Then use controls in web gui. Do you see error messages in the console? The esp32 seems to miss the some commands from the browser.
I have made modifications to index.h so it does retries. This makes saving list, and other web controls much more reliable.
See #182 last post for a zip file.

I too get the small click when changing stations which I cannot seem to eliminate. I haven't tried it yet, but connecting the audio  L+R+Gnd signals using capacitors (e.g. 4.7uF) in series with each signal which may help. To eliminate squeak  noise I use this code in setStation function:

```
    SerialPrintfln("action: ...  switch to station " ANSI_ESC_CYAN "%d", sta);

    if(_f_isWebConnected && sta == _cur_station && _state == RADIO){ // Station is already selected
        _f_newStreamTitle = true;
    }
    else{
        int v = _cur_volume;
        while ( v > 0) audioSetVolume(v--);
        if (audioIsRunning()) audioStopSong();  // crashes if stop when no audio is running
        _streamTitle[0] = '\0';
        _icyDescription[0] = '\0';
        _f_newStreamTitle = true;
        _f_newIcyDescription = true;
        connecttohost(_stationURL);
        if(DECODER == 0) vTaskDelay(150);            // try different delays here
        while ( v < _cur_volume) { 
            vTaskDelay(10); 
            audioSetVolume(v++);
        }
    }
    _cur_station = sta;
    StationsItems();
    if(_state == RADIO || _state == RADIOico) showLogoAndStationName();
    showFooterStaNr();
}

```
For TP initialisation, try adding a capacitor from EN pin on esp32 to GND. What TFT are you using?

valera39, 2023-06-22:
ILI9488,I hadn't noticed the dumps before. And my previously assembled receivers work without problems, but with older firmware, and there are no clicks either.

closed 2023-06-22 by valera39