schreibfaul1/ESP32-MiniWebRadio #200: Questions about version 2.8

valera39, 2023-06-22:
Hello Wolle!
If I may, I have decided to post in a separate thread the questions that are relevant to me.
My configuration: ESP32S3 N16R8, VS1053, ILI9488.
1. When you switch the stations in the speakers for 1/2 second, first you hear a small squeak and then a click. 
On an oscilloscope connected to the VS1053 output you can see a DC voltage spike at the moment of switching. It started around firmware 2.7.3f. Before that the switching was completely silent. I tried the changes suggested by steve6375, but they didn't help.
2. The touch does not work after loading the radio, it can be turned on by briefly disconnecting the TP_IRQ of the display from gpio 39, or by briefly shorting it to ground. Changing the gpio did not help.
Again, on firmware 2.7 these problems were not at all.
3. is it possible to adjust the speed of the VU meter segments and can it be placed horizontally?
4. With PlatformIO Serial Monitor enabled the log became less informative, like this:
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x39 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
[   934][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
E (934) esp_core_dump_flash: No core dump partition found!
E (934) esp_core_dump_flash: No core dump partition found!
E (248) ledc: ledc_get_duty(739): LEDC is not initialized
[  2063][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: ASUS_AC66U
[  2070][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: ASUS_AC66U
[  2071][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_01
[  2075][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_02
[  2083][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_03
[  2091][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_04
[  2099][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_04
[  8562][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[  8563][I][WiFiMulti.cpp:119] run(): [WIFI] 6 networks found
[  8564][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 2C:FD:A1:C2:44:F0 SSID: ASUS_AC66U Channel: 1 (-59)
[  8638][I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done
is there any bugs in it and how to enable a more detailed output like before?
5. how to enable a more detailed output as before?
6. I already wrote about the great convenience of having a drop-down list of stations on the display. If you think this is a necessary option, do you have a solution to this problem?
Unfortunately, I'm not a programmer, but I'm always willing to help you make your radio even better.
Thank you!

Arne65, 2023-06-23:
Hello vallera

To your 3rd point with the ESP 32 with VS 1053, I also only have the yellow bars. On the ESP 32-S3 1N8 with an external DAC, I set AUDIOTASK_CORE to 0 because the VU display works perfectly and all other displays are considerably faster. With the ESP 32, however, this did not work for me

valera39, 2023-06-24:
What about me with my questions, is there a solution?

schreibfaul1, 2023-06-24:
1) this is probably a bug in VS1053? presumably these are disturbances before the codec and the sample rate are known. I suppressed the first 200ms, it's now okay with my board.
2) I know, it's my turn
3) the values ​​are read from the VS1053 from a register and displayed. With the I2S Dac I calculate this myself.
4) these are entries from the stations.csv. This is just to show what SSID it might be able to connect to. If you only have one WiFi network, you can delete the entries 'network01'... or hide them with #
5) is done
6) that's something for the future

clear-sky, 2023-06-25:
Hi dear wolle! after upgrading to 2.8 some stations stopped working
please test this streams:
http://us3.internet-radio.com:8342/stream
http://199.189.111.28:8012/
http://radio.stereoanime.net:8000/128

closed 2023-06-26 by valera39