schreibfaul1/ESP32-MiniWebRadio #202: Insufficient Memory and SD card write errors when set DECODER to 1 (for PCM5102A DAC)

steve6375, 2023-06-24:
I have both a VS1053B and PCM5102A DAC connected to an esp32. The only problem the VS1053 has is that m4a files don't seem to play and sometimes flac files dont fully play. SD card works OK and esp32 doesn't crash.

I used 'Release' build (not debug) and -DCORE_DEBUG_LEVEL=5 

So I tried the DAC and just changed DECODER to 1 and recompiled. With DAC it does play m4a (not FLAC files as no PSRAM) and some radio stations but on other stations it will give memory errors and also reads and writes to SD card often fail and leaves blank stations.csv or settings.csv.
Also crashes on occasion even from fresh boot. I tried several clean rebuilds but still a problem.
I am using the small buttons previous build still (not latest main) release.

Is it using too much memory and is there anything I can do to make DAC work reliably?

**This is DECODER = 0 which is basically stable and OK:**
```
 *  Executing task: C:\Users\Steve\.platformio\penv\Scripts\platformio.exe run --target upload --target monitor --environment esp32 

Processing esp32 (board: esp32dev; platform: https://github.com/platformio/platform-espressif32.git#v6.2.0; framework: arduino, espidf)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.2.0+sha.1d6e199) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 3MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 2.0.8+sha.d3254f7
 - framework-espidf @ 3.40404.0 (4.4.4)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - tool-ninja @ 1.9.0 
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Warning! Arduino framework as an ESP-IDF component doesn't handle the `variant` field! The default `esp32` variant will be used.
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 12 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP32-IR-Remote-Control @ 1.0.10+sha.869e149
|-- ESP32-vs1053_ext @ 1.0.10+sha.297b304
|-- ESP32-audioI2S-master @ 2.0.0+sha.669e61d
|-- Arduino_JSON @ 0.2.0+sha.89b92d9
|-- ac101
|-- es8388
|-- ftpsrv
|-- rtime
|-- SoapESP32 @ 1.1.5
|-- tftLib
|-- websrv
|-- wm8978
Building in release mode
Compiling .pio\build\esp32\src\audiotask.o
Compiling .pio\build\esp32\src\main.o
Linking .pio\build\esp32\firmware.elf
Retrieving maximum program size .pio\build\esp32\firmware.elf
Checking size .pio\build\esp32\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  17.3% (used 56744 bytes from 327680 bytes)
Flash: [========  ]  79.1% (used 3006009 bytes from 3801088 bytes)
Building .pio\build\esp32\firmware.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 25 ELF sections
Successfully created esp32 image.
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
[... 241 lines cut ...]
  #5  0x40196b5d:0x3ffd1d10 in __cxa_allocate_exception at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_alloc.cc:300
  #6  0x40196c1c:0x3ffd1d30 in operator new(unsigned int) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/new_op.cc:54
  #7  0x40120752:0x3ffd1d50 in __gnu_cxx::new_allocator<std::_Sp_counted_ptr_inplace<VFSFileImpl, std::allocator<VFSFileImpl>, (__gnu_cxx::_Lock_policy)2> >::allocate(unsigned int, void const*) at c:\users\steve\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\ext/new_allocator.h:111
      (inlined by) std::allocator_traits<std::allocator<std::_Sp_counted_ptr_inplace<VFSFileImpl, std::allocator<VFSFileImpl>, (__gnu_cxx::_Lock_policy)2> > >::allocate(std::allocator<std::_Sp_counted_ptr_inplace<VFSFileImpl, std::allocator<VFSFileImpl>, (__gnu_cxx::_Lock_policy)2> >&, unsigned int) at c:\users\steve\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/alloc_traits.h:436
      (inlined by) std::__allocated_ptr<std::allocator<std::_Sp_counted_ptr_inplace<VFSFileImpl, std::allocator<VFSFileImpl>, (__gnu_cxx::_Lock_policy)2> > > std::__allocate_guarded<std::allocator<std::_Sp_counted_ptr_inplace<VFSFileImpl, std::allocator<VFSFileImpl>, (__gnu_cxx::_Lock_policy)2> > >(std::allocator<std::_Sp_counted_ptr_inplace<VFSFileImpl, std::allocator<VFSFileImpl>, (__gnu_cxx::_Lock_policy)2> >&) at c:\users\steve\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/allocated_ptr.h:97
      (inlined by) std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<VFSFileImpl, std::allocator<VFSFileImpl>, VFSImpl*, char const*&, char const*&>(VFSFileImpl*&, std::_Sp_alloc_shared_tag<std::allocator<VFSFileImpl> >, VFSImpl*&&, char const*&, char const*&) at c:\users\steve\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/shared_ptr_base.h:675
      (inlined by) std::__shared_ptr<VFSFileImpl, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<VFSFileImpl>, VFSImpl*, char const*&, char const*&>(std::_Sp_alloc_shared_tag<std::allocator<VFSFileImpl> >, VFSImpl*&&, char const*&, char const*&) at c:\users\steve\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/shared_ptr_base.h:1342
      (inlined by) std::shared_ptr<VFSFileImpl>::shared_ptr<std::allocator<VFSFileImpl>, VFSImpl*, char const*&, char const*&>(std::_Sp_alloc_shared_tag<std::allocator<VFSFileImpl> >, VFSImpl*&&, 
char const*&, char const*&) at c:\users\steve\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/shared_ptr.h:359
      (inlined by) std::shared_ptr<VFSFileImpl> std::allocate_shared<VFSFileImpl, std::allocator<VFSFileImpl>, VFSImpl*, char const*&, char const*&>(std::allocator<VFSFileImpl> const&, VFSImpl*&&, char const*&, char const*&) at c:\users\steve\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/shared_ptr.h:706
      (inlined by) std::shared_ptr<VFSFileImpl> std::make_shared<VFSFileImpl, VFSImpl*, char const*&, char const*&>(VFSImpl*&&, char const*&, char const*&) at c:\users\steve\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/shared_ptr.h:722
  #8  0x401208f9:0x3ffd1d80 in VFSImpl::open(char const*, char const*, bool) at C:/Users/Steve/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/libraries/FS/src/vfs_api.cpp:47
  #9  0x40117341:0x3ffd1e10 in fs::FS::open(char const*, char const*, bool) at C:/Users/Steve/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/libraries/FS/src/FS.cpp:234
  #10 0x400d9535:0x3ffd1e40 in updateSettings() at src/main.cpp:417
  #11 0x400dc7af:0x3ffd1ea0 in check_1sec() at src/main.cpp:2243
  #12 0x400dc8aa:0x3ffd1f00 in loop() at src/main.cpp:2269 (discriminator 1)
  #13 0x40116075:0x3ffd1f20 in loopTask(void*) at C:/Users/Steve/.platformio/packages/framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa/cores/esp32/main.cpp:50

ELF file SHA256: b3d397abfc4fb8c7

E (12025) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

schreibfaul1, 2023-06-25:
Finding a trade-off between storage usage and features is difficult. Without PSRAM with I2S DAC only mp3 and aac will work properly.
1) The InBuffer, which temporarily stores the raw audio data, is about 16000 bytes large without PSRAM and about 640000 bytes with PSRAM. So FLAC with a block size of 8KB, only 2 frames fit into the buffer, which is very little.
2) Some WLAN functions are swapped out to the PSRAM, if available
3) mbetTLS (https://...) needs more than 20KB and should therefore be written to PSRAM.
4) VS1053 is a HW decoder, the SW decoders need additional memory, PSRAM is used where possible
5) I reduced the stack in the audio task by 2000 DWORDs (8000bytes), that might help. If you click on the question mark, the 'highest water level' is displayed in the log. There should still be some space at the top, if you reach zero there is a risk of a crash.
If you want to use all functions, PSRAM is advisable. Especially OPUS, VORBIS and FLAC are very memory hungry.

steve6375, 2023-06-26:
Thanks for reply. Changing CONFIG_MBEDTLS_INTERNAL_MEM_ALLOC didn't help.
The memory stats when using the test ? button in the web ui station screen dont show much difference (but playing some radio stations cause buffer errors as soon as they are played on the DAC still).
I guess an esp32 with PSRAM is needed. 
Do you have a recommendation for a suitable esp32, audio board and TFT that will work the best?
Is any model of esp32+PSRAM OK?

schreibfaul1, 2023-06-26:
An ESP32 with 4MB PSRAM is sufficient. PSRAM is faster with ESP32-S3, even faster with octal PSRAM. AAC with SBR is then also activated for the S3 (only ext. DAC). An external DAC would also be good, like PCM5102 instead of the old VS1053B

steve6375, 2023-07-01:
FreeNove WROVER + PSRAM + camera board with PCM5102 solves the problem. 

I get warnings 

08:21:06 AUDIO_info:  slow stream, dropouts are possible
08:21:07 AUDIO_info:  slow stream, dropouts are possible
08:21:08 AUDIO_info:  slow stream, dropouts are possible

but seems OK.

closed 2023-07-01 by steve6375