schreibfaul1/ESP32-MiniWebRadio #204: loading error

valera39, 2023-06-24:
Hello Wolle!
After the last update there is a boot error on ESP32 N8R8
Log:
ESP32 Chip: ESP32-S3
Arduino Version: 2.0.9
ESP-IDF Version: 4.4.4
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
PSRAM total size: 8388608 bytes
00:00:00
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V2     *    
00:00:00        ***************************    
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9488
00:00:00 setup: ....  Init SD card
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403774c9
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
[   935][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
E (935) esp_core_dump_flash: No core dump partition found!
E (936) esp_core_dump_flash: No core dump partition found!

Back to the old firmware

schreibfaul1, 2023-06-24:
If you mean the core dump partition, it doesn't exist.

valera39, 2023-06-24:
All good now, thank you!

closed 2023-06-25 by valera39