schreibfaul1/ESP32-MiniWebRadio #205: some stations stopped in new version

clear-sky, 2023-06-25:


valera39, 2023-06-25:
[clear-sky](https://github.com/clear-sky). hello!
I have on the latest firmware works the first and second stations. The third does not play.
Configuration ESP32S3 N8R8 and VS1053.

clear-sky, 2023-06-26:
Hi dear valera!
thanks for the reply -my config is not similar yours!
i use external dac(esp-s3-12k + pcm5102)
i updated "ESP32-audioI2S-master" to latest and "http://199.189.111.28:8012/" started working
but this stations is does not work yet!
http://us3.internet-radio.com:8342/stream
http://radio.stereoanime.net:8000/128

schreibfaul1, 2023-06-26:
Tested the station http://radio.stereoanime.net:8000/128 and get a connection timeout. The connection is very bad. I found that I can't increase the timeout in common.h, it was always 250ms. This works now with CONN_TIMEOUT=500ms, but  drop outs are still. The other two stations are okay.

closed 2023-06-26 by clear-sky