schreibfaul1/ESP32-MiniWebRadio #206: endless loop esp32-S3-16R8

Arne65, 2023-06-26:
Hello Wolle !

do you have any idea what could be causing this.

endless loop.

I've tried everything and the result is always the same.

Thanks

00:04:30   

00:04:30        ***************************   

00:04:30        *     MiniWebRadio V2     *   

00:04:30        ***************************   

00:04:30   

00:04:30 setup: ....  Arduino is pinned to core 1

00:04:30 tft_info: .  init ILI9486

00:04:30 setup: ....  Init SD card

ESP-ROM:esp32s3-20210327

Build:Mar 27 2021

rst:0x8 (TG1WDT_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)

Saved PC:0x4037767d

SPIWP:0xee

mode:DIO, clock div:1

load:0x3fce3808,len:0x44c

load:0x403c9700,len:0xbe4

load:0x403cc700,len:0x2a38

entry 0x403c98d4

[   975][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled

E (986) esp_core_dump_flash: Core dump data check failed:

Calculated checksum='0ef9d667'

 

ESP32 Chip: ESP32-S3

Arduino Version: 2.0.9

ESP-IDF Version: 4.4.4

ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)

PSRAM total size: 8388608 bytes

 

 

00:04:32   

00:04:32        ***************************   

00:04:32        *     MiniWebRadio V2     *   

00:04:32        ***************************   

00:04:32   

00:04:32 setup: ....  Arduino is pinned to core 1

00:04:32 tft_info: .  init ILI9486

00:04:33 setup: ....  Init SD card

ESP-ROM:esp32s3-20210327

Build:Mar 27 2021

rst:0x1 (POWERON),boot:0xa (SPI_FAST_FLASH_BOOT)

SPIWP:0xee

mode:DIO, clock div:1

load:0x3fce3808,len:0x44c

load:0x403c9700,len:0xbe4

load:0x403cc700,len:0x2a38

entry 0x403c98d4

[   973][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled

E (983) esp_core_dump_flash: Core dump data check failed:

Calculated checksum='0ef9d667'

 

ESP32 Chip: ESP32-S3

Arduino Version: 2.0.9

ESP-IDF Version: 4.4.4

ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)

PSRAM total size: 8388608 bytes

 

 

00:00:00   

00:00:00        ***************************   

00:00:00        *     MiniWebRadio V2     *   

00:00:00        ***************************   

00:00:00   

00:00:00 setup: ....  Arduino is pinned to core 1

00:00:00 tft_info: .  init ILI9486

00:00:00 setup: ....  Init SD card

ESP-ROM:esp32s3-20210327

Build:Mar 27 2021

rst:0x8 (TG1WDT_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)

Saved PC:0x4037767d

SPIWP:0xee

mode:DIO, clock div:1

load:0x3fce3808,len:0x44c

load:0x403c9700,len:0xbe4

load:0x403cc700,len:0x2a38

entry 0x403c98d4

[   975][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled

E (986) esp_core_dump_flash: Core dump data check failed:

Calculated checksum='0ef9d667'

 

ESP32 Chip: ESP32-S3

Arduino Version: 2.0.9

ESP-IDF Version: 4.4.4

ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)

PSRAM total size: 8388608 bytes

schreibfaul1, 2023-06-26:
Hello Arne,
the 'coredump partition' is available, but is not necessary for normal operation. In the partition table, this is always at the end of the flash area. I suspect you selected a larger flash than actually exists. Please set a smaller size in platformio.ini for your environment. This is just flash and has nothing to do with the PSRAM.

[image]

schreibfaul1, 2023-06-26:
Correction: Forget what I wrote, somehow the partition table can't cope with the 16MB limit. I reduced the factory partition a bit, now it works.
But I'm not sure if that's the reason for the continuous loop.

Arne65, 2023-06-26:
Hello Wolle

Thank you for the quick reply and for the great work you are doing here.
I have set the following.

platformio.ini

[image]

miniwebradio16MB.csv

# Name,     Type,   SubType,   Offset,   Size,     Flags
  phy_init, data,   phy,       0x9000,   0x7000,
  nvs,      data,   nvs,       0x10000,  0x40000,
  factory,  app,    factory,   0x50000,  0xF90000,
  coredump, data,   coredump,  0xFE0000, 0x10000

the loop now looks like this

--- Terminal on COM7 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
�e3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
[   971][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377699
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
[   973][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377699
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
[   973][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled

schreibfaul1, 2023-06-26:
The watchdog timer 1 had triggered.
[image]
Could be a problem with the wiring of the pins. If the ESP32S3 Boad is not soldered, unplug it from the circuit to see if the problem goes away.

Arne65, 2023-06-27:
Good morning Wolle,

yes you were right

I soldered the ESP pins again and it works.

closed 2023-06-27 by Arne65