schreibfaul1/ESP32-MiniWebRadio #208: Radio Stations not playing and 'nvs_get_str len fail' error (esp32 + PSRAM)

steve6375, 2023-06-29:
I had MiniWebRadio working on an esp32 without PSRAM.
I programmed a brand new esp32 WROVER which has PSRAM (actually a FREENOVE esp32 WROVER-E with 4MB PSRAM - but with no camera attached).
The SD card loads OK. 
The Web GUI can be displayed and accessed.
Media files play OK.
Stations are listed in the web GUI OK.
TFT display is OK.
BUT - no station name is displayed on the TFT and no radio station plays.
Also the web GUI URL fields do not update.

The code is the same (I only changed to NVRAM enabled in PlatformIO.ini build flags.
I am using the previous small_buttons branch and same code works on WROOM esp32 with no PSRAM.

[image]

Here is the serial monitor log

```
0:26:30 rtime_info:  20:26:30
20:26:30 audiotask:   is pinned to core 1
20:26:30 audiotask:   priority is 2      
20:26:31 VS1053 chipID = -1372388535, version = 3 
20:26:31 setup: ....  Number of saved stations: 28
20:26:31 setup: ....  current station number: 1
20:26:31 setup: ....  current volume: 18
20:26:31 setup: ....  last connected host: https://stream.live.vc.bbcmedia.co.uk/bbc_world_service
20:26:31 setup: ....  connection timeout: 500 ms
20:26:31 setup: ....  connection timeout SSL: 2000 ms
[ 12410][E][Preferences.cpp:483] getString(): nvs_get_str len fail: station_001 NOT_FOUND
20:26:31 action: ...  switch to station 1
20:26:31 AUDIO_info:  Connect to new host: "http://"
20:26:31 AUDIO_info:  Song stopped correctly after 0 msec
20:26:31 AUDIO_info:  PSRAM found, inputBufferSize: 295903 bytes
[ 12468][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for 
20:26:31 AUDIO_info:  Request http:// failed!
20:26:31 StreamTitle:
20:26:31 file "/logo/s/.jpg" not found
20:26:31 DLNA_info:   SSDP search for media servers started
20:26:35 DLNA_info:   SSDP query discovered 1 media servers
20:26:35 DLNA_info:   connected successfully to 192.168.1.1:8200
20:26:35 ftpServer:   Ftp server waiting for connection on port 21
[ 23117][E][Preferences.cpp:483] getString(): nvs_get_str len fail: station_002 NOT_FOUND
20:26:41 action: ...  switch to station 2
20:26:41 AUDIO_info:  Connect to new host: "http://"
20:26:41 AUDIO_info:  Song stopped correctly after 0 msec
[ 23155][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for 
20:26:41 AUDIO_info:  Request http:// failed!
20:26:41 StreamTitle:
20:26:42 file "/logo/s/.jpg" not found
```

It seems the issue may be around
getString(): nvs_get_str len fail: station_001 NOT_FOUND

The URL does not seem to be obtained. It just has the http:// start but not the rest.

any clues where I might look for the error? 
The board is set as esp32dev - should I set esp-wrover-kit?

steve6375, 2023-06-29:
[image]
Stations.csv is OK
If I load a URL into web GUI filed then it plays OK
[image]

It seems the storage to nvs is failing?

schreibfaul1, 2023-06-29:
I added partition tables for 8 and 16MB flash. In doing so, I put the nvs partition from the end of the table in an area that is in the same place for everyone. By doing so you probably overwrote the old nvs area. Please delete the settings.json so that nvs is rewritten.

steve6375, 2023-06-29:
That fixed it! Thanks :-)

closed 2023-06-29 by steve6375

steve6375, 2023-06-30:
P.S. I cloned the latest version and got the same issue. Could you add a way to auto-detect this problem and automatically delete and remake the settings.csv? It is not obvious that the settings.csv file needs to be deleted.

schreibfaul1, 2023-06-30:
That's strange. The nvs partition is not deleted during the update, which is also not intended because it will destroy your own channel lists.
I don't have a solution for this at the moment.
The only thing I can think of to destroy the nvs partition would be 'Erase Flash'