schreibfaul1/ESP32-MiniWebRadio #209: Touchscreen initialization

valera39, 2023-06-29:
Hello Wolle!
I replaced VS1053 with PCM5102 and the noise and clicking problem is completely solved. The VU meter also started working normally. So the problem was in the VS1053 settings?
But it still does not work with the touchscreen, only if you jerk the gpio, I wrote earlier about this.
Thanks, looking forward to the corrections!

valera39, 2023-06-30:
Turning the time announcement on and off works. But the touchscreen does not start after rebooting or powering on until T_IRQ is shorted to ground or disconnected from ESP.
ESP32S3 N8R8, ILI9488, PCM5102. No errors in the log

Arne65, 2023-06-30:
Hallo Valera 

I had a similar problem and swapped the GPIO TP IRQ and TP_CS in the code that helped me

valera39, 2023-06-30:
Hello [Arne65](https://github.com/Arne65)
change in common.h?

Arne65, 2023-06-30:
Yes

valera39, 2023-06-30:
Ok

schreibfaul1, 2023-07-01:
Great that you were able to solve the problem by changing the GPIOs. I rarely had a TP blockade, but then no SW or HW reset helped. Only interrupting the operating voltage.
The TP controller has no reset pin nor any registers that could be reset.
I have now added a fake measurement for x and y to the setup, maybe that will help.

closed 2023-07-06 by valera39