schreibfaul1/ESP32-MiniWebRadio #212: Station Logos not being fetched reliably when using web GUI station finder

steve6375, 2023-06-30:
Latest (todays) version. When using the station finder, the icon file is not being fetched and displayed.
There are timeout errors.
If I click the arrow button the logo is displayed in the browser - so the URL is OK but the MiniWebRadio html index.h seems to be having timeout issues.
I dont recall this being an issue with the previous version??

[image]

schreibfaul1, 2023-06-30:
Hello Steve,
Thanks for the feedback. Images cannot be directly embedded from one web page into another, CORS policy prevents this. So I used a proxy that removes the CORS flag. The old proxy doesn't seem to be reachable anymore, I chose a different one.

steve6375, 2023-06-30:
Thats fixed it! Thanks!

closed 2023-06-30 by steve6375

steve6375, 2023-07-01:
Still some icons blocked though!
I found at least 5 other cases.

[image]

schreibfaul1, 2023-07-01:
Basically, the cors policy is implemented by the web browser. That's what we want to avoid. In addition, the web host can explicitly forbid cors. The old cors proxy seems to be back online. If you have a good proxy, we'll install it.