schreibfaul1/ESP32-MiniWebRadio #214: Crash when playing Arabella Rock (PCM5102)

steve6375, 2023-07-01:
Sometimes when changing to Arabella Rock station, using PCM5102 DAC + esp32 WROVER+PSRAM, the esp32 crashes and reboots.
Here is log playing **ASDA** and then switching to **Arabella Rock**,
This is reproducible but does not always happen but at moment happens every time when ASDA -> Arabella.
May depend on what is currently being played?

```
09:15:36 AUDIO_info:  icy-name: ASDA
09:15:36 StationName: ASDA
09:15:36 AUDIO_info:  MP3Decoder has been initialized, free Heap: 88516 bytes , free stack 32 DWORDs
09:15:36 lastURL: ..  https://dmmatrix01test.moodmedia.nl:8004/ASDA
09:15:36 Audiocodec:  MP3
09:15:36 AUDIO_info:  stream ready
09:15:36 AUDIO_info:  syncword found at pos 29
09:15:36 AUDIO_info:  syncword found at pos 0
09:15:36 AUDIO_info:  Channels: 2
09:15:36 AUDIO_info:  SampleRate: 44100
09:15:36 AUDIO_info:  BitsPerSample: 16
09:15:36 AUDIO_info:  BitRate: 192000
09:15:36 StreamTitle: 
09:15:36 AUDIO_info:  VBR recognized, audioFileDuration is estimated
09:26:20 action: ...  switch to station 26
09:26:20 AUDIO_info:  Connect to new host: "https://static.arabella.at/streamadressen/arabella-rock.m3u"
09:26:20 AUDIO_info:  buffers freed, free Heap: 111904 bytes
09:26:23 AUDIO_info:  SSL has been established in 2963 ms, free Heap: 111676 bytes
09:26:23 AUDIO_info:  Connect to new host: "https://frontend.streams.arabella.at/arabella-rock/stream/mp3?aggregator=arabella-playlistfile"
09:26:23 AUDIO_info:  buffers freed, free Heap: 111124 bytes
09:26:24 AUDIO_info:  SSL has been established in 1128 ms, free Heap: 111716 bytes
09:26:25 AUDIO_info:  redirect to new host "https://edge03.streams.arabella.at/arabella-rock/stream/mp3?aggregator=arabella-playlistfile"
09:26:25 AUDIO_info:  Connect to new host: "https://edge03.streams.arabella.at/arabella-rock/stream/mp3?aggregator=arabella-playlistfile"
09:26:25 AUDIO_info:  buffers freed, free Heap: 111944 bytes
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (audioplay) 
Core  1 register dump:
PC      : 0x400916d1  PS      : 0x00060036  A0      : 0x4008f9e3  A1      : 0x3ffddd60
A2      : 0x3ffba2b4  A3      : 0x3ffba480  A4      : 0x3ffba484  A5      : 0x00000001
A6      : 0x00000001  A7      : 0x00000002  A8      : 0x3ffba2bc  A9      : 0x3ffddd40  
A10     : 0x3ffdf4e8  A11     : 0x3ffdf4e8  A12     : 0x00000014  A13     : 0x00000004  
A14     : 0x3ffdf4e0  A15     : 0x80000001  SAR     : 0x0000001f  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x401675a8  LEND    : 0x40167664  LCOUNT  : 0x00000000  

Backtrace: 0x400916ce:0x3ffddd60 0x4008f9e0:0x3ffddd90 0x4008f990:0x4016762d |<-CORRUPTED

  #0  0x400916ce:0x3ffddd60 in taskSelectHighestPriorityTaskSMP at C:\Users\Steve\.platformio\packages\framework-espidf\components\freertos/tasks.c:3436
      (inlined by) vTaskSwitchContext at C:\Users\Steve\.platformio\packages\framework-espidf\components\freertos/tasks.c:3519
  #1  0x4008f9e0:0x3ffddd90 in _frxt_dispatch at C:\Users\Steve\.platformio\packages\framework-espidf\components\freertos\port\xtensa/portasm.S:436
  #2  0x4008f990:0x4016762d in _frxt_int_exit at C:\Users\Steve\.platformio\packages\framework-espidf\components\freertos\port\xtensa/portasm.S:231

ELF file SHA256: baea1828482cda17

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
[   446][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
```

[image]

steve6375, 2023-07-01:
Here is a podcast URL

http://traffic.libsyn.com/theamphour/TheAmpHour-435-AndreasSpiess.mp3

seems to always crash...

steve6375, 2023-07-01:
Also crashing if VS1053 used. So not related to audio decoder?

steve6375, 2023-07-03:
POP Lounge cafe also resets system.

steve6375, 2023-07-04:
I am using VS1053 and the crashing/reboot issue still persists.
I recompiled for PCM5102 and it works OK (except for Speiss mp3 which is forbidden).
So I dont think you fixed the bug in the VS1053 ????

schreibfaul1, 2023-07-04:
yes, I could understand that. At the station 'Arabella Rock' an ssl connection is established and immediately afterwards an ssl redirection. 
The stack was 'under water' with a 'canary wdt' event. Hope this works now.

steve6375, 2023-07-05:
Still same issue in todays version. Arabella Rock resets system
WROVER with PSRAM, VS1053.

https://stream.laut.fm/poploungecafe
https://static.arabella.at/streamadressen/arabella-rock.m3u
http://traffic.libsyn.com/theamphour/TheAmpHour-435-AndreasSpiess.mp3

all cause crash and reboot.

schreibfaul1, 2023-07-05:
that's bad, I don't have the error (anymore).
Still getting `Debug exception reason: Stack canary watchpoint triggered (audioplay)`?
If so, please increase the stack in audiotask.cpp (line  182) from 7000 DWORDs to 7500. Would that help?

steve6375, 2023-07-05:
7500 seems OK. No crash now.

steve6375, 2023-07-05:
but after playing a few stations, trying to play the Adreas Spiess podcast causes reset

Seems to have a very long URL when expanded???

http://traffic.libsyn.com/theamphour/TheAmpHour-435-AndreasSpiess.mp3

```
11:36:14 action: ...  switch to station 38
11:36:14 AUDIO_info:  Connect to new host: "http://traffic.libsyn.com/theamphour/TheAmpHour-435-AndreasSpiess.mp3"
11:36:14 AUDIO_info:  Song stopped correctly after 20 msec
11:36:14 AUDIO_info:  Connection has been established in 100 ms, free Heap: 146392 bytes
11:36:15 AUDIO_info:  file "/logo/s/TheAmpHour Andres Spiess CRASH!.jpg" not found
11:36:15 AUDIO_info:  redirect to new host "https://traffic.libsyn.com/theamphour/TheAmpHour-435-AndreasSpiess.mp3"
11:36:15 AUDIO_info:  Connect to new host: "https://traffic.libsyn.com/theamphour/TheAmpHour-435-AndreasSpiess.mp3"
11:36:15 AUDIO_info:  Song stopped correctly after 0 msec
11:36:16 AUDIO_info:  SSL has been established in 1764 ms, free Heap: 102920 bytes
11:36:16 AUDIO_info:  redirect to new host "https://content.libsyn.com/p/a/3/8/a38e7f486749ff06/TheAmpHour-435-AndreasSpiess.mp3?c_id=37971800&cs_id=37971800&no-libsyn-stats=true&response-content-type=audio%2Fmpeg&Expires=1688556049&Signature=Fe66pDVB-6IYo~mLZv0UXJRmsBX-skSLSzRtBY1fJDNW9VQGrcWNlzuPf~oX3-GWyHb88Xm7~WilhUemK2UT~jVrrr-ZQVgN2o1jzAQ0AQ9KCIP-WqN0~5uNWxwuAtgniY8vLm-kI7NO-XV1TN2dPJXJrw59~OlwQ8rZ0kzxfnvKfiTBzxdYOyS0k8uTS5hgd9u98y4FMOY9eY9fc62MpeNPCpcMixTXoF-duxtS5grdtGFU3M8Lv6A8RrGa6FxvJGZT7CbQbs9ce2GriOmyfqxEi34j2ljCovVABXtuEIOHC"
11:36:17 AUDIO_info:  Connect to new host: "https://content.libsyn.com/p/a/3/8/a38e7f486749ff06/TheAmpHour-435-AndreasSpiess.mp3?c_id=37971800&cs_id=37971800&no-libsyn-stats=true&response-content-type=audio%2Fmpeg&Expires=1688556049&Signature=Fe66pDVB-6IYo~mLZv0UXJRmsBX-skSLSzRtBY1fJDNW9VQGrcWNlzuPf~oX3-GWyHb88Xm7~WilhUemK2UT~jVrrr-ZQVgN2o1jzAQ0AQ9KCIP-WqN0~5uNWxwuAtgniY8vLm-kI7NO-XV1TN2dPJXJrw59~OlwQ8rZ0kzxfnvKfiTBzxdYOyS0k8uTS5hgd9u98y4FMOY9eY9fc62MpeNPCpcMixTXoF-duxtS5grdtGFU3M8Lv6A8RrGa6FxvJGZT7CbQbs9ce2GriOmyfqxEi34j2ljCovVABXtuEIOHC"
11:36:17 AUDIO_info:  Song stopped correctly after 0 msec
11:36:17 AUDIO_info:  SSL has been established in 553 ms, free Heap: 100816 bytes
[ 58749][I][vs1053_ext.cpp:1761] parseHttpResponseHeader(): httpResponseHeader: HTTP/1.1 403 Forbidden
11:36:17 StreamTitle: HTTP/1.1 403 Forbidden
11:36:17 AUDIO_info:  Song stopped correctly after 0 msec
11:36:17 Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008cb99  PS      : 0x00060230  A0      : 0x801869fc  A1      : 0x3ffc69a0  
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffc6ce0  
A10     : 0x00000003  A11     : 0x00060223  A12     : 0x00060220  A13     : 0x00000000  
A14     : 0x007c70dc  A15     : 0x003fffff  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4008cb99  LEND    : 0x4008cba9  LCOUNT  : 0xffffffff

Backtrace: 0x4008cb96:0x3ffc69a0 0x401869f9:0x3ffc69b0 0x4018ebf9:0x3ffc6cc0 0x4018ec3f:0x3ffc6d50 0x40101e72:0x3ffc6d90 0x400dc8cb:0x3ffc6e30 0x40101dc5:0x3ffc6e90

  #0  0x4008cb96:0x3ffc69a0 in strlen at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:82
  #1  0x401869f9:0x3ffc69b0 in _svfprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1528
  #2  0x4018ebf9:0x3ffc6cc0 in _vsnprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vsnprintf.c:70 (discriminator 4)
  #3  0x4018ec3f:0x3ffc6d50 in vsnprintf at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vsnprintf.c:40
  #4  0x40101e72:0x3ffc6d90 in Print::printf(char const*, ...) at C:/Users/Steve/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.cpp:55
  #5  0x400dc8cb:0x3ffc6e30 in loop() at src/main.cpp:2312
  #6  0x40101dc5:0x3ffc6e90 in loopTask(void*) at C:/Users/Steve/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50

ELF file SHA256: dcccc3d7944baa20

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
[   446][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled

ESP32 Chip: ESP32-D0WD-V3
Arduino Version: 2.0.9
ESP-IDF Version: 4.4.4
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
PSRAM total size: 8388608 bytes

```

steve6375, 2023-07-05:
I added debug code to print out c

[image]

```
11:45:50 AUDIO_info:  SSL has been established in 539 ms, free Heap: 100780 bytes
[ 31408][I][vs1053_ext.cpp:1761] parseHttpResponseHeader(): httpResponseHeader: HTTP/1.1 403 Forbidden
11:45:50 StreamTitle: HTTP/1.1 403 Forbidden
11:45:50 AUDIO_info:  Song stopped correctly after 0 msec
11:45:50 Audiocodec:  83
11:45:50 Audiocodec:  � b
11:45:50 AUDIO_info:  file "/common/s/�A�.jpg" not found
```

Does not crash if accessed after reboot, but does crash after playing several stations.

steve6375, 2023-07-05:
I don't know C++ but testing for  c < 10 stops that crash.

But now it is crashing when selecting Radio 1 after trying to play Andres Spiess mp3???

http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_one/bbc_radio_one.isml/bbc_radio_one-audio=128000.norewind.m3u8

```
12:01:55 action: ...  switch to station 1
12:01:55 AUDIO_info:  Connect to new host: "http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_one/bbc_radio_one.isml/bbc_radio_one-audio=128000.norewind.m3u8"
12:01:55 AUDIO_info:  Song stopped correctly after 0 msec
[ 74494][I][vs1053_ext.cpp:3102] ts_parsePacket(): parseTS reset
12:01:55 AUDIO_info:  connect to as-hls-uk-live.akamaized.net on port 80 path /pool_904/live/uk/bbc_radio_one/bbc_radio_one.isml/bbc_radio_one-audio=128000.norewind.m3u8
12:01:55 AUDIO_info:  Connection has been established in 41 ms, free Heap: 145532 bytes
[ 74578][I][vs1053_ext.cpp:1761] parseHttpResponseHeader(): httpResponseHeader: HTTP/1.1 200 OK
[ 74578][I][vs1053_ext.cpp:1761] parseHttpResponseHeader(): httpResponseHeader: Server: openresty/1.13.6.2
[ 74587][I][vs1053_ext.cpp:1761] parseHttpResponseHeader(): httpResponseHeader: X-Cache-Status: HIT
[ 74597][I][vs1053_ext.cpp:1761] parseHttpResponseHeader(): httpResponseHeader: X-Pkgr-Instance: 00ba0568b2f392307
[ 74608][I][vs1053_ext.cpp:1761] parseHttpResponseHeader(): httpResponseHeader: Content-Type: application/vnd.apple.mpegurl
[ 74619][I][vs1053_ext.cpp:1761] parseHttpResponseHeader(): httpResponseHeader: Accept-Ranges: bytes
[ 74629][I][vs1053_ext.cpp:1761] parseHttpResponseHeader(): httpResponseHeader: ETag: "16885513472000000"
[ 74639][I][vs1053_ext.cpp:1761] parseHttpResponseHeader(): httpResponseHeader: X-USP: version=1.8.4 (13557)
[ 74650][I][vs1053_ext.cpp:1761] parseHttpResponseHeader(): httpResponseHeader: Last-Modified: Wed, 05 Jul 2023 10:01:52 GMT
[ 74662][I][vs1053_ext.cpp:1761] parseHttpResponseHeader(): httpResponseHeader: X-Cache-Info: caching
[ 74672][I][vs1053_ext.cpp:1761] parseHttpResponseHeader(): httpResponseHeader: Content-Length: 598
12:01:55 AUDIO_info:  content-length: 598
[ 74692][I][vs1053_ext.cpp:1761] parseHttpResponseHeader(): httpResponseHeader: Cache-Control: max-age=2
[ 74696][I][vs1053_ext.cpp:1761] parseHttpResponseHeader(): httpResponseHeader: Expires: Wed, 05 Jul 2023 10:01:58 GMT
[ 74708][I][vs1053_ext.cpp:1761] parseHttpResponseHeader(): httpResponseHeader: Date: Wed, 05 Jul 2023 10:01:56 GMT
[ 74718][I][vs1053_ext.cpp:1761] parseHttpResponseHeader(): httpResponseHeader: Connection: keep-alive
[ 74729][I][vs1053_ext.cpp:1761] parseHttpResponseHeader(): httpResponseHeader: Akamai-Mon-Iucid-Del: 496262
[ 74741][I][vs1053_ext.cpp:1761] parseHttpResponseHeader(): httpResponseHeader: Access-Control-Expose-Headers: Server,Range,Content-Length,Content-Range,Akamai-Mon-Iucid-Ing,Akamai-Mon-Iucid-Del,Akamai-Request-BC
[ 74759][I][vs1053_ext.cpp:1761] parseHttpResponseHeader(): httpResponseHeader: Access-Control-Allow-Methods: GET,HEAD,OPTIONS
[ 74772][I][vs1053_ext.cpp:1761] parseHttpResponseHeader(): httpResponseHeader: Access-Control-Allow-Headers: Origin,Range,Accept-Encoding,Referer,CMCD-Request,CMCD-Object,CMCD-Status,CMCD-Session
[ 74789][I][vs1053_ext.cpp:1761] parseHttpResponseHeader(): httpResponseHeader: Access-Control-Allow-Origin: *
[ 74799][I][vs1053_ext.cpp:1935] parseHttpResponseHeader(): now parse playlist
[ 74820][I][vs1053_ext.cpp:1290] readPlayListData(): pl=0 "#EXTM3U"
[ 74820][I][vs1053_ext.cpp:1290] readPlayListData(): pl=1 "#EXT-X-VERSION:3"
[ 74824][I][vs1053_ext.cpp:1290] readPlayListData(): pl=2 "## Created with Unified Streaming Platform(version=1.8.4)"
[ 74835][I][vs1053_ext.cpp:1290] readPlayListData(): pl=3 "#EXT-X-MEDIA-SEQUENCE:263836144"
[ 74844][I][vs1053_ext.cpp:1290] readPlayListData(): pl=4 "#EXT-X-INDEPENDENT-SEGMENTS"
[ 74853][I][vs1053_ext.cpp:1290] readPlayListData(): pl=5 "#EXT-X-TARGETDURATION:6"
[ 74861][I][vs1053_ext.cpp:1290] readPlayListData(): pl=6 "#USP-X-TIMESTAMP-MAP:MPEGTS=5086400928,LOCAL=2023-07-05T10:01:55.200000Z"
[ 74874][I][vs1053_ext.cpp:1290] readPlayListData(): pl=7 "#EXT-X-PROGRAM-DATE-TIME:2023-07-05T10:01:55.200000Z"
[ 74884][I][vs1053_ext.cpp:1290] readPlayListData(): pl=8 "#EXTINF:6.4, no desc"
[ 74893][I][vs1053_ext.cpp:1290] readPlayListData(): pl=9 "bbc_radio_one-audio=128000-263836144.ts"
[ 74902][I][vs1053_ext.cpp:1290] readPlayListData(): pl=10 "#EXTINF:6.4, no desc"
[ 74910][I][vs1053_ext.cpp:1290] readPlayListData(): pl=11 "bbc_radio_one-audio=128000-263836145.ts"
[ 74920][I][vs1053_ext.cpp:1290] readPlayListData(): pl=12 "#EXTINF:6.4, no desc"
[ 74928][I][vs1053_ext.cpp:1290] readPlayListData(): pl=13 "bbc_radio_one-audio=128000-263836146.ts"
[ 74938][I][vs1053_ext.cpp:1290] readPlayListData(): pl=14 "#EXTINF:6.4, no desc"
[ 74946][I][vs1053_ext.cpp:1290] readPlayListData(): pl=15 "bbc_radio_one-audio=128000-263836147.ts"
[ 74956][I][vs1053_ext.cpp:1290] readPlayListData(): pl=16 "#EXTINF:6.4, no desc"
[ 74964][I][vs1053_ext.cpp:1290] readPlayListData(): pl=17 "bbc_radio_one-audio=128000-263836148.ts"
[ 74975][I][vs1053_ext.cpp:1507] parsePlaylist_M3U8(): m_m3u8_targetDuration 6
[ 74982][I][vs1053_ext.cpp:1507] parsePlaylist_M3U8(): m_m3u8_targetDuration 6
[ 74990][I][vs1053_ext.cpp:1507] parsePlaylist_M3U8(): m_m3u8_targetDuration 6
[ 74998][I][vs1053_ext.cpp:1507] parsePlaylist_M3U8(): m_m3u8_targetDuration 6
[ 75006][I][vs1053_ext.cpp:1507] parsePlaylist_M3U8(): m_m3u8_targetDuration 6
[... cut ...]
  #4  0x40099729:0x3ffd6f50 in free at C:\Users\Steve\.platformio\packages\framework-espidf\components\newlib/heap.c:39
  #5  0x400e8772:0x3ffd6f70 in VS1053::parsePlaylist_M3U8() at .pio/libdeps/esp32/ESP32-vs1053_ext/src/vs1053_ext.cpp:1563
  #6  0x400ea8c4:0x3ffd6fc0 in VS1053::loop() at .pio/libdeps/esp32/ESP32-vs1053_ext/src/vs1053_ext.cpp:596
  #7  0x400d382f:0x3ffd6fe0 in audioTask(void*) at src/audiotask.cpp:174

ELF file SHA256: 5be6ca445be7959e

Rebooting...
ets Jul 29 2019 12:21:46

```

schreibfaul1, 2023-07-05:
Thanks for the good analysis, the crash has to do with the length of the URL. I don't think I've ever had a URL with more than 500 characters. I do not want to change anything about the permitted length at this point in time, this would have to be done in several places.
Since the audioI2S Lib works, I copied the connecttohost() code to the vs1053ext. First of all, this is the simple and time-saving method and I would like to keep the basic routines for both libraries the same anyway. And the stack is also increased. to 7500 DWORDs.

steve6375, 2023-07-05:
Do you need to fix this code in main too?

const char *codecname[10] = {"unknown", "WAV", "MP3", "AAC", "M4A", "FLAC", "AACP", "OPUS", "OGG", "VORBIS"};

        if(_cur_Codec == 0){
            uint8_t c = audioGetCodec();
            if(c != 0 && c != 8){ // unknown or OGG   ***************************  NO CHECK !!!!!
                _cur_Codec = c;
                SerialPrintfln("Audiocodec:  " ANSI_ESC_YELLOW "%s", codecname[c]);
                if(_state == PLAYER || _state == PLAYERico) showFileLogo(_state);
                if(_state == RADIO && _f_logoUnknown == true) {_f_logoUnknown = false; showFileLogo(_state);}
            }
        }

No check is done on size of c
c could be larger than the array and cause a crash, as happened in my case above.

closed 2023-07-05 by steve6375