schreibfaul1/ESP32-MiniWebRadio #215: Recompile without PSRAM causes SSID wifi connect issues

steve6375, 2023-07-02:
I have esp32+PSRAM.
network file has my SSID and password in the file on SD card.

If I compile for PSRAM enabled, all is OK:
```
ELF file SHA256: 2d2c2de7118783bf

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
[   446][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled

ESP32 Chip: ESP32-D0WD-V3
Arduino Version: 2.0.9
ESP-IDF Version: 4.4.4
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
PSRAM total size: 8388608 bytes

08:27:50
08:27:50        ***************************    
08:27:50        *     MiniWebRadio V2     *    
08:27:50        ***************************    
08:27:50
08:27:50 setup: ....  Arduino is pinned to core 1
08:27:50 tft_info: .  init ILI9341
08:27:50 setup: ....  Init SD card
08:27:51 setup: ....  SD card found, 3694.5 MB by 3886.5 MB free
08:27:51 setup: ....  seek for stations.csv
08:27:51 setup: ....  stations.csv found
08:27:51 setup: ....  seek for WiFi networks
[  1152][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: gertrude
[  1158][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: gertrude
[  1166][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: gertrude
[  7286][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[  7286][I][WiFiMulti.cpp:119] run(): [WIFI] 2 networks found
[  7287][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 84:16:F9:2B:6C:AF SSID: gertrude Channel: 4 (-26)
[  8063][I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
08:27:58 WiFI_info:   Connecting WiFi...
08:27:58 setup: ....  connected to gertrude, IP address is 192.168.1.114
08:27:58 ftpServer:   Buffers allocated: 4622 bytes
08:27:58 rtime_info:  Initializing SNTP
09:27:58 rtime_info:  09:27:58
09:27:58 audiotask:   is pinned to core 1
09:27:58 audiotask:   priority is 2

```

However, if I comment out PSRAM in platformio.ini, then I cannot connect to Wifi (even if I use mobile phone and set  SSID and pwd via 192.168.4.1 . It still won't connect.

;        -DBOARD_HAS_PSRAM 

`

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0

ESP32 Chip: ESP32-D0WD-V3
Arduino Version: 2.0.9
ESP-IDF Version: 4.4.4
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)

00:00:00    
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V2     *    
00:00:00        ***************************    
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9341
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 3694.5 MB by 3886.5 MB free
00:00:00 setup: ....  seek for stations.csv
00:00:00 setup: ....  stations.csv found
00:00:00 setup: ....  seek for WiFi networks
[   735][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: gertrude
[   741][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: gertrude
[   750][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: gertrude
[  6674][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[  6675][I][WiFiMulti.cpp:119] run(): [WIFI] 1 networks found
[  6676][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 84:16:F9:2B:6C:AF SSID: gertrude Channel: 4 (-25)
[  9691][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT
[ 11708][E][WiFiMulti.cpp:187] run(): [WIFI] Connecting Failed (6).
00:00:11 WiFI_info:   WiFi credentials are not correct
[ 11763][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 8 - ASSOC_LEAVE
00:00:11 Accesspoint: IP: 192.168.4.1
00:00:18 setup: ....  1 WiFi networks found
00:00:18 setup: ....  gertrude (-25)
00:00:18 ftpServer:   Ftp server waiting for connection on port 21

`

steve6375, 2023-07-02:
I have added dbeug code to print out passphrase in WiFiMulti.cpp and it is using the correct SSID and passphrase.
Also increased timeout to 20 seconds.

```
00:00:00 setup: ....  stations.csv found
00:00:00 setup: ....  seek for WiFi networks
[   715][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: gertrude mypass
[   722][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: gertrude mypass
[   732][I][main.cpp:1152] connectToWiFi(): s_ssid=gertrude  s_password=mypass s_info=home
[   736][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: gertrude mypass
[  6869][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[  6869][I][WiFiMulti.cpp:119] run(): [WIFI] 2 networks found
[  6870][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 84:16:F9:2B:6C:AF SSID: gertrude mypass Channel: 4 (-27)
[  9886][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT
[ 12896][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT
[ 15897][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT
[ 18896][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT
[ 21896][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT
[ 24896][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT
[ 26903][E][WiFiMulti.cpp:187] run(): [WIFI] Connecting Failed (6).
00:00:26 WiFI_info:   WiFi credentials are not correct
[ 26958][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 8 - ASSOC_LEAVE
```

steve6375, 2023-07-02:
Here is terminal log for recompile with PSRAM enabled (no problem!):
```

00:00:00 setup: ....  SD card found, 3694.5 MB by 3886.5 MB free
00:00:01 setup: ....  seek for stations.csv
00:00:01 setup: ....  stations.csv found
00:00:01 setup: ....  seek for WiFi networks
[  1145][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: gertrude mypass
[  1152][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: gertrude mypass
[  1161][I][main.cpp:1152] connectToWiFi(): s_ssid=gertrude  s_password=mypass s_info=home
[  1166][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: gertrude mypass
[  7095][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[  7095][I][WiFiMulti.cpp:119] run(): [WIFI] 1 networks found
[  7096][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 84:16:F9:2B:6C:AF SSID: gertrude mypass Channel: 4 (-26)
[  7869][I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
00:00:07 WiFI_info:   Connecting WiFi...
00:00:07 setup: ....  connected to gertrude, IP address is 192.168.1.114
00:00:07 ftpServer:   Buffers allocated: 4622 bytes
00:00:07 rtime_info:  Initializing SNTP
```

Batzi1967, 2023-07-02:
I have the same problem with several ESP32 without psram:

ESP32 Chip: ESP32-D0WD-V3
Arduino Version: 2.0.9
ESP-IDF Version: 4.4.4
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)

00:00:00
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V2     *    
00:00:00        ***************************    
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9341
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 15166.5 MB by 15199.0 MB free
00:00:00 setup: ....  seek for stations.csv
00:00:00 setup: ....  stations.csv found
00:00:00 setup: ....  seek for WiFi networks
[   733][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: FRITZ!Box 7490
[   741][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: FRITZ!Box Fon WLAN 7320
[   754][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: FRITZ!Box 7490
[   755][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: FRITZ!Box Fon WLAN 7320
[   764][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: FRITZ!Box Fon WLAN 7320
[   773][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_04
[   781][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_04
[   812][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   904][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
[   905][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 2 - STA_START
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
[     1][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=21, ...)
[     2][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=50009, ...)
[     7][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
[    14][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)

ESP32 Chip: ESP32-D0WD-V3
Arduino Version: 2.0.9
ESP-IDF Version: 4.4.4
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)

00:00:00
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V2     *    
00:00:00        ***************************    
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9341
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 15166.5 MB by 15199.0 MB free
00:00:00 setup: ....  seek for stations.csv
00:00:00 setup: ....  stations.csv found
00:00:00 setup: ....  seek for WiFi networks
[   733][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: FRITZ!Box 7490
[   741][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: FRITZ!Box Fon WLAN 7320
[   754][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: FRITZ!Box 7490
[   755][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: FRITZ!Box Fon WLAN 7320
[   764][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: FRITZ!Box Fon WLAN 7320
[   773][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_04
[   781][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_04
[   812][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 0 - WIFI_READY
[... 12 lines cut ...]
[  9835][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: FRITZ!Box Fon WLAN 7320, BSSID: bc:05:43:94:6f:eb, Reason: 15
[  9837][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  9846][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT
[  9855][D][WiFiGeneric.cpp:1077] _eventCallback(): WiFi Reconnect Running
[  9864][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[ 11803][E][WiFiMulti.cpp:187] run(): [WIFI] Connecting Failed (6).
00:00:11 WiFI_info:   WiFi credentials are not correct
[ 11857][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: FRITZ!Box Fon WLAN 7320, BSSID: bc:05:43:94:6f:eb, Reason: 8
[ 11860][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 11868][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 8 - ASSOC_LEAVE
[ 11878][V][WiFiGeneric.cpp:392] _arduino_event_cb(): AP Started
[ 11889][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 10 - AP_START
[ 11890][V][WiFiGeneric.cpp:395] _arduino_event_cb(): AP Stopped
[ 11897][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 11 - AP_STOP
[ 11898][V][WiFiGeneric.cpp:392] _arduino_event_cb(): AP Started
[ 11912][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 10 - AP_START
00:00:11 Accesspoint: IP: 192.168.4.1
[ 18164][V][WiFiGeneric.cpp:383] _arduino_event_cb(): SCAN Done: ID: 129, Status: 0, Results: 4
[ 18165][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 1 - SCAN_DONE
00:00:18 setup: ....  4 WiFi networks found
00:00:18 setup: ....  FRITZBox7490Gastzugang (-47)
00:00:18 setup: ....  FRITZ!Box Fon WLAN 7320 (-53)
00:00:18 setup: ....  FRITZ!Box 7490 (-78)
00:00:18 setup: ....  FRITZBox7490Gastzugang (-79)
00:00:18 ftpServer:   Ftp server waiting for connection on port 21

TubaFonse, 2023-07-02:
I have the same connection problem with ESP32 without psram, 4M Flash, VS1053 decoder, TFT+TP ILI9341 configuration: 

00:00:18 setup: ....  11 WiFi networks found
00:00:18 setup: ....  Nw5Box7490GastLF (-40)
00:00:18 setup: ....  FRITZ!Box7490LF (-41)
00:00:18 setup: ....  Nw5Box7490GastLF (-77)
00:00:18 setup: ....  FRITZ!Box7490LF (-78)
00:00:18 setup: ....  Gast bei LaLeLuFred (-81)
00:00:18 setup: ....  LaLeLuFred (-82)
00:00:18 setup: ....  Gast bei LaLeLuFred (-88)
00:00:18 setup: ....  FRITZ!Box Fon WLAN 7360 (-89)
00:00:18 setup: ....  Fritz-Box7430Mar3 (-89)
00:00:18 setup: ....  FRITZ!Box 7430 MP (-91)
00:00:18 setup: ....  LaLeLuFred (-91)
00:00:18 ftpServer:   Ftp server waiting for connection on port 21
ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
ESP32 Chip: ESP32-D0WDQ6
Arduino Version: 2.0.9
ESP-IDF Version: 4.4.4
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
00:00:00
00:00:00        ***************************
00:00:00        *     MiniWebRadio V2     *    
00:00:00        ***************************
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9341
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 1920.0 MB by 1938.5 MB free
00:00:00 setup: ....  seek for stations.csv
00:00:00 setup: ....  stations.csv found
00:00:00 setup: ....  seek for WiFi networks
[   669][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: FRITZ!Box7490LF
[  7011][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[  7011][I][WiFiMulti.cpp:119] run(): [WIFI] 11 networks found
[  7012][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: C8:0E:14:0C:93:11 SSID: FRITZ!Box7490LF Channel: 6 (-40)
[ 10107][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT
[ 12041][E][WiFiMulti.cpp:187] run(): [WIFI] Connecting Failed (6).
00:00:12 WiFI_info:   WiFi credentials are not correct
[ 12095][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 8 - ASSOC_LEAVE
00:00:12 Accesspoint: IP: 192.168.4.1
00:00:18 setup: ....  10 WiFi networks found
00:00:18 setup: ....  FRITZ!Box7490LF (-38)
00:00:18 setup: ....  Nw5Box7490GastLF (-38)
00:00:18 setup: ....  FRITZ!Box7490LF (-78)
00:00:18 setup: ....  Nw5Box7490GastLF (-79)
00:00:18 setup: ....  LaLeLuFred (-79)
00:00:18 setup: ....  Gast bei LaLeLuFred (-80)
00:00:18 setup: ....  Fritz-Box7430Mar3 (-87)
00:00:18 setup: ....  Gast bei LaLeLuFred (-88)
00:00:18 setup: ....  LaLeLuFred (-89)
00:00:18 setup: ....  FRITZ!Box Fon WLAN 7360 (-93)
00:00:18 ftpServer:   Ftp server waiting for connection on port 21

TubaFonse, 2023-07-03:
Wifi in Old Firmware Version 2.4.0 plays, but has sporadic hang-up problem: 

ESP32 Chip: ESP32-D0WDQ6
Arduino Version: 2.0.3

00:00:00
00:00:00        ***************************
00:00:00        *     MiniWebRadio V2     *
00:00:00        ***************************
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9341
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found
00:00:00 setup: ....  seek for stations.csv
00:00:00 setup: ....  stations.csv found
00:00:00 setup: ....  seek for WiFi networks
00:00:02 WiFI_info:   Connecting WiFi...
00:00:02 setup: ....  connected to FRITZ!Box7490LF, IP address is 192.168.178.35
00:00:02 ftpServer:   Buffers allocated: 4622 bytes
00:00:02 rtime_info:  Initializing SNTP
00:00:02 rtime_info:  Waiting for system time to be set... (1/10)
23:04:36 rtime_info:  23:04:36
23:04:36 setup: ....  Number of saved stations: 48
23:04:36 setup: ....  current station number: 1
23:04:36 setup: ....  current volume: 12
23:04:37 VS1053 chipID = 1279067256, version = 3
23:04:37 action: ...  current volume is 12
23:04:37 action: ...  switch to station 1
23:04:38 ftpServer:   Ftp server waiting for connection on port 21
23:04:38 icy-descr:   BAYERN 1 Oberbayern
23:04:38 StationName: BAYERN 1 Oberbayern
23:04:38 lastURL: ..  http://d121.rndfnk.com/ard/br/br1/obb/mp3/128/stream.mp3?cid=01FB9ME7567MEQ5G72EVEEVVCS&sid=2S25Xw1uph61WoXv7qhGfQ95fIf&token=9wwWw5a2pwuGR7YtFE1ZVnIy_CKMZ-Z3yrEpQ3E_mlk&tvf=bEFqBYs8bhdkMTIxLnJuZGZuay5jb20
23:04:38 StreamTitle: +++ Zehntausende Polizisten sollen Unruhen in Frankreich verhindern
23:04:48 StreamTitle: Boston: Amanda
23:05:51 StreamTitle: Die Nacht auf BAYERN 1
23:06:05 StreamTitle: Boston: Amanda
23:06:16 icy-descr:   BAYERN 1 Oberbayern
23:06:16 StationName: BAYERN 1 Oberbayern
23:06:16 lastURL: ..  http://d121.rndfnk.com/ard/br/br1/obb/mp3/128/stream.mp3?cid=01FB9ME7567MEQ5G72EVEEVVCS&sid=2S25Xw1uph61WoXv7qhGfQ95fIf&token=9wwWw5a2pwuGR7YtFE1ZVnIy_CKMZ-Z3yrEpQ3E_mlk&tvf=bEFqBYs8bhdkMTIxLnJuZGZuay5jb20
23:06:16 StreamTitle: Boston: Amanda
23:08:30 StreamTitle: Mavis Staples: Blood Is Thicker Than Time
23:09:22 StreamTitle: Die Nacht auf BAYERN 1
23:09:55 StreamTitle: Mavis Staples: Blood Is Thicker Than Time
23:10:56 AUDIO_info:  Request http://d121.rndfnk.com/ard/br/br1/obb/mp3/128/stream.mp3?cid=01FB9ME7567MEQ5G72EVEEVVCS&sid=2S25Xw1uph61WoXv7qhGfQ95fIf&token=9wwWw5a2pwuGR7YtFE1ZVnIy_CKMZ-Z3yrEpQ3E_mlk&tvf=bEFqBYs8bhdkMTIxLnJuZGZuay5jb20 failed!
23:10:56 StreamTitle:

after reset:

TubaFonse, 2023-07-03:
Test of latest version July 3rd 2023 Time: 01:52 fails again: 

ESP32 Chip: ESP32-D0WDQ6
Arduino Version: 2.0.9
ESP-IDF Version: 4.4.4
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)

00:00:00
00:00:00        ***************************
00:00:00        *     MiniWebRadio V2     *
00:00:00        ***************************
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9341
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 1920.0 MB by 1938.5 MB free
00:00:00 setup: ....  seek for stations.csv
00:00:00 setup: ....  stations.csv found
00:00:00 setup: ....  seek for WiFi networks
[   662][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: FRITZ!Box7490LF
[  7196][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[  7197][I][WiFiMulti.cpp:119] run(): [WIFI] 11 networks found
[  7198][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: C8:0E:14:0C:93:11 SSID: FRITZ!Box7490LF Channel: 1 (-46)
[  7237][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 202 - AUTH_FAIL
[  7308][I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
00:00:07 WiFI_info:   Connecting WiFi...
00:00:07 setup: ....  connected to FRITZ!Box7490LF, IP address is 192.168.178.35
00:00:07 ftpServer:   Buffers allocated: 4622 bytes
00:00:07 rtime_info:  Initializing SNTP
00:00:07 rtime_info:  Waiting for system time to be set... (1/10)
01:46:39 rtime_info:  01:46:39
01:46:39 audiotask:   is pinned to core 1
01:46:39 audiotask:   priority is 2
01:46:40 VS1053 chipID = 1279067256, version = 3
01:46:40 setup: ....  Number of saved stations: 48
01:46:40 setup: ....  current station number: 3
01:46:40 setup: ....  current volume: 12
01:46:40 setup: ....  last connected host: http://d131.rndfnk.com/ard/br/br1/obb/mp3/128/stream.mp3?cid=01FB9ME7567MEQ5G72EVEEVVCS&sid=2Rz5ivRgCPXx1lfPrfB9tOFNhIt&token=ZA_rOb4NRyH5FVdLFCsDwiZOF0JaQ4ZnUcNJwGwjU8s&tvf=ubzAeSnpbRdkMTMxLnJuZGZuay5jb20
01:46:40 setup: ....  connection timeout: 500 ms
01:46:40 setup: ....  connection timeout SSL: 2000 ms
01:46:40 action: ...  current volume is 12
[  9906][E][Preferences.cpp:483] getString(): nvs_get_str len fail: station_003 NOT_FOUND
01:46:40 action: ...  switch to station 3
01:46:40 AUDIO_info:  Connect to new host: "http://not_found"
01:46:42 AUDIO_info:  Song stopped incorrectly!
01:46:42 AUDIO_info:  after song stopped incorrectly
01:46:42 AUDIO_info:  REG            dec      bin               hex
01:46:42 AUDIO_info:  -----------  -------  ----------------  -------
01:46:42 AUDIO_info:  MODE          02056   0000100000001000   0808
01:46:42 AUDIO_info:  STATUS        00576   0000001001000000   0240
01:46:42 AUDIO_info:  BASS          00000   0000000000000000   0000
01:46:42 AUDIO_info:  CLOCKF        24576   0110000000000000   6000
01:46:42 AUDIO_info:  DECODE_TIME   00002   0000000000000010   0002
01:46:42 AUDIO_info:  AUDATA        44101   1010110001000101   AC45
01:46:42 AUDIO_info:  WRAM          00000   0000000000000000   0000
01:46:42 AUDIO_info:  WRAMADDR      07682   0001111000000010   1E02
01:46:42 AUDIO_info:  HDAT0         00000   0000000000000000   0000
01:46:42 AUDIO_info:  HDAT1         00000   0000000000000000   0000
01:46:42 AUDIO_info:  AIADDR        00000   0000000000000000   0000
01:46:42 AUDIO_info:  VOL           06425   0001100100011001   1919
01:46:42 AUDIO_info:  AICTRL0       00000   0000000000000000   0000
01:46:42 AUDIO_info:  AICTRL1       00000   0000000000000000   0000
01:46:42 AUDIO_info:  AICTRL2       00000   0000000000000000   0000
01:46:42 AUDIO_info:  AICTRL3       00000   0000000000000000   0000
01:46:42 AUDIO_info:  PSRAM not found, inputBufferSize: 14399 bytes
[ 12069][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for not_found
01:46:42 AUDIO_info:  Request http://not_found failed!
01:46:42 StreamTitle:
01:46:42 WiFI_info:   RSSI is -39 dB
01:46:42 DLNA_info:   SSDP search for media servers started
01:46:46 DLNA_info:   SSDP query discovered 2 media servers
01:46:46 DLNA_info:   connected successfully to 192.168.178.24:50001
01:46:46 DLNA_info:   connected successfully to 192.168.178.29:2869
01:46:46 ftpServer:   Ftp server waiting for connection on port 21
01:59:53 Time: ...... Announcement at 2 o'clock is silent

steve6375, 2023-07-05:
seems OK now.

closed 2023-07-05 by steve6375