schreibfaul1/ESP32-MiniWebRadio #216: The radio won't load

valera39, 2023-07-02:
Hello Wolle!
The latest firmware version does not load the radio. ESP32-S3 N8R8, PCM5102.
Log:
ESP32 Chip: ESP32-S3
Arduino Version: 2.0.9
ESP-IDF Version: 4.4.4
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
PSRAM total size: 8388608 bytes

00:00:18
00:00:18        ***************************    
00:00:18        *     MiniWebRadio V2     *    
00:00:18        ***************************    
00:00:18
00:00:18 setup: ....  Arduino is pinned to core 1
00:00:18 tft_info: .  init ILI9488
00:00:18 setup: ....  Init SD card
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377975
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
[   983][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=21, ...)
[   983][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=50009, ...)
[   988][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
[   996][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
[  1004][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
E (1027) esp_core_dump_flash: Core dump data check failed:
Calculated checksum='cf46c0bf'
Image checksum='ffffffff'

clear-sky, 2023-07-02:
hi dear valera39
my config is esp-s3-12k + pcm5102 but have no problem
radio work perfect!

valera39, 2023-07-02:
[clear-sky](https://github.com/clear-sky), hello!
Was the data on the map changed by the author? I'll certainly try, but I have no problem flashing the previous version and it works

clear-sky, 2023-07-02:
Hi Dear valera39!
when i updated 2.0.8 to 2.0.9  i confirm "Content_on_SD_Card.zip" is updated!

valera39, 2023-07-02:
Unfortunately, updating the files on the SD card did not help.

valera39, 2023-07-02:
[clear-sky](https://github.com/clear-sky), Thanks for the tip. It all worked after completely clearing the ESP memory and SD card. Apparently this is sometimes necessary :)
The question can be withdrawn

closed 2023-07-02 by valera39