schreibfaul1/ESP32-MiniWebRadio #217: feature request(sd card file browser)

clear-sky, 2023-07-02:


valera39, 2023-07-02:
Yes, that's very cool. It would be nice to be able to control it with the touchscreen as much as possible.

clear-sky, 2023-07-02:
Hi dear valera39!
100% agree with you

steve6375, 2023-07-02:
Just use FileZilla to upload files to the SD card.

clear-sky, 2023-07-02:
Hi dear steve6375!
thanks for the reply 
can you explain more?
in host section i entered radio ip
but give error-- may be not entered port number
this is error message:
```
Connecting to 192.168.1.3:21...
Status:	Connection established, waiting for welcome message...
Response:	220--- Welcome to FTP for ESP32 ---
Response:	220---   By David Paiva   ---
Response:	220 --   Version FTP-2020-12-12   --
Command:	AUTH TLS
Response:	500 Syntax error
Command:	AUTH SSL
Response:	530 user not found
Status:	Insecure server, it does not support FTP over TLS.
Command:	USER anonymous
Error:	Could not read from socket: ECONNRESET - Connection reset by peer
Error:	Could not connect to server
```

valera39, 2023-07-02:
[clear-sky](https://github.com/clear-sky) download and install Filezila on your computer and configure the program as recommended by Wolle

steve6375, 2023-07-02:
P.s. you need the non-server version of FileZilla (desktop version)

valera39, 2023-07-02:
[clear-sky](https://github.com/clear-sky), here it is https://filezilla-project.org/

clear-sky, 2023-07-03:
Hi All!
thank you for your replays
problem solved

closed 2023-07-03 by clear-sky