schreibfaul1/ESP32-MiniWebRadio #218: feature request(show playing album picture)

clear-sky, 2023-07-02:
request is album picture if avalable!
as far as i know you are linux user like me (this makes me very happy!)
if you are on kde please try to add a widget (right click on panel> add widget> get new widget > download new plasma widget > use filter > show highest rated first > advanced radio player
this widget downloads playing album picture and shows it as thumbnail on panel
HOW THIS IDEA MOST BE ON YOUR RADIO?
when switching to new station radio shows station logo for a while (for example 5 seconds )
then if playing album picture found replaces it with radio logo!!
do you like this idea?

schreibfaul1, 2023-07-04:
Don't have kde but cinnamon. But I know what you mean. Many audio files have embedded images.
Do you know the ESPuino project? This is a music box for children. If images are available, they will be displayed on the webpage.

clear-sky, 2023-07-05:
Hi dear wolle
Thanks for the reply
as far as i understood about that widget it searches stream title in last.fm and grabs album cover from those site!
https://www.last.fm/

closed 2024-02-09 by github-actions[bot]