schreibfaul1/ESP32-MiniWebRadio #220: the radio station causes a reboot

valera39, 2023-07-03:
Hello Wolle!
This is the station where the radio restarts https://sovetskie-pesni.ru:8005/stream
Hello Wolle!
вот на этой станции радио перезагружается 
ESP32 Chip: ESP32-S3
Arduino Version: 2.0.9
ESP-IDF Version: 4.4.4
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
PSRAM total size: 8388608 bytes

00:00:00
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V2     *    
00:00:00        ***************************    
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9488
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 30119.1 MB by 30159.0 MB free
00:00:01 setup: ....  seek for stations.csv
00:00:01 setup: ....  stations.csv found
00:00:01 setup: ....  seek for WiFi networks
[  2571][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: ASUS_AC66U
[  2583][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: ASUS_AC66U
[  2585][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_01
[  2592][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_02
[  2600][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_03
[  2608][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_04
[  2616][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_04
[  2641][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 0 - WIFI_READY
[  2676][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
[  2677][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 2 - STA_START
[  8881][V][WiFiGeneric.cpp:383] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 6
[  8881][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[  8888][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[  8893][I][WiFiMulti.cpp:119] run(): [WIFI] 6 networks found
[  8900][D][WiFiMulti.cpp:149] run():  --->   0: [1][2C:FD:A1:C2:44:F0] ASUS_AC66U (-54) *
[  8909][D][WiFiMulti.cpp:151] run():        1: [5][04:D9:F5:56:3C:68] TP-LINK_E345E0_RPT (-60) *
[  8918][D][WiFiMulti.cpp:151] run():        2: [1][1C:AF:F7:B3:8E:CB] ASUS_AC66U_REPITER (-72) *
[  8928][D][WiFiMulti.cpp:151] run():        3: [5][30:B5:C2:E3:45:E0] TP-LINK_E345E0 (-79) *
[  8937][D][WiFiMulti.cpp:151] run():        4: [4][D8:07:B6:B7:FE:0F] GG (-89) *
[  8945][D][WiFiMulti.cpp:151] run():        5: [12][2C:FD:A1:03:D7:60] ASUS_60 (-91) *
[  8954][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 2C:FD:A1:C2:44:F0 SSID: ASUS_AC66U Channel: 1 (-54)
[  8966][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0      
[  9037][V][WiFiGeneric.cpp:355] _arduino_event_cb(): STA Connected: SSID: ASUS_AC66U, BSSID: 2c:fd:a1:c2:44:f0, Channel: 1, Auth: WPA2_PSK
[  9040][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[  9057][V][WiFiGeneric.cpp:369] _arduino_event_cb(): STA Got New IP:192.168.1.130
[  9057][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[  9064][D][WiFiGeneric.cpp:1098] _eventCallback(): STA IP: 192.168.1.130, MASK: 255.255.255.0, GW: 192.168.1.1
[  9075][I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
[  9082][D][WiFiMulti.cpp:175] run(): [WIFI] SSID: ASUS_AC66U
[  9088][D][WiFiMulti.cpp:176] run(): [WIFI] IP: 192.168.1.130
[  9094][D][WiFiMulti.cpp:177] run(): [WIFI] MAC: 2C:FD:A1:C2:44:F0
[  9101][D][WiFiMulti.cpp:178] run(): [WIFI] Channel: 1
00:00:08 WiFI_info:   Connecting WiFi...
00:00:08 setup: ....  connected to ASUS_AC66U, IP address is 192.168.1.130
00:00:08 ftpServer:   Buffers allocated: 4622 bytes
00:00:08 rtime_info:  Initializing SNTP
00:00:08 rtime_info:  Waiting for system time to be set... (1/10)
00:00:10 rtime_info:  Waiting for system time to be set... (2/10)
16:44:53 rtime_info:  16:44:53
16:44:53 audiotask:   is pinned to core 0
16:44:53 audiotask:   priority is 2
16:44:53 setup: ....  Number of saved stations: 89
16:44:53 setup: ....  current station number: 16
16:44:53 setup: ....  current volume: 3
16:44:53 setup: ....  last connected host: https://sovetskie-pesni.ru:8005/stream
16:44:53 setup: ....  connection timeout: 500 ms
16:44:53 setup: ....  connection timeout SSL: 2000 ms
16:44:54 action: ...  current volume is 3
16:44:54 action: ...  switch to station 16
16:44:54 AUDIO_info:  Connect to new host: "https://sovetskie-pesni.ru:8005/stream"
16:44:54 AUDIO_info:  PSRAM found, inputBufferSize: 638965 bytes
[ 14100][V][ssl_client.cpp:321] stop_ssl_socket(): Cleaning SSL connection.
16:44:54 AUDIO_info:  buffers freed, free Heap: 169984 bytes
[ 14115][V][ssl_client.cpp:62] start_ssl_client(): Free internal heap before TLS 169984
[ 14118][V][ssl_client.cpp:68] start_ssl_client(): Starting socket
[ 14161][V][ssl_client.cpp:146] start_ssl_client(): Seeding the random number generator
[ 14164][V][ssl_client.cpp:155] start_ssl_client(): Setting up the SSL/TLS structure...
[ 14168][D][ssl_client.cpp:176] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
[ 14177][V][ssl_client.cpp:254] start_ssl_client(): Setting hostname for TLS session...
[ 14187][V][ssl_client.cpp:269] start_ssl_client(): Performing the SSL/TLS handshake...
[ 15104][V][ssl_client.cpp:290] start_ssl_client(): Verifying peer X.509 certificate...
[ 15105][V][ssl_client.cpp:298] start_ssl_client(): Certificate verified.
[ 15109][V][ssl_client.cpp:313] start_ssl_client(): Free internal heap after TLS 169028
16:44:55 AUDIO_info:  SSL has been established in 1004 ms, free Heap: 169028 bytes
[ 15129][V][ssl_client.cpp:369] send_ssl_data(): Writing HTTP request with 127 bytes...
16:44:55 bitRate:     128Kbit/s
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400556d5  PS      : 0x00060930  A0      : 0x82020f91  A1      : 0x3fcc0fb0
A2      : 0xb0d09dd0  A3      : 0xb0d09dcc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x000000d0
A10     : 0x3fcc11ea  A11     : 0xffffffce  A12     : 0x00000031  A13     : 0xffffffd1
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c
EXCVADDR: 0xb0d09dd0  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff

Backtrace: 0x400556d2:0x3fcc0fb0 |<-CORRUPTED
Thank you!

steve6375, 2023-07-03:
See #214

valera39, 2023-07-03:
Thank you, I have checked. In youradio project, which uses modified audio libraries from Wolle, the station works without problems.
We will wait for a solution

schreibfaul1, 2023-07-03:
https://sovetskie-pesni.ru:8005/stream has a long icy description. Encoded in UTF-8, the string has exceeded the 512 byte limit. I had not provided a correct limit. Der Fehler ist in beiden Bibliotheken audioI2S und vs1053ext. The bug is in both audioI2S and vs1053ext libraries. Exceeding the buffer limit does not necessarily lead to a crash, but it often does. In order for the error to disappear, the libraries must be re-imported with 'Full Clean'. 
http://traffic.libsyn.com/theamphour/TheAmpHour-435-AndreasSpiess.mp3  is blocked for Germany

steve6375, 2023-07-03:
I downloaded new zip file of current main and added into my existing project so all source files should be the same.
Then did a full clean and rebuilt.
Still crashing on various stations as before.
I did full clean twice and rebuild about 4 times.
????

schreibfaul1, 2023-07-03:
You can sync with the master. Or clone the repository via the link. Please never copy files from the zip into the project (the only exception are the files for the SD card)

steve6375, 2023-07-03:
I made a new git clone in PlatformIO and I still get exactly the same problem with crashing on Arabella and POP LOUNGE CAFE and the Speiss podcast URLS, etc.

valera39, 2023-07-03:
After the update, the problem disappeared, everything works
Thank you!

closed 2023-07-04 by valera39