schreibfaul1/ESP32-MiniWebRadio #222: 7 Minutes(!) after reboot Radio plays first, latest Firmware and ESP32dev (4M) without PSRAM, VS1053, ILI9341

TubaFonse, 2023-07-04:
Dear Wolle! 
I'm using ESP32-ESP32dev (4M) module without PSRAM in standard configuration: VS1053, ILI9341 since 6 months. I flashed the latest Firmware and updated my SD-Card. After reboot the radio plays the first song after 7 MINUTES, (with Firmware based on Version 2.4.0 after 12 Seconds, no dropouts!). Please help to get a reliable MiniWebRadio V2 again - many thanks! 
See the following trace from today, Time: 11:30 

Serial Monitor (nach FW-Update und Reset), es dauert 7 Minuten, bis das Radio spielt: 

--- Terminal on COM3 | 115200 8-N-1
E (1015) psram: PSRAM ID read error: 0xffffffff
[     3][W][esp32-hal-psram.c:71] psramInit(): PSRAM init failed!
ESP32 Chip: ESP32-D0WDQ6
Arduino Version: 2.0.9
ESP-IDF Version: 4.4.4
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
PSRAM not found! MiniWebRadio will not work properly!
00:00:00
00:00:00        ***************************
00:00:00        *     MiniWebRadio V2     *
00:00:00        ***************************
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9341
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 944.8 MB by 968.8 MB free
00:00:00 setup: ....  seek for stations.csv
00:00:00 setup: ....  stations.csv found
00:00:00 setup: ....  seek for WiFi networks
[   855][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: FRITZ!Box7490LF
[  7186][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[  7186][I][WiFiMulti.cpp:119] run(): [WIFI] 10 networks found
[  7187][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: C8:0E:14:0C:93:11 SSID: FRITZ!Box7490LF Channel: 6 (-39)
[  7235][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 202 - AUTH_FAIL
[  7313][I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
00:00:07 WiFI_info:   Connecting WiFi...
00:00:07 setup: ....  connected to FRITZ!Box7490LF, IP address is 192.168.178.35
00:00:07 ftpServer:   Buffers allocated: 4622 bytes
00:00:07 rtime_info:  Initializing SNTP
00:00:07 rtime_info:  Waiting for system time to be set... (1/10)
00:00:09 rtime_info:  Waiting for system time to be set... (2/10)
11:35:46 rtime_info:  11:35:46
11:35:46 audiotask:   is pinned to core 1
11:35:46 audiotask:   priority is 2
11:35:46 VS1053 chipID = 1279067256, version = 3
11:35:46 setup: ....  Number of saved stations: 48
11:35:46 setup: ....  current station number: 1
11:35:46 setup: ....  current volume: 12
11:35:46 setup: ....  last connected host:
11:35:46 setup: ....  connection timeout: 500 ms
11:35:46 setup: ....  connection timeout SSL: 2000 ms
11:35:46 action: ...  current volume is 12
11:35:46 action: ...  switch to station 1
11:35:46 AUDIO_info:  Connect to new host: "http://streams.br.de/bayern1obb_2.m3u"
11:35:48 AUDIO_info:  Song stopped incorrectly!
11:35:48 AUDIO_info:  after song stopped incorrectly
11:35:48 AUDIO_info:  REG            dec      bin               hex
11:35:48 AUDIO_info:  -----------  -------  ----------------  -------
11:35:48 AUDIO_info:  MODE          02056   0000100000001000   0808
11:35:48 AUDIO_info:  STATUS        00576   0000001001000000   0240
11:35:48 AUDIO_info:  BASS          00000   0000000000000000   0000
11:35:48 AUDIO_info:  CLOCKF        24576   0110000000000000   6000
11:35:48 AUDIO_info:  DECODE_TIME   00002   0000000000000010   0002
11:35:48 AUDIO_info:  AUDATA        44101   1010110001000101   AC45
11:35:48 AUDIO_info:  WRAM          00000   0000000000000000   0000
11:35:48 AUDIO_info:  WRAMADDR      07682   0001111000000010   1E02
11:35:48 AUDIO_info:  HDAT0         00000   0000000000000000   0000
11:35:48 AUDIO_info:  HDAT1         00000   0000000000000000   0000
11:35:48 AUDIO_info:  AIADDR        00000   0000000000000000   0000
11:35:48 AUDIO_info:  VOL           06425   0001100100011001   1919
11:35:48 AUDIO_info:  AICTRL0       00000   0000000000000000   0000
11:35:48 AUDIO_info:  AICTRL1       00000   0000000000000000   0000
11:35:48 AUDIO_info:  AICTRL2       00000   0000000000000000   0000
[... 424 lines cut ...]
11:40:32 AUDIO_info:  Song stopped correctly after 0 msec
11:40:32 AUDIO_info:  Connection has been established in 33 ms, free Heap: 115948 bytes
11:40:32 AUDIO_info:  redirect to new host "http://d111.rndfnk.com/ard/br/br1/obb/mp3/128/stream.mp3?cid=01FB9ME7567MEQ5G72EVEEVVCS&sid=2S6Oc4xNUAsFFzr6HsaBXadHiKH&token=9JcpE5StgXSX5otsLHeOgvMstsdmdeDsrD5ahjWcTag&tvf=0hvxKmG0bhdkMTExLnJuZGZuay5jb20"
11:40:32 AUDIO_info:  Connect to new host: "http://d111.rndfnk.com/ard/br/br1/obb/mp3/128/stream.mp3?cid=01FB9ME7567MEQ5G72EVEEVVCS&sid=2S6Oc4xNUAsFFzr6HsaBXadHiKH&token=9JcpE5StgXSX5otsLHeOgvMstsdmdeDsrD5ahjWcTag&tvf=0hvxKmG0bhdkMTExLnJuZGZuay5jb20"
11:40:32 AUDIO_info:  Song stopped correctly after 0 msec
11:40:33 AUDIO_info:  Connection has been established in 28 ms, free Heap: 115652 bytes
11:40:33 WiFI_info:   RSSI is -43 dB
11:40:33 bitRate:     128Kbit/s
11:40:33 icy-descr:   BAYERN 1 Oberbayern
11:40:33 AUDIO_info:  icy-name: BAYERN 1 Oberbayern
11:40:33 StationName: BAYERN 1 Oberbayern
11:40:33 lastURL: ..  http://d111.rndfnk.com/ard/br/br1/obb/mp3/128/stream.mp3?cid=01FB9ME7567MEQ5G72EVEEVVCS&sid=2S6Oc4xNUAsFFzr6HsaBXadHiKH&token=9JcpE5StgXSX5otsLHeOgvMstsdmdeDsrD5ahjWcTag&tvf=0hvxKmG0bhdkMTExLnJuZGZuay5jb20
11:40:33 DLNA_info:   SSDP search for media servers started
11:40:33 AUDIO_info:  stream ready
11:40:33 StreamTitle: Peter Kent: It's A Real Good Feeling
11:40:37 DLNA_info:   SSDP query discovered 2 media servers
11:40:37 DLNA_info:   connected successfully to 192.168.178.24:50001
11:40:37 DLNA_info:   connected successfully to 192.168.178.29:2869
11:40:37 ftpServer:   Ftp server waiting for connection on port 21
11:40:38 Audiocodec:  MP3
11:42:21 StreamTitle: Amy MacDonald: This Is The Life
11:45:41 StreamTitle: Udo Lindenberg: Horizont
11:46:13 StreamTitle: BAYERN 1 am Vormittag
11:46:50 StreamTitle: Udo Lindenberg: Horizont
11:49:19 StreamTitle: Roxette: Joyride

valera39, 2023-07-04:
[TubaFonse](https://github.com/TubaFonse), The latest firmware requires ESP32 with PSRAM https://github.com/schreibfaul1/ESP32-MiniWebRadio/blob/master/README.md

schreibfaul1, 2023-07-04:
Hallo TubaFonse,
das ist schon richtig, was valera39 hier schreibt. Wenn du einen externen DAC benutzt geht nichts ohne PSRAM. In deinem Fall hast du einen HW Dekoder und damit wird es gerade so funktionieren. 
Mein allererstes Radio hat auch nur einen ESP32 ohne PSRAM und läuft trotzdem. Aber das interne SRAM ist extrem knapp. Ich denke, du hast einen älteren VS1053, der nicht mit dem Userpatch kompatibel ist. Leider habe ich dieses Patch eingeschaltet (war in den Vorgängerversionen immer deaktiviert)
Nimm bitte die neueste Version und aktualisiere bitte auch den Inhalt der SD, sichere vorher die stations.csv damit deine Senderliste nicht überschrieben wird. Und jetzt das wichtige: in der vs1053_ext.cpp muss `LOAD_VS0153_PLUGIN`deaktiviert werden.
[image]

Im Projekt kann es die Datei doppelt geben, für ESP32 und ESP32-S3. Das wird die Problemursache sei. Ohne den Userpatch funktioniert das VU Meter leider nicht.

Überlege, ob du nicht lieber einen exteren DAC, z.B. PCM5102 einsetzen möchtest.

viele Grüße aus Berlin
Wolle

TubaFonse, 2023-07-04:
Hallo Wolle, erstmals herzlichen Dank für Deine vorbildliche Unterstützung mit Erklärungen! 
Ich war nur etwas irritiert von der Notwendigkeit des PSRAM, da meine Konfiguration mit einfachem ESP32 (ohne PSRAM) mit der aktuellen FW und gepatchtem VS1053 gerade so läuft (sogar das VU-Meter!). Jedoch erst nachdem der Controller nach ca. 7 Minuten ab Power-On und unzähligen dropouts in die Gänge kommt (siehe o.g. Trace) spielt das Radio. 
Heute ist der ESP32-S3-WROOM-1-N16R8 und PCM5102A angekommen, aber ich finde nicht das Menü bzw menuconfig, wo ich OCTAL einstellen muss, wo ist das im VisualStudio / PlatformIO zu finden? Dein Hinweis in platformio.ini: 
in menuconfig the type of PSRAM must be changed from QUAD to OCTAL!
;       component config ---> ESP32S3-Specific ---> SPI RAM config ---> MODE (QUAD/OCT)... ---> (x) Octal Mode PSRAM

viele Grüße vom Starnberger See
TubaFonse

schreibfaul1, 2023-07-05:
ja, das menuconfig ist etwas versteckt, aber auch das gibt es doppelt je nach Umgebung.
 
[image]

TubaFonse, 2023-07-09:
danke Wolle, endlich geschafft das PSRAM zu konfigurieren und dank Deiner professionellen Hilfe wieder sehr viel dazugelernt. 
Bin gerade dabei den Fehler einzugrenzen, weil der Zugriff auf die SD-Card mit meinem neuen ESP32-S3-WROOM-1-N16R8 leider  nicht funktioniert, bin fast am verzweifeln (d.h. bootet endlos, wenn die SD-Karte eingesteckt ist). 
Hat mein folgendes Problem evtl. mit dem Boot-Loop-Problem des ESP32-S3 zu tun? 
ece3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
[   981][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377991
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
[   985][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377991
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
[   985][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377991
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
[   985][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
ESP-ROM:esp32s3-20210327

P.S.: Ich warte noch Deine Antwort ab, werde dann aber dieses Issue schließen, weil es mit dem ursprünglichen Problem (siehe Titel) in diesem Chat nichts mehr zu tun hat. 
Schönen Sonntag und LG, 
TubaFonse

schreibfaul1, 2023-07-09:
Hallo @TubaFonse,
das sieht soweit gut aus, die Konfiguration ist okay. Aber du kommst nicht bis ins setup(), da ist vorher schon ein Problem. Ich hoffe, dein Board befindet sich noch auf einem Steckbrett. Ich vermute einen Fehler in der Außenbeschaltung. Wenn du das Board aus der Schaltung ziehst, müsste es bis zur SD_MMC laufen und erkennen, dass diese fehlt. Und das ohne WDT1 SYS_RST
Bei mir sieht es dann so aus:
[image]
Wenn du auch soweit kommst liegt der Fehler in der Beschaltung, andernfalls ist das Board defekt (glaube ich aber nicht) dann kannst du vielleicht runter vom 16MB auf die  8MB Version, vom Flash werden nur fast 4MB benutzt,

TubaFonse, 2023-07-19:
Problem gelöst mit aktueller Firmware.

closed 2023-07-19 by TubaFonse