schreibfaul1/ESP32-MiniWebRadio #223: help wanted

valera39, 2023-07-04:
Hello Wolle!
Knowing that you always answer thoroughly and in detail, I want to ask you a few questions that are of interest to me. I may have asked them before and some of them have not been answered, so
Please forgive me in advance, including the inaccurate translation in some places.
You can replace the "mute" button with a "stop" button on the main screen to stop the radio stream, and after about 30 seconds, display the clock on the screen. In this way, if there is no playback, the clock is displayed in full screen.
Is it possible to apply this image
[image]
 instead of the # in front of the station numb
[image]
er. I think that would be nice.
Do you have any successes and developments for the implementation of the list of stations on the display, or you have not dealt with this option?
I think it is extremely difficult or impossible to make DLNA or player mode control playback, rewind, random playback and so on from the display. Or am I wrong?
I would like to control the functions with the touch screen and remote control as much as possible.
Please explain how to prescribe another remote control. I tried to convert the remote control code in the IR.cpp file with a HEX calculator, but it did not work. What is the right way to do this,
to get the correct values, for example case 39015: ir_resp=0; break; //ZERO 0xFF0019. I would like to do a full on and off with a separate miniweb remote button, like I do with any household device.
Is it possible for the user himself, not having much programming skills, to move and change the VU metr on the display?
I very much hope that you will have no difficulty in answering my questions.
I tried to email you questions, but your box doesn't exist.
Please continue to support your wonderful project. You can always count on my help to make your radio even better!
Thank you!

Arne65, 2023-07-04:
Hello Valera

I think the following protocols are supported by arduino  ir

Currently included protocols: Denon / Sharp, JVC, LG / LG2, NEC / Onkyo / Apple, Panasonic / Kaseikyo, RC5, RC6, Samsung, Sony, (Pronto), BangOlufsen, BoseWave, Lego, Whynter, FAST, MagiQuest.<br/><br/><b>New: </b>Added FAST Protocol. Changed some function signatures. Improved handling of PULSE_DISTANCE + PULSE_WIDTH protocols.<br/><a href="https://github.com/Arduino-IRremote/Arduino-IRremote/blob/master/changelog.md">Release notes</a><br/>

look here

https://github.com/Arduino-IRremote/Arduino-IRremote

if your remote has one of these protocols it should work

Arne65, 2023-07-04:
is your calculation correct?

                   case 39015: ir_resp=0;  break; //ZERO   0x9867

valera39, 2023-07-04:
[Arne65](https://github.com/Arne6,5), Yes, this is an example of the original IR.cpp

valera39, 2023-07-04:
You can also add to the log output in PlatformIO the number of the current version?

schreibfaul1, 2023-07-04:
Hi @valera39 , I've made symbols for station number and sleep in time, but I'm not good at it. Your symbols seem better. The height can be max 30px and the width for both 68px. There is a constraint on bmp that the width must be a multiple of 4.

valera39, 2023-07-04:
Thank you so much! I would appreciate it if you could answer my questions and suggestions.

valera39, 2023-07-06:
It would be nice if you could shift the VU metr, like this
[image]

steve6375, 2023-07-06:
Why do you want the VU moved? Is it being partially hidden by the TFT screen bezel on your case?

I don't think an alarm clock symbol is correct. The timer is an 'off' timer not an alarm timer. It switches off the device after a preset time. Maybe something like:
[image]
and
[image]

valera39, 2023-07-06:
This picture is just the bottom of the screen.
When all the elements are on the same line at the bottom, it looks more organic.
You can put your own picture of the clock or an original one from the archive.

valera39, 2023-07-06:
Hello dear Wolle!
I still hope that you will answer my questions

closed 2024-02-08 by github-actions[bot]