schreibfaul1/ESP32-MiniWebRadio #224: Enhancement: Set Timezone via web UI and settings

steve6375, 2023-07-05:
Timezone is currently hard coded so it must be recompiled by the end user for their own timezone (not convenient if moving from east coast to west coast USA or sending to a relative). Could TZ be set via web UI?
Also loaded and saved in settings file on SD card so user can set it that way too?

steve6375, 2023-07-05:
Could provide link for timezone codes?
https://github.com/nayarsystems/posix_tz_db/blob/master/zones.csv
Or file in your GitHub project?

schreibfaul1, 2023-07-06:
there is now a copy of the csv file on SD_MMC. The time zone can be changed via the web interface. The changes are saved in the settings.json and remain until the settings.json is overwritten.
[image]
The entry for the time zone is deleted in common.h.

steve6375, 2023-07-06:
Looks great 😃. Nice job! I will try it out...

closed 2023-07-10 by steve6375