schreibfaul1/ESP32-MiniWebRadio #228: Change sleep/off timer button (moon and stars icon)  to match footer icon

steve6375, 2023-07-06:
[image]

This picture on this button should match the icon used in the footer. The current picture does not really convey that it sets an off timer.

valera39, 2023-07-06:
Take any picture that you like and replace the original in the archive. It's not hard.

steve6375, 2023-07-06:
Of course I know that, I was thinking of all other users. The button icon is really not intuitive.

schreibfaul1, 2023-07-06:
adjusted the symbols
[image]

steve6375, 2023-07-07:
I like the new icons - great!

One suggestion. When a sleep timer is set, could the zz.bmp icon change to red colour (to match red timer text)?

clear-sky, 2023-07-07:
my be an icon like this have better harmony with other icons (zz.bmp)
[image]

closed 2023-07-08 by steve6375