schreibfaul1/ESP32-MiniWebRadio #230: It would be nice if you could shift the VU metr, like this

valera39, 2023-07-06:
It would be nice if you could shift the VU metr, like this
[image]

_Originally posted by @valera39 in https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/223#issuecomment-1623273545_

clear-sky, 2023-07-07:
Hi Dear valera39!
it is not possible ! imagine if section 3 of your ip had 3 digits(for example 192.168.150.130)
unless you make icons to be closer to each other or use another font

valera39, 2023-07-17:
You can put the bitrate value in the top line, there is a lot of free space there. Then everything will fit in the bottom line, it will look great on one line.

steve6375, 2023-07-17:
Top header could contain day of week (Tue) and date (2023/07/17) and weather symbol (e.g. ☁️ or 🌞 or 🌨️ etc )

closed 2024-02-08 by github-actions[bot]