schreibfaul1/ESP32-MiniWebRadio #231: Crashes if settings.json does not contain Timezone entry

steve6375, 2023-07-07:
I did not update my Settings.json file on SD card because I wanted to keep my old settings but new version crashes due to empty timezone string - so setting timezone to initial value actually has no effect in main.cpp because it is always updated by SD card settings.json even if it is missing.

I fixed by adding lines at 1444 in main.cpp:
    if (_TZString == "") _TZString = "CET-1CEST,M3.5.0,M10.5.0/3";
    if (_TZName == "") _TZName = "Europe/Berlin";
    setRTC(_TZString.c_str());

Maybe it would it be better to modify the **defaultsettings** routine to ensure it always gets valid value for TZname and String values and all other values too in case bad or old settings.json file?

Also, the Timezone list has a blank line at the **beginning** and also one **end** of the list after Zulu
[image]

This causes settings.json to have blank contents (but does not crash)
"Timezone_Name":"&undefined","Timezone_String":""}

The last line can be fixed by editing csv file and **deleting last empty line**,
I don't know how to fix the first blank line problem :-(

schreibfaul1, 2023-07-07:
This is unpleasant if the program crashes due to an empty Timezone_String. There are now no more empty fields in the selection list. And the RTC module refuses strings of length 0

steve6375, 2023-07-08:
I tried a previous settings.json file (which I have not edited - it was written by rogue/corrupted code or due to SD card read errors maybe?) and it caused a crash.

This is the file I tested
[settings_BAD.zip](https://github.com/schreibfaul1/ESP32-MiniWebRadio/files/11990469/settings_BAD.zip)

{"volume":0,"ringvolume":0,"alarmtime":0,"alarm_weekdays":0,"timeAnnouncing":false,"mute":false,"brightness":0,"sleeptime":0,"lastconnectedhost":null,"station":0,"sumstations":26,"toneha":0,"tonehf":0,"tonela":0,"tonelf":0,"toneLP":0,"toneBP":0,"toneHP":0}

It got changed to this on first boot, but the esp32 still crashes and reboots
[settings_BAD2.zip](https://github.com/schreibfaul1/ESP32-MiniWebRadio/files/11990470/settings_BAD2.zip)

{"volume":0,"ringvolume":0,"alarmtime":0,"alarm_weekdays":0,"timeAnnouncing":false,"mute":false,"brightness":0,"sleeptime":0,"lastconnectedhost":"","station":0,"sumstations":26,"toneha":0,"tonehf":0,"tonela":0,"tonelf":0,"toneLP":0,"toneBP":0,"toneHP":0,"Timezone_Name":null,"Timezone_String":null}

I know it is a 'bad' settings file, but it really shouldn't cause a crash - the average end user will have no idea what is wrong and not know how to fix the settings.json file (e.g. delete it) even if he/she knew that was the problem! If any values in settings.json are invalid then code should cope with it.

```

[  7294][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[  7295][I][WiFiMulti.cpp:119] run(): [WIFI] 2 networks found
[  7296][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 84:16:F9:2B:6C:AF SSID: gertrude Channel: 4 (-32)
[  8090][I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
00:00:26 WiFI_info:   Connecting WiFi...
00:00:26 setup: ....  connected to gertrude, IP address is 192.168.1.107
00:00:26 ftpServer:   Buffers allocated: 4622 bytes
00:00:26 connection to NTP failed, trying again    
[  8123][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 8 - ASSOC_LEAVE
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344 
load:0x40078000,len:13924
ho 0 tail 12 room 4      
load:0x40080400,len:3600 
entry 0x400805f0
[   446][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled

ESP32 Chip: ESP32-D0WD-V3
Arduino Version: 2.0.9
ESP-IDF Version: 4.4.4
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
PSRAM total size: 8388608 bytes

00:00:27    
00:00:27        ***************************    
00:00:27        *     MiniWebRadio V2     * 
```

clear-sky, 2023-07-08:
Hi dear wolle!
design it like network.csv
radio must at first look for time zone in  sd card if not found go to TZNAME set up in common.h

schreibfaul1, 2023-07-09:
This shouldn't happen of course, setting a string to NULL is bad in this case.

steve6375, 2023-07-24:
I havent retested but assume fixed! thanks