schreibfaul1/ESP32-MiniWebRadio #232: Suggestion: Add Standby button

steve6375, 2023-07-08:
When the Radio is on, we can mute it but not turn it off.
Could you add a Standby button so we can tap screen and press Standby button?
This would put device into Sleep mode.
Then can be woken up by TFT touch or Alarm going off.
[image]

valera39, 2023-07-08:
I support it! And the ability to control this mode with the remote control

schreibfaul1, 2023-07-08:
It's integrated, I think Valera got the idea. If you press the mute button longer than 2.5s (button must be green) it goes into sleep mode

steve6375, 2023-07-08:
Oh yes, but how is a user supposed to know that! Most devices have an obvious Standby button.

valera39, 2023-07-10:
Hello Wolle!
Big request to make the transition to standby mode by remote control button, for example by long pressing the mute button.
Thank you!

schreibfaul1, 2023-07-10:
The idea is good. My IR Library is an in-house development and it's time to expand it for most NEC remote controls. Then many other remote controls (almost all) would be usable, such as from the video recorder or television...

steve6375, 2023-07-10:
I used an mp3 player remote
[image]

codes are (starting top left):

41565  25245  57885
8925 765 49725
57375 43095  36975
26775 39015 43135
12495 6375 31365
4335 14535 23205
17085 19125 21165

However I am having some issues with it.

Also when press a number button and get a large number (or two or 3) on TFT - the level meter vertical bars on right hand side of screen is destroyed.

steve6375, 2023-07-12:
IR bug fixed - see #241

steve6375, 2023-07-12:
@schreibfaul1 
I see you have modified IR just now so we can define different controller values by modifying code in main.cpp. 
I am not at all good at C++ but ..
Does this mean you could now put code in main.cpp permanently like

#if __has_include("IR_USER_KEYS.h") 
# include "IR_USER_KEYS.h"
#endif

If possible could you please put some lines like this in main.cpp so that all a user has to do is add their own IR_USER_KEYS.h file and then there is no need to modify any code at all in main.cpp. This means whenever a new version of MinIWebRadio is released, we simply need to add our ir key file and do not need to modify main.cpp each time.

This means you or other users can add various different files for different remotes and the user just needs to rename the correct one to IR_USER_KEYS.h and add it to the src folder.

steve6375, 2023-07-12:
P.S. Just another thought, it would be even easier if the code loaded the IR button values from a file on the SD card.
This means we would not have to recompile for different remotes and the same firmware could be used with a variety of different controllers simply by changing a file on the SD card.
Is this possible? If the file does not exist then just use the default IR codes.

steve6375, 2023-07-12:
I was looking at your new code (havent downloaded or tested it yet) but I saw this line which looked wrong to me as the array could be exceeded?

[image]

steve6375, 2023-07-13:
My MP3 remote has **21 buttons** which could be supported. Some have 60 buttons.

Would be nice to have a few new screens (states) - e.g.  add these three extra screen pages:
gain\freq\EQ screen - adjust gain/freq/vol whilst radio or media file is playing
Options screen - set Timezone/AP point from list/announce time/factory reset
DLNA screen. - select file

```
STANDBY/SLEEP - put into sleep mode immediately with power saving
MODE - cycle through screen - change to state=CLOCK screen - next press - change to ALARM screen - next press change to 
   set SLEEP - next press change to BRIGHTNESS - OPTIONS screen - next press RADIO screen.
MUTE - mute on - mute off
PLAY/PAUSE - stop music (radio or mp3) - start music
|<< and >>| - station up/down or mp3 file up/down
EQ - Show gain/freq sliders screen (allow changes whilst playing current radio or mp3 file
VOL-/VOL+ - adjust vol
RPT - if playing media files from SD, cycle through all files in current folder
U/SD - RADIO screen or MEDIA folder screen (or same as MODE button?)
```

OPTIONS screen allows selection of access point from list, announce time and TimeZone (and Factory Reset?).

User can also define additional buttons if you allowed for larger array of 60 buttons, user can define multiple buttons to have same affect, e.g.

```
STANDBY button = SLEEP button
HOME button = EXIT button = RADIO screen
RETURN button - previous screen
> (PLAY) button =  play current radio or media file
|| (PAUSE) button =  stop playing radio or pause media file
[] (STOP) button = stop radio or stop media file
AUDIO button - select mp3 media play screen
GUIDE and i+ info button and OPT+ button - select OPTIONS screen
DIGITAL/ANALOGUE button - RADIO/MEDIA screen toggle
SOURCE - same as MODE
```

red/green/yellow/blue buttons - select screen RADIO/MEDIA/DLNA/OPTIONS screens.

schreibfaul1, 2023-07-13:
Yes, that's right, some remote controls have a lot of buttons, but no more than 256. Each button is assigned a command code, from 0x00 to 0xFF. In addition, each remote control has an address, also from 0x00 to 0xFF, which applies to all buttons.
My idea is to build a website where the buttons can be assigned a function. In this way, many remote controls that use the NEC protocol can be used.

steve6375, 2023-07-13:
OK - and save those settings to a file on the SD card please (e.g. \IR_CODES.csv which is automatically read in by firmware)?
Then the file can be kept on SD card even if we update the firmware/rebuild, etc.
Also can keep files for all types of controllers in a folder on SD card so user can pick a readymade one from a list?

steve6375, 2023-07-13:
P.S. just tried latest version - you still have IR.cpp bug - my controller does not work  properly - see #241

steve6375, 2023-07-13:
Also latest code crashes when I use my controller. Can you try different controllers so generate unrecognised codes and see if crashes?

maybe because of unrecognised code and 256 array bug I mentioned before?

```
0:47:33 AUDIO_info:  Connect to new host: "http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_one/bbc_radio_one.isml/bbc_radio_one-audio=128000.norewind.m3u8"
10:47:33 AUDIO_info:  Song stopped correctly after 40 msec
10:47:33 AUDIO_info:  Connection has been established in 34 ms, free Heap: 146676 bytes
[ 81650][I][IR.cpp:179] isr_IR(): ir_value 18e700ff
10:47:33 AUDIO_info:  new request: "http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_one/bbc_radio_one.isml/bbc_radio_one-audio=128000-263944009.ts"
10:47:33 ir_code: ..  IR address 0x00, IR command 0x18
10:47:34 Audiocodec:  AAC
10:47:34 AUDIO_info:  new request: "http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_one/bbc_radio_one.isml/bbc_radio_one-audio=128000-263944010.ts"
10:47:35 AUDIO_info:  new request: "http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_one/bbc_radio_one.isml/bbc_radio_one-audio=128000-263944011.ts"
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Core  1 register dump:
PC      : 0x40092622  PS      : 0x00060335  A0      : 0x8009140a  A1      : 0x3ffb18cc  
A2      : 0x3ffc142c  A3      : 0x3ffc6e4c  A4      : 0x00000004  A5      : 0x00060323  
A6      : 0x00060323  A7      : 0x00000001  A8      : 0x3ffc6e4c  A9      : 0x00000018  
A10     : 0x3ffc6e4c  A11     : 0x00000018  A12     : 0x3ffb9974  A13     : 0x00060323  
A14     : 0x007b1a98  A15     : 0x003fffff  SAR     : 0x00000020  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x4008cb79  LEND    : 0x4008cb89  LCOUNT  : 0xfffffffe  
Core  1 was running in ISR context:
EPC1    : 0x40119d2f  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000

Backtrace: 0x4009261f:0x3ffb18cc |<-CORRUPTED

  #0  0x4009261f:0x3ffb18cc in vListInsert at C:\Users\Steve\.platformio\packages\framework-espidf\components\freertos/list.c:183

Core  0 register dump:
PC      : 0x400927bb  PS      : 0x00060035  A0      : 0x80091033  A1      : 0x3ffb11bc
A2      : 0x3ffb1a98  A3      : 0xb33fffff  A4      : 0x0000abab  A5      : 0x00060023
A6      : 0x00060021  A7      : 0x0000cdcd  A8      : 0x0000abab  A9      : 0xffffffff
A10     : 0x3ffb950c  A11     : 0x00000000  A12     : 0x3ffb9508  A13     : 0x00000007
A14     : 0x007b1a98  A15     : 0x003fffff  SAR     : 0x0000001d  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x400927b8:0x3ffb11bc |<-CORRUPTED

  #0  0x400927b8:0x3ffb11bc in compare_and_set_native at C:\Users\Steve\.platformio\packages\framework-espidf\components\esp_hw_support\include/soc/compare_set.h:25
      (inlined by) spinlock_acquire at C:\Users\Steve\.platformio\packages\framework-espidf\components\esp_hw_support\include/soc/spinlock.h:103
      (inlined by) xPortEnterCriticalTimeout at C:\Users\Steve\.platformio\packages\framework-espidf\components\freertos\port\xtensa/port.c:301

ELF file SHA256: 8cfd1144ba3912ed

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
```

steve6375, 2023-07-13:
Toadys code - bug in IR table

'o'  should be 'k'

in both main.cpp commented code and ir.cpp

???

steve6375, 2023-07-13:
can ir_buttons array be increased? My controller has 21 buttons.
Others may have more - so say array of 40 should be enough???
This allows for different buttons having same function
Also add note in main,cpp that array size must not be changed (as fixed size array in ir.cpp)?

steve6375, 2023-07-13:
I found cause of crashing in ir.cpp - it is due to the i_log line inside interrupt routine !!
[image]

Please comment this out.

Also, should this line be uint32_t  not int32_t or doesn't it matter ???
[image]

steve6375, 2023-07-13:
P.S. How will the user know what codes are generated by his/her own IR remote controller?
Will you output codes to TFT screen or to Web UI.
We can't expect the end user to have a USB Terminal running.

steve6375, 2023-07-14:
I modified main.cpp to show IR remote key codes when CLOCK is shown

```
void ir_code(const char *adr, const char* cmd, uint8_t address, uint8_t command){
    SerialPrintfln("ir_code: ..  " ANSI_ESC_YELLOW "IR address " ANSI_ESC_BLUE "%s, "
                                   ANSI_ESC_YELLOW "IR command " ANSI_ESC_BLUE "%s", adr, cmd);
    if(_state != CLOCK) return;
    tft.setFont(_fonts[1]);
    tft.setTextColor(TFT_GOLD);
    clearIPaddr();
    tft.setCursor(_winIPaddr.x + 6 , _winIPaddr.y + 2);
    tft.printf("IR:a=%s c=%s",adr,cmd);   // show adr and cmd for 1/2 second
    vTaskDelay(500);
    showFooterIPaddr();
}

```
This allows a user to see IR codes without needing terminal and USB cable.

I added my code (one button is missing)
```

     irBtn_t ir_buttons[20];
     ir_buttons[ 0].val = 0x16; ir_buttons[ 0].ch = '0';
     ir_buttons[ 1].val = 0x0c; ir_buttons[ 1].ch = '1';
     ir_buttons[ 2].val = 0x18; ir_buttons[ 2].ch = '2';
     ir_buttons[ 3].val = 0x5e; ir_buttons[ 3].ch = '3';
     ir_buttons[ 4].val = 0x08; ir_buttons[ 4].ch = '4';
     ir_buttons[ 5].val = 0x1c; ir_buttons[ 5].ch = '5';
     ir_buttons[ 6].val = 0x5a; ir_buttons[ 6].ch = '6';
     ir_buttons[ 7].val = 0x42; ir_buttons[ 7].ch = '7';
     ir_buttons[ 8].val = 0x52; ir_buttons[ 8].ch = '8';
     ir_buttons[ 9].val = 0x4a; ir_buttons[ 9].ch = '9';
     ir_buttons[10].val = 0x46; ir_buttons[10].ch = 'k';  // OK  MODE
     ir_buttons[11].val = 0x43; ir_buttons[11].ch = 'u';  // UP
     ir_buttons[12].val = 0x40; ir_buttons[12].ch = 'd';  // DOWN
     ir_buttons[13].val = 0x09; ir_buttons[13].ch = 'r';  // RIGHT
     ir_buttons[14].val = 0x15; ir_buttons[14].ch = 'l';  // LEFT
     ir_buttons[15].val = 0x47; ir_buttons[15].ch = '#';  // #  MUTE
     ir_buttons[16].val = 0x45; ir_buttons[16].ch = '*';  // * 
     ir_buttons[17].val = 0x07; ir_buttons[17].ch = 'k';  // EQ
     ir_buttons[18].val = 0x44; ir_buttons[18].ch = '#';  // Play/Pause
     ir_buttons[19].val = 0x0D; ir_buttons[19].ch = 'k';  // U/SD
//     ir_buttons[20].val = 0x19; ir_buttons[20].ch = '#';  // RPT
    ir.set_irButtons(ir_buttons);
    ir.set_irAddress(0x00);
	
```

steve6375, 2023-07-14:
I just tried the new July 14 release - good job! 

However, my remote will not work unless I change the order in ir.cpp so it firsts checks for 1 or 0 pulse and then checks for 32 pulses. i.e.
```

    if(ir_begin==false) return;
    levelcounter++;
    if(levelcounter%2==1)return;                // only falling edge can pass

    if((intval > 400) && (intval < 750)){       // Short pulse?
        ir_value += bit;                        // Count number of received bits
        pulsecounter++;
        bit <<= 1;
    }
    else if((intval > 1400) && (intval < 1900)){     // Long pulse?
        pulsecounter++;
        bit <<= 1;
    }
    else{
        pulsecounter = 0;
        bit = 1;
    }
    
    if(pulsecounter==32){

        ir_begin=false;
        userCode = ir_value & 0xFFFF;
        uint8_t a, b, c, d;
        a = (userCode & 0xFF00) >> 8;
        b = (userCode & 0x00FF);
        if(a + b == 0xFF){
            dataCode=(ir_value & 0xFFFF0000) >> 16;
            c = (dataCode & 0xFF00) >> 8;
            d = (dataCode & 0x00FF);
            if(c + d == 0xFF){
                ir.setIRresult(a, c);
            }
        }
    }

}
//----------------------------------------------------------------------------------------------------------------------------------------------------

```

I think if you changed the code it would still work for both your controller and mine?

schreibfaul1, 2023-07-14:
Changed it, makes no difference to me. Maybe your remote doesn't have a repeat sequence?
I'll take a look at your other suggestions.

steve6375, 2023-07-14:
On the scope it looks OK - same as text book!
It sends 32 bits and a stop bit
then after a while it sends a REPEAT packet (not a duplicate but the special repeat packet) ONLY if you hold down the button
but if you just press quick and release quick - only the 32/33 bit packet is sent.
[image]

You code seems to rely on a 2nd packet that must follow the first packet - but my controller only sends one packet (if I dont hold down the button).

schreibfaul1, 2023-07-14:
The main thing is that it works, waiting for the 'repeat code' which is also not intended.

steve6375, 2023-07-14:
@Wolle - I notice that for the last few days the VU meter bars are not working.
It seems to compile without error but the

steve6375, 2023-07-15:
I bought an arduino remote controller yesteday from Amazon and it just came this morning (same as yours) to test.
The outputs of my mp3 controller and the arduino controller are identical (as on my scope) and they behave similarly too though the arduino remote seems to generate REPEAT packets slightly more often.
I suspect because you do not have an IR indicator LED, if you give a short button press you may think it did not send a packet and so you press it again (or always give a 1/2 second press-and-hold on the button). But actually it is relying on the REPEAT packet to make the pulsecounter go to 32.
I can see my IR red LED Rx indicator on my shield and it is easy to see if a single packet has been received or if it is followed by REPEAT packets because then it will give multiple flashes. Only multiple flashes make it work correctly with your IR.cpp code.

You can test this by pressing button 1 **very quickly**  and then press button 2 very quickly - then press button 3 very quickly and wait 2 seconds (the TFT should always show '123') - then repeat presses 1 -2 -3, etc. Keep repeating this and you should find that at some point it will go wrong - e.g. when you press button '1' the TFT display will print '3'  and sometimes when you press button '2' the TFT display will print '1', etc.

If you the code is changed to test for pulsecounter==32  **AFTER** testing for a 1 or 0 pulse and incrementing the pulsecounter, then you won't get this problem. It will always give the correct code even on very short button presses.

steve6375, 2023-07-15:
Just for your info:
I can put esp32 into deep sleep (approx 150uA only) and screen+VS1053+SD card draws **45mA** only.
It can be woken (esp32 starts as if rebooting) by pressing TP screen (or by any IR press).
https://mischianti.org/2021/03/23/esp32-practical-power-saving-preserve-gpio-status-external-and-ulp-wake-up-5/
Here is code I used  at the end of the fall_asleep function just for test.

```

  /*
  First we configure the wake up source
  We set our ESP32 to wake up for an external trigger.
   Note that using internal pullups/pulldowns also requires
  RTC peripherals to be turned on.
  */

// esp_sleep_enable_ext0_wakeup(GPIO_NUM_35,LOW);  // for IR wakeup
 esp_sleep_enable_ext0_wakeup(GPIO_NUM_39,LOW);    // for TP wakeup
//Go to sleep now
  Serial2.println("Going to sleep now");
  vTaskDelay(3000);            //delay for user finger off TP screen or else causes trigger
  esp_deep_sleep_start();   // esp32 will restart (boot) when it next wakes up

```

valera39, 2023-07-16:
On my standard arduino remote, long pressing the mute button does not work correctly. Maybe the stand by mode should be put on a separate button for remotes with a large number of buttons?

schreibfaul1, 2023-07-16:
This can be, as Steve wrote, the 'repeat code' can be delayed. With my remote controls, the 'Repeat Code' is repeated about 20 times in 2 seconds. If 15 repetitions are detected, it is a long key press.
I will adjust the code again, then it will work better.

steve6375, 2023-07-16:
Both my mp3 controller and arduino controller work with latest download for me.
Takes about 2 seconds of holding down. Same button (quick press) wakes.
code is  if(ir_rc > 14){

steve6375, 2023-07-16:
If still not working maybe check battery in remote is OK?

valera39, 2023-07-16:
[steve6375](https://github.com/steve6375) The battery is new, all commands go through fine, not always double tap works, have to press many times to get standby to work.

steve6375, 2023-07-16:
why do you mention double tap?
To go into sleep you hold down (press down) the # key and keep it held down for 2-3 seconds.

valera39, 2023-07-16:
I misspoke, I apologize. When pressing for a long time, switching to stasnd by does not happen,or happens, but not always.

valera39, 2023-07-17:
after the last update, the radio does not come out of stand by mode
log part:
17:59:00 AUDIO_info:  Connect to new host: "http://harddanceradio.ddns.is74.ru:8000/eurodisco.m3u"
[ 30702][V][ssl_client.cpp:321] stop_ssl_socket(): Cleaning SSL connection.
17:59:00 AUDIO_info:  buffers freed, free Heap: 167412 bytes
17:59:00 AUDIO_info:  Connection has been established in 80 ms, free Heap: 166648 bytes
[ 30864][D][WiFiClient.cpp:546] connected(): Disconnected: RES: 0, ERR: 128
17:59:00 AUDIO_info:  Connect to new host: "http://harddanceradio.ddns.is74.ru:8000/eurodisco"
[ 30877][V][ssl_client.cpp:321] stop_ssl_socket(): Cleaning SSL connection.
17:59:00 AUDIO_info:  buffers freed, free Heap: 167024 bytes
17:59:00 AUDIO_info:  Connection has been established in 78 ms, free Heap: 166688 bytes
17:59:00 bitRate:     192Kbit/s
17:59:00 icy-descr:   Eurodisco Syntpop Italodisco 80s 90s
17:59:00 AUDIO_info:  icy-name: EURODISCO '80 [HDR]
17:59:00 StationName: EURODISCO '80 [HDR]
17:59:00 icy-url: ..  http://harddanceradio.ddns.is74.ru:8000
[ 31085][D][aac_decoder.cpp:1679] AACDecoder_AllocateBuffers(): AAC Spectral Band Replication enabled, 50788 additional bytes allocated
17:59:00 AUDIO_info:  AACDecoder has been initialized, free Heap: 165356 bytes , free stack 2616 DWORDs
17:59:00 lastURL: ..  http://harddanceradio.ddns.is74.ru:8000/eurodisco
17:59:00 AUDIO_info:  stream ready
17:59:00 AUDIO_info:  syncword found at pos 538
17:59:00 AUDIO_info:  syncword found at pos 0
17:59:00 AUDIO_info:  Channels: 2
17:59:00 AUDIO_info:  SampleRate: 44100
17:59:00 AUDIO_info:  BitsPerSample: 16
17:59:00 AUDIO_info:  BitRate: 182601
17:59:00 AUDIO_info:  AAC HeaderFormat: ADTS
17:59:00 AUDIO_info:  AAC Codec: MPEG-4 LowComplexity
17:59:00 AUDIO_info:  VBR recognized, audioFileDuration is estimated
17:59:00 StreamTitle: Joy - Chinese girls [3]
17:59:01 Audiocodec:  AAC
17:59:07 StreamTitle: Joy - Chinese girls [4]
17:59:11 ir_code: ..  IR address 0x00, IR command 0x16
[ 44348][I][main.cpp:2757] ir_long_key(): long pressed key nr: 10
[ 44949][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: ASUS_AC66U, BSSID: 2c:fd:a1:c2:44:f0, Reason: 8
[ 44950][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 44951][V][WiFiGeneric.cpp:343] _arduino_event_cb(): STA Stopped
[ 44959][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 8 - ASSOC_LEAVE
[ 44973][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 3 - STA_STOP
[ 45284][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 6 = 80 Mhz, APB: 80000000 Hz
17:59:14 falling asleep
17:59:20 ir_code: ..  IR address 0x00, IR command 0x16
17:59:20 awake
[ 52283][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[ 52298][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 0 - WIFI_READY
[ 52300][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
[ 52302][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 2 - STA_START
[ 53816][V][WiFiGeneric.cpp:383] _arduino_event_cb(): SCAN Done: ID: 129, Status: 0, Results: 4
[ 53816][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[ 53823][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[ 53828][I][WiFiMulti.cpp:119] run(): [WIFI] 4 networks found
[ 53835][D][WiFiMulti.cpp:151] run():        0: [4][04:D9:F5:56:3C:68] TP-LINK_E345E0_RPT (-61) *
[ 53844][D][WiFiMulti.cpp:151] run():        1: [11][1C:AF:F7:B3:8E:CB] ASUS_AC66U_REPITER (-64) *
[ 53854][D][WiFiMulti.cpp:151] run():        2: [4][30:B5:C2:E3:45:E0] TP-LINK_E345E0 (-82) *
[ 53863][D][WiFiMulti.cpp:151] run():        3: [4][D8:07:B6:B7:FE:0F] GG (-85) *
[ 53871][E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
17:59:23 START WIFI
17:59:23 .
17:59:23 .
17:59:23 .
17:59:23 .
17:59:23 .
17:59:23 .
17:59:24 .
17:59:24 .
17:59:24 .
17:59:24 .
17:59:24 .
17:59:24 .
17:59:24 .
17:59:24 .
17:59:24 .
17:59:24 .
17:59:25 .
17:59:25 .
17:59:25 .
17:59:25 .

schreibfaul1, 2023-07-17:
During sleep, the WiFi turns off, but when it wake up, your network is not found.
Isn't it one of these?
[image]

valera39, 2023-07-17:
my network isn't on the list for some reason. SSID: ASUS_AC66U,

valera39, 2023-07-17:
stand by mode and exiting it suddenly started working normally. I don't know why it didn't work earlier

valera39, 2023-07-17:
and still, putting stand by on a separate button would be a better solution for reliability.

valera39, 2023-07-17:
What if in stand by mode only the clock is displayed and the time is taken from the RTC module DS3231 for example?

valera39, 2023-07-20:
Is it possible to do something like this?

closed 2024-02-07 by github-actions[bot]