schreibfaul1/ESP32-MiniWebRadio #233: Sleep mode does not turn off WiFi

steve6375, 2023-07-08:
When in sleep mode (e.g. after setting 5 minute timer) the TFT screen is off but WiFi is still working and Web UI still works.
Is it really sleeping?

schreibfaul1, 2023-07-08:
No, actually it just mutes and the screen goes off. Would it make sense to switch off WiFi in battery operation?

steve6375, 2023-07-08:
I guess IR sensor and TP still need to be active too.
Currently my Wrover+PSRAM+VS1053+ILI9341 takes approx -**306mA** via USB port at 4.95V at full brightness or **210mA** at very low brightness.
After sleep timer triggers it takes **195mA**.
If I hold Mute button it takes **181mA**  - not sure why it is lower than sleep timer???

So would last about 5.5 hours with 1000mA/h Li battery in sleep mode.

When in 'sleep mode', I can use Web UI to select and play a station (unmute) but screen is still off which seems a little odd too.

If turning off wifi or reducing wifi power whilst in sleep mode helps to reduce current then it would be worth it.

It would be nice to save more power. e.g. just run loop for a 1/4 second once every 5 seconds so need to touch TFT or press remote for 5 seconds max to wake it up?? 

If we have a MiniWebRadio with mains power supply + Li battery, we don't want to turn off the power because then battery would stop charging.

steve6375, 2023-07-09:
I added some WiFi disable code and CPU frequency code to wake and sleep functions.
I do not really know what I am doing however!!! I read this...
https://docs.espressif.com/projects/espressif-esp-faq/en/latest/software-framework/wifi.html
https://mischianti.org/2021/03/06/esp32-practical-power-saving-manage-wifi-and-cpu-1/

Now  Sleep = **52mA**    at 10Mhz !!!
Touch wakes up display (but have not tested IR sensor)

If do not change frequency (keep at 240MHz) then sleep current with WiFi off = 100mA.

```
void fall_asleep(){
    xSemaphoreTake(mutex_display, portMAX_DELAY);
    _f_sleeping = true;
    audioStopSong();
    if(_state != CLOCK){
        clearAll();
        setTFTbrightness(0);
    }

// NEW CODE
    WiFi.disconnect(true);  // Disconnect from the network
    WiFi.mode(WIFI_OFF);    // Switch WiFi off
    setCpuFrequencyMhz(10);

    SerialPrintfln("falling asleep");
    xSemaphoreGive(mutex_display);
}

void wake_up(){
    if(_f_sleeping == true){ //awake
        _f_sleeping = false;
        SerialPrintfln("awake");
        setTFTbrightness(_brightness);
        
       // NEW CODE
       setCpuFrequencyMhz(240);
        WiFi.disconnect(false);  // Reconnect the network
        WiFi.mode(WIFI_STA);    // Switch WiFi off
        wifiMulti.run();
        SerialPrintfln("START WIFI");
        while (WiFi.status() != WL_CONNECTED) {
            delay(100);
            SerialPrintfln(".");
        }
        SerialPrintfln("WiFi connected");
        strcpy(_myIP, WiFi.localIP().toString().c_str());
        SerialPrintfln("WiFi wake: Connected to " ANSI_ESC_CYAN "%s" ANSI_ESC_WHITE
                   ", IP address is " ANSI_ESC_CYAN "%s", WiFi.SSID().c_str(), _myIP);

        changeState(RADIO);
        showVUmeter();
        connecttohost(_lastconnectedhost.c_str());
        showLogoAndStationName();
        showFooter();
        showHeadlineItem(RADIO);
        showHeadlineVolume();
        _f_mute = true;
        mute();
    }
}
```

schreibfaul1, 2023-07-14:
If you operate the radio with a battery, you should not only switch off WiFi but also the display via a MOSFET. And then go into 'deep sleep' mode. But I don't know enough about that. The people from the ESPuino project do that.

steve6375, 2023-07-14:
Sure, but above code can easily be added to existing sleep/wake function and it will save 100mA power when in sleep mode (on battery or on mains supply - it saves $$$ in energy).
esp32 board consumes most power anyway.
The code above seems OK and esp32 can still be woken by touch or IR remote.
ir_key routine now also needs to call wake_up()  as well
```
void ir_key(const char* key){
    if(_f_sleeping == true && key[0] != 'k') return;
    if(_f_sleeping == true && key[0] == 'k'){ //awake
        wake_up();
etc.

```

closed 2023-07-20 by steve6375

steve6375, 2023-07-21:
P.S. Found this video which explains why 80MHz is best low speed
https://youtu.be/JFDiqPHw3Vc