schreibfaul1/ESP32-MiniWebRadio #239: Web UI - Media files - downloading any jpg icon file always gives smurf icon

steve6375, 2023-07-10:
When using Web UI - Media - I try to download jpg icons, but all I get is icon of blue smirf when opened.

[image]

[image]
[image]
[image]
[image]

schreibfaul1, 2023-07-10:
I have that problem too. Only image files seem to be affected. Images are treated differently by the web browser, I don't have a solution for this yet.

schreibfaul1, 2023-07-10:
This was not a problem with the web browser. Images are often duplicated, each for small and medium-sized displays. And additionally there was an error in the SD path. If nothing was found on the SD, the smurf came.

steve6375, 2023-07-13:
Still getting the pesky smurf !
here is browsedr F12 console
[image]
and terminal o/p
[image]

closed 2023-07-13 by steve6375