schreibfaul1/ESP32-MiniWebRadio #240: DLNA control

valera39, 2023-07-10:
Hello Wolle!
On the web page, DLNA tab, there are no control buttons, so only one track can be selected for playback. How do I stop playback? When I switch to the radio tab, playback continues from DLNA. What if I want to listen to the whole folder?
How do I listen to music from my home media server using a playlist? I made a playlist:
#EXTM3U
#EXTINF:-1, Brothers On The 4th Floor - Dance with Me
\\RT-AC66U_B1-44F\sda5\музыка\2 Brothers on the 4th floor\2 Brothers On The 4Th Floor - Dance With Me.mp3
but it's not playing
[ 28063][I][main.cpp:1110] getNextAudioFile(): /audiofiles/Default.m3u
[ 28063][I][main.cpp:1116] getNextAudioFile(): /audiofiles/Default.m3u
16:53:15 Playlist:    #EXTM3U
16:53:15 Playlist:    #EXTINF:-1, Brothers On The 4th Floor - Dance with Me
16:53:15 Playlist:    Title:  Brothers On The 4th Floor - Dance with Me
16:53:16 Playlist:    playtime: -1s
[ 31269][W][main.cpp:1176] processPlaylist(): pls path /audiofiles/\\RT-AC66U_B1-44F\sda5\музыка\2 Brothers on the 4th floor\2 Brothers On The 4Th Floor - Dance With Me.mp3, 11, 105
16:53:16 Playlist:    /audiofiles/\\RT-AC66U_B1-44F\sda5\музыка\2 Brothers on the 4th floor\2 Brothers On The 4Th Floor - Dance 
With Me.mp3
[ 31333][V][ssl_client.cpp:321] stop_ssl_socket(): Cleaning SSL connection.
16:53:16 AUDIO_info:  buffers freed, free Heap: 152060 bytes
16:53:16 AUDIO_info:  Reading file: "/audiofiles/\\RT-AC66U_B1-44F\sda5\музыка\2 Brothers on the 4th floor\2 Brothers On The 4Th Floor - Dance With Me.mp3"
16:53:16 AUDIO_info:  Failed to open file for reading
16:53:16 AUDIO_info:  next playlist file
16:53:16 end of playlist
[ 32539][D][WiFiClient.cpp:546] connected(): Disconnected: RES: -1, ERR: 104
[ 32544][D][WiFiClient.cpp:546] connected(): Disconnected: RES: 0, ERR: 128

schreibfaul1, 2023-07-24:
Hallo valera39,
in den nächsten Tagen möchte ich die DLNA Seite neu gestalten. Das soll ähnlich der audio Seite mit  einem Verzeichnisbaum werden. Dann ist viel mehr möglich.

valera39, 2023-07-24:
All right, I'll be waiting.

closed 2024-02-07 by github-actions[bot]