schreibfaul1/ESP32-MiniWebRadio #241: Possible logic bug in IR.cpp

steve6375, 2023-07-11:
In IR.cpp it tests for completed packet (pulsecounter=32)  before it actually gets the last bit and increments pulsecounter to 32.
This means that a second packet is needed and then the first packet is recognised and sent to main.cpp

The order needs changing
first check for 0 or 1 pulse
then test if pulsecounter = 32  as we have complete packet.

I tested this and now seems to work much better.

What do you think ???

```

    if(levelcounter%2==1)return;                // only falling edge can pass

// my remote = 640 or 1700  - test for bit 0 or 1 first  ZZZZZZZZZZZZZZZZZZ

    if((intval > 400) && (intval < 750)){       // Short pulse?
        ir_value=(ir_value << 1) + 0;           // Shift in a "zero" bit
        pulsecounter++;                         // Count number of received bits
    }
    if((intval > 1400) && (intval < 1900)){     // Long pulse?
        ir_value=(ir_value << 1) + 1;           // Shift in a "one" bit
        pulsecounter++;                         // Count number of received bits
    }

// now get complete data   ZZZZZZZZZZ

    if(pulsecounter==32){          
        ir_begin=false;
        pulsecounter = 0;
        ir_resp=(-1);
        address=(ir_value&0xFFFF0000)>>16;
        command=(ir_value&0x0000FFFF);
        ir_value=0;
        if(address==0x00FF){

```

steve6375, 2023-07-11:
P.S. I also added a line which aborts if a pulse is missed or too long.  See ZZZ line

```
    if ( (ir_begin) && intval > 9000) ir_begin=false;   // added in case miss change, etc. ZZZZZZZ
    if(ir_begin==false) return;
    levelcounter++;
    if(levelcounter%2==1)return;                // only falling edge can pass

```

steve6375, 2023-07-11:
At the moment user cannot see remote key code numbers without adding terminal output debug code (which can cause crashes and timing issues). So difficult to use different bespoke remotes.
It would be nice if the remote's keypad number was always output to the terminal so user can see the codes generated.
Also if user can define key codes within extra user file ir_user.h file (present in project as sample_ir_user.h) and include this by using (In pseudo-code)...

if exist ir_user.h then load ir_user.h

so no change needed to source files and user just adds his own ir_user.h file by renaming and changing sample_ir_user.h file to ir_user.h?

Is something like this possible?

steve6375, 2023-07-13:
Loading remote button codes from a file on SD card would be a much better solution.
Then can have different files for different remotes inside zip file and user can select correct one for their remote and copy it to root of SD card to be loaded by firmware.

clear-sky, 2023-07-13:
it is much better to use external ir lib like irmp
in this condition user can use his own remote (all protocols can be chosen)

closed 2023-07-19 by steve6375