schreibfaul1/ESP32-MiniWebRadio #248: IR remote set via web interface

clear-sky, 2023-07-14:
Hi dear wolle!
this is a miracle!!!
very nice work!
but i have problem with it can not set my nec remote via web interface 
when i press remote keys and click on each 19 box nothing shown
when i edit each box by hand and press save in terminal i can see data is saved in NVS but nothing really is saved
when i press load botton in web interface all box is 0x00
my remote address is 0xFD01 (extracted from IRMP) > https://github.com/IRMP-org/IRMP
in your old lib  this address did not work i used first 4 byte of raw data in this case 0x80BF and second 4 byte of row data used for command 
for example raw data on key 1 of my remote was 0x80BF49B6 (0x80BF is address and command is 0x49b6)

schreibfaul1, 2023-07-14:
I have that too with one of my remote controls. According to the NEC protocol, the address consists of 8 bits that are repeated inverted for control. So 0xFD01 is invalid, it should be called 0xFD02 or 0xFE01.
I thought mine was broken, but there are probably deviations from the protocol.
 I will have to change something.

schreibfaul1, 2023-07-14:
There is an extended protocol
[image]

Now your remote control will also work properly.

clear-sky, 2023-07-14:
Hi Dear Wolle!
Thanks for your great work!
problem solved!

schreibfaul1, 2023-07-14:
thanks, i forgot that

steve6375, 2023-07-15:
Maybe add
(long press = SLEEP)

[image]

steve6375, 2023-07-15:
P.S. Can you test wake from sleep using IR button? Doesn't seem to work? Can use station button to start playing radio from sleep but screen is off...