schreibfaul1/ESP32-MiniWebRadio #249: ESP32-S3-WROOM-1-N16R8 “Boot Loop Problem”, only when SD-Card is connected

TubaFonse, 2023-07-15:
Dear Users and Tester of ESP32-S3-WROOM-1-N16R8 module, I need your help on following issue: 

with the latest ESP32-MiniWebRadio V2 Firmware (14.07.2023) used, also actual content on the SD-Cards: 

1.)	I tested 7 different SD-Cards (1GB … 4GB FAT up to 32GB FAT32) in SD-Card adapter (on ILI9341) with short wires to GPIO2, GPIO14, GPIO15 of ESP32-DevKitC_V4 (ESP32-WROOM-32), 
TFT&TP: ILI9341, Decoder: VS1053 (according to common.h CONFIG_IDF_TARGET_ESP32). 
Result: Access to SD-Card without problems, ESP32-MiniWebRadio V2 plays! 
2.)	I tested the same 7 SD-Cards in SD-Card adapter (on ILI9488) with short wires to GPIO11, GPIO13, GPIO14 of ESP32-S3-WROOM-1-N16R8 
TFT&TP: ILI9488, Decoder: PCM5102A (according to common.h CONFIG_IDF_TARGET_ESP32S3). 
3.)	I tested the same 7 SD-Cards in a single SD-Card adapter with direct short wires (no resistors, no pullups, direct to GPIO11, GPIO13, GPIO14 of ESP32-S3-WROOM-1-N16R8 (no other peripherals connected). 
4.)	Who can do a correlation test in that configuration with ESP32-S3-WROOM-1-N16R8 to help me? 

Did you get the same results? Have you solved the problem? 

Reboot (without SD-Card), result is normal, as I expected: 

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
[   983][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
E (237) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (238) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
[  1232][E][SD_MMC.cpp:138] begin(): Failed to initialize the card (0x107). Make sure SD card lines have pull-up resistors in place.

Reboot (with SD-Card), result is a "Boot-Loop": 

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
[   983][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403778ad
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
[   988][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
ESP-ROM:esp32s3-20210327

… and so on (Boot Loop problem)!

Batzi1967, 2023-07-15:
Bei mir genau das gleiche!
mit SD-Karte hängt er im boot loop:
"ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
[   986][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=21, ...)
[   987][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=50009, ...)
[   992][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
[   999][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
[  1008][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403778e9
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
[   991][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=21, ...)
[   991][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=50009, ...)
[   996][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
[  1004][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
[  1012][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403778e9
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
[   991][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=21, ...)
[   991][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=50009, ...)
[   996][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
[  1004][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
[  1012][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403778e9
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
[   991][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=21, ...)
[   991][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=50009, ...)
[   996][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
[  1004][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
[  1012][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled"
sobald die Karte herausen ist:
"ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403778e9
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
[   991][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=21, ...)
[   991][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=50009, ...)
[   996][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
[  1004][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
[  1012][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
E (156) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (157) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
[  1177][E][SD_MMC.cpp:138] begin(): Failed to initialize the card (0x107). Make sure SD card lines have pull-up resistors in place."

schreibfaul1, 2023-07-16:
Aber was ist die Ursache? Der ESP32-S3 verlangt keine festen GPIOs. Es kann jeder Port verwendet werden. Ermittlet doch bitte andere freie Ports auf dem Board und und beschaltet diese mit der SD_MMC (und in common.h ändern) 
Der ESP kommt aus dem Boot nicht heraus, obwohl die verwendete GPIOs keine "strapping pins" sind.

TubaFonse, 2023-07-16:
Hallo Wolle, 
genauso hab ich es beschaltet (an einen SD-Card-auf-Micro-SD-Adapter ca. 5cm lange Drähte angelötet und an den Sockel des ESP32-S3-DevKitC-1-N16R8 (Bord Version V1.3) gesteckt. 
Außer Deiner Standard-Konfiguration in common.h der ESP32S3 Variante 
(MMC_D0 = 11, MMC_CLK = 13, MMC_CMD = 14) 
habe ich heute noch mit 
(MMC_D0 = 4, MMC_CLK = 5, MMC_CMD = 6) sowie  
(MMC_D0 = 19, MMC_CLK = 20, MMC_CMD = 21) 
probiert mit dem gleichen Fehlerbild.

Batzi1967, 2023-07-16:
Hallo Ihr Zwei,
ich hab die gleiche Lösung mit dem Adapter,
Ich habe es gerade mit  SD_MMC_D0 = 40; SD_MMC_CLK = 39; SD_MMC_CMD = 45
probiert, genau das gleiche.

Arne65, 2023-07-16:
Hallo

Probleme hatte ich immer mit ESP Boards aus China .Espressif Originale funktionierten bei mir immer . 
Mit den China replicas hatte ich die von euch beschriebenen Probleme.

Arne65, 2023-07-16:
Das nochmalige löten der Pins hat dann aber die Probleme behoben .

TubaFonse, 2023-07-16:
hallo @Arne65 und an Alle: 
1.) mich würde interessieren, um den Fehler einzugrenzen, hast Du ein Board (egal woher ...) mit einem Original ESPRESSIF ESP32-S3-WROOM-1 Chip drauf (on-Chip PSRAM und Flash!) 
Laserbeschriftung mit Lupe lesbar: 
(oben) ESP32-S3-WROOM-1 
(unten) MON4R8 od. MON4R2 od. MON8R2 od. MON16R2 od. MON8R8 od. MON16R8), 
die Rückseite des Boards (Platine) ist bestempelt mit: 
YD-ESP32-S3
2022-V1.3
das in der aktuellen MiniWebRadio V2 problemlos läuft? 
2.) ich hab mir 2 Stück Espressif ESP32-S3-WROOM-1 MON16R8 (16MB Flash, 8MB PSRAM) Module gekauft, die Stiftleisten selbst gelötet (glaubt mir, ich kann löten!), und beide Module zeigen das gleiche o.g. Fehlverhalten. 
3.) wer hat KEIN Problem mit o.g. Board UND welche Konfiguration habt ihr für das MiniWebRadio mit ESP32-S3-WROOM-1 im Einsatz? 

Danke & LG, 
euer TubaFonse

valera39, 2023-07-16:
I have used different Chinese boards, almost always sd card connection errors are due to bad contacts 
this mapping to the default connection works without any problems
https://aliexpress.ru/item/1005004609461716.html?sku_id=12000029817335306
https://aliexpress.ru/item/32886466598.html?spm=a2g2w.orderdetail.0.0.6c774aa6KSn0dQ&sku_id=65655306566

Batzi1967, 2023-07-16:
Hallo zusammen!
Das sind Bilder von meinem Board, ich hab nur das eine. Die Stiftleisten waren bereits gelötet.
[image]
[image]
Funktionieren sollte er eigentlich, ich hatte darauf schon EspHome am laufen, damit hat es bestens funktioniert.
Ich hab mir heute noch mal 2 neue von einem anderen Chinamann bestellt. Mal schauen, wie die sich dann verhalten.
LG
Franz

steve6375, 2023-07-16:
I don't have an S3 setup.
I notice that clock div is 2 in my output ?? Not sure what it should be for S3.

The reset is apparently a watchdog timer because no interrupts are happening?
https://esp32.com/viewtopic.php?t=6469

I notice that you compile for the DAC and not the VS1053.
If you compile for the VS1053, does it boot any further? Just change common.h for VS1053 (even if one is not connected) and try it maybe?

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
[   446][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled

ESP32 Chip: ESP32-D0WD-V3
Arduino Version: 2.0.9
ESP-IDF Version: 4.4.4
MiniWebRadio     Version 2.8.3 Jul 15/2023
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
PSRAM total size: 8388608 bytes
```

dorinik, 2023-07-16:
Hello,
I have 2 cards purchased from here:
https://vi.aliexpress.com/item/1005004609461716.html?spm=a2g0o.order_list.order_list_main.16.21ef1802TDO1Qy&gatewayAdapt=glo2vnm

- ESP32-S3 N8R2 board works perfectly
- ESP32-S3 N16R8 board I have the same Boot Loop error

steve6375, 2023-07-16:
@dorinik - what other devices are you using? TFT? Audio?

dorinik, 2023-07-16:
[image]
[image]

steve6375, 2023-07-16:
OK, so only difference is maybe the partitions as you are using DAC with 8MB and it works - just not with 16MB.
miniwebradio16MB.csv will be used for a 16R8.
What if you just replace that with miniwebradio8MB.csv contents just as a test?
(I am not an expert - just trying to help!).

dorinik, 2023-07-16:
It's the same, it doesn't work!

TubaFonse, 2023-07-16:
@steve6375  
1.) I just tested in "ESP32S3 with VS1053"-Configuration - same Boot-Loop Problem! - when SD-Card inserted, see here: 

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377741
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
[   944][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377741
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
[   944][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
E (216) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (216) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
[  1170][E][SD_MMC.cpp:138] begin(): Failed to initialize the card (0x107). Make sure SD card lines have pull-up resistors in place.

so you'll see, only disconnect (pull-out) of SD-Card stops the "Boot-Loop" - undependant of Decoder-Configuration. 

2.) Who can write a little SDMMC Test Program (maybe Read / Write / Speed-Test under this ESP32-S3-WROOM-1 Conditions with PSRAM and Flash? I only found https://github.com/espressif/arduino-esp32/blob/master/libraries/SD_MMC/examples/SDMMC_Test/SDMMC_Test.ino and don't know to import this to PlatformIO with ESP32-S3-N16R8 environment, exactly as used for MiniWebRadio V2.

TubaFonse, 2023-07-16:
Test (only for @steve6375 )
You want to know, what happens, when creating an image for ESP32-S3-N8R2 (8M Flash, 2M quad PSRAM) just for test - and load it into an S3-N16R8 (16M Flash, 8M octo PSRAM)? Here it is (it don't will work - and Boot-Loop): 

Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420c3802
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
E (511) psram: PSRAM ID read error: 0x00ffffff
E (512) spiram: SPI RAM enabled but initialization failed. Bailing out.
[   528][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled

ESP32 Chip: ESP32-S3
Arduino Version: 2.0.9
ESP-IDF Version: 4.4.4
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
E (65) spiram: SPI RAM not initialized

abort() was called at PC 0x42044450 on core 1

Backtrace: 0x4037763a:0x3fcecf20 0x4037ddf1:0x3fcecf40 0x40387b6d:0x3fcecf60 0x42044450:0x3fcecfe0 0x4200c41d:0x3fced000 0x420325de:0x3fced0a0

  #0  0x4037763a:0x3fcecf20 in panic_abort at C:\Users\Hoeckmayr\.platformio\packages\framework-espidf@3.40404.0\components\esp_system/panic.c:408
  #1  0x4037ddf1:0x3fcecf40 in esp_system_abort at C:\Users\Hoeckmayr\.platformio\packages\framework-espidf@3.40404.0\components\esp_system/esp_system.c:137
  #2  0x40387b6d:0x3fcecf60 in abort at C:\Users\Hoeckmayr\.platformio\packages\framework-espidf@3.40404.0\components\newlib/abort.c:46
  #3  0x42044450:0x3fcecfe0 in esp_spiram_get_size at C:\Users\Hoeckmayr\.platformio\packages\framework-espidf@3.40404.0\components\esp_hw_support\port\esp32s3/spiram.c:299
      (inlined by) esp_spiram_get_size at C:\Users\Hoeckmayr\.platformio\packages\framework-espidf@3.40404.0\components\esp_hw_support\port\esp32s3/spiram.c:295
  #4  0x4200c41d:0x3fced000 in setup() at src/main.cpp:1424
  #5  0x420325de:0x3fced0a0 in loopTask(void*) at C:/Users/Hoeckmayr/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42

ELF file SHA256: 21e2c80d79f9914a

Rebooting...

schreibfaul1, 2023-07-16:
It could be that the new S3 boards from Espressif (with octal PSRAM) need the IDF 4.4.5.
https://github.com/espressif/esp-idf/issues/11438
The reboot is then only a consequential error that is triggered by the TFT.
If you remove this (in setup()): 
`
    if(!SD_MMC.begin("/sdcard", true, false, sdmmc_frequency)){
        clearAll();
        tft.setFont(_fonts[5]);
        tft.setTextColor(TFT_YELLOW);
        tft.setCursor(50,100);
        tft.print("SD Card Mount Failed");
        setTFTbrightness(80);
        SerialPrintfln(ANSI_ESC_RED "SD Card Mount Failed");
        while(1){};  // endless loop, MiniWebRadio does not work without SD
    }
`
the real error message will appear

TubaFonse, 2023-07-16:
Hi Wolle, 
do you have one of that new S3 board from Espressif (with octal PSRAM) -N8R8 / -N16R8 ?

schreibfaul1, 2023-07-17:
I use the AI ​​Thinker ESP32-S3-12K.
Looking at the N8R8, it may be that the IR PIN (GPIO33) is used internally. GPIO33 is not routed to the outside. What happens if you change the IR_PIN from 33 to 12 in common.h?

TubaFonse, 2023-07-18:
Maestro Wolle @schreibfaul1 
Changing the IR_PIN from 33 to GPIO4 solves the "Boot Loop" problem with ESP32-S3-WROOM-1-N16R8 and -N8R8 and -N4R8 (all Octal SPI PSRAM), see esp32-s3_datasheet_en-1.pdf, page 26 chapter 2.7 "Pin Mapping between Chip and Flash/PSRAM": 
Don't use GPIO33, GPIO34, GPIO35, GPIO36, GPIO37 (already used between Chip and the In-package Octal SPI Flash/PSRAM).

TubaFonse, 2023-07-19:
[image]

closed 2023-07-19 by TubaFonse

schreibfaul1, 2023-07-19:
thanks, we should not use the GPIO listed in the table for the ESP32-S3