schreibfaul1/ESP32-MiniWebRadio #250: VU meter VS1053

valera39, 2023-07-17:
Hello Wolle!
Updated the firmware on the old radio with the ESP32 Wrower, VS1053 configuration but the VU meter does not work. I updated completely with new project

steve6375, 2023-07-19:
VU meter also stopped working for me since last three versions, as I previously reported.

schreibfaul1, 2023-07-20:
the VS1053 plugin was disabled

schreibfaul1, 2023-07-20:
I have a board that doesn't work at all with the patch and keeps crashing. After a test I forgot to activate it. This will work again now.
A new AAC decoder that can play back aac+ in stereo is being tested. If this is successful, it is better to do without VS1053 boards in the future.

valera39, 2023-07-20:
Good!
It's just that I have some receivers built on VS1053 and there is no way to convert them to another DAC. So if possible, don't give up on the VS1053 support.

schreibfaul1, 2023-07-20:
Not that, but I don't want to invest any more time in the VS1053. If you are planning a new project, please use an external DAC or audio board from AT Thinker, Olimex or the Lyra series from Espressif

valera39, 2023-07-20:
PCM5102 will do? This one is more accessible to me

schreibfaul1, 2023-07-20:
yes, the PCM5102 is well suited
[image]

closed 2023-07-23 by valera39