schreibfaul1/ESP32-MiniWebRadio #254: After time announce mp3 file plays, MWR connects to wrong radio station

steve6375, 2023-07-19:
Playing station 2 = BBC Radio 4. 
When the time announce MP3 file plays - afterwards it plays the wrong station - Classic FM  which is station #7.
The display still says BBC Radio 4  but the info field below says Classic FM or the title of the classical music that is playing.
I added debug code to print out current station in SetStation in main.cp
12:59:21 sta 2, _cur_station 4

```
12:59:04 Audiocodec:  MP3
12:59:08 ir_code: ..  IR address 0x00, IR command 0x0d
12:59:13 ir_code: ..  IR address 0x00, IR command 0x0d
12:59:19 ir_code: ..  IR address 0x00, IR command 0x46
12:59:21 ir_result:   Stationnumber 2
12:59:21 sta 2, _cur_station 4
12:59:21 action: ...  switch to station 2
12:59:21 AUDIO_info:  Connect to new host: "https://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/uk/sbr_high/ak/bbc_radio_fourfm.m3u8"
12:59:21 AUDIO_info:  Song stopped correctly after 20 msec
12:59:22 AUDIO_info:  SSL has been established in 564 ms, free Heap: 102176 bytes
12:59:22 AUDIO_info:  new request: "http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d320000.norewind.m3u8"
12:59:22 AUDIO_info:  The host has disconnected, reconnecting
12:59:22 AUDIO_info:  new request: "http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=320000-264026245.ts"
12:59:22 Audiocodec:  AAC
12:59:24 AUDIO_info:  new request: "http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=320000-264026246.ts"
12:59:27 AUDIO_info:  new request: "http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=320000-264026247.ts"
12:59:34 AUDIO_info:  new request: "http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=320000-264026248.ts"
[658211][E][IR.cpp:152] loop(): something went wrong, intval_l 175, intval_h 0, pulsecounter 33
12:59:40 AUDIO_info:  new request: "http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=320000-264026249.ts"
12:59:47 AUDIO_info:  new request: "http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d320000.norewind.m3u8"
12:59:47 AUDIO_info:  new request: "http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=320000-264026250.ts"
12:59:53 AUDIO_info:  new request: "http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio=320000-264026251.ts"
12:59:53 Time: ...... play Audiofile /voice_time/13_00.mp3
12:59:53 AUDIO_info:  Song stopped correctly after 0 msec
12:59:53 AUDIO_info:  Reading file: "/voice_time/13_00.mp3"
12:59:53 Audiocodec:  MP3
12:59:53 AUDIO_info:  Content-Length: 144621
12:59:53 AUDIO_info:  ID3 framesSize: 55    
12:59:53 AUDIO_info:  ID3 version: 2.4   
12:59:53 AUDIO_info:  ID3 extended header
12:59:53 id3data: ..  UserDefinedText: SoftwareLavf57.20.100
12:59:53 AUDIO_info:  Audio-Length: 144566
12:59:53 AUDIO_info:  stream ready        
13:00:01 AUDIO_info:  Song stopped correctly after 20 msec
13:00:01 AUDIO_info:  End of file "13_00.mp3"
13:00:01 end of file: 13_00.mp3
13:00:01 AUDIO_info:  Connect to new host: "http://media-ice.musicradio.com/ClassicFMMP3"
13:00:01 AUDIO_info:  Song stopped correctly after 0 msec
13:00:01 AUDIO_info:  Connection has been established in 23 ms, free Heap: 141600 bytes
13:00:01 bitRate:     128Kbit/s
13:00:01 icy-descr:   Classic FM
13:00:01 AUDIO_info:  icy-name: Classic FM
13:00:01 StationName: Classic FM
13:00:01 lastURL: ..  http://media-ice.musicradio.com/ClassicFMMP3
13:00:01 AUDIO_info:  stream ready
13:00:01 StreamTitle:
13:00:01 AUDIO_info:  StreamUrl=''
13:00:02 Audiocodec:  MP3

```

steve6375, 2023-07-19:
Another example

Note: station selected using IR remote...

1. IR select station 9 Asian network
2. Time announce
3. Pop Lounge Cafe plays

```
17:55:22 ir_code: ..  IR address 0x00, IR command 0x09
17:55:24 ir_result:   Stationnumber 9
17:55:24 sta 9, _cur_station 4
17:55:24 action: ...  switch to station 9
17:55:24 AUDIO_info:  Connect to new host: "https://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/uk/sbr_high/ak/bbc_asian_network.m3u8"
17:55:25 AUDIO_info:  Song stopped correctly after 30 msec
17:55:25 AUDIO_info:  SSL has been established in 508 ms, free Heap: 96980 bytes
17:55:25 AUDIO_info:  file "/logo/s/BBC Asian Network [UK ONLY].jpg" not found
17:55:25 AUDIO_info:  new request: "http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_asian_network/bbc_asian_network.isml/bbc_asian_network-audio%3d320000.norewind.m3u8"
17:55:25 AUDIO_info:  The host has disconnected, reconnecting
17:55:25 sta 9, _cur_station 9
17:55:25 AUDIO_info:  new request: "http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_asian_network/bbc_asian_network.isml/bbc_asian_network-audio=320000-264029020.ts"
17:55:26 Audiocodec:  AAC
17:55:27 AUDIO_info:  new request: "http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_asian_network/bbc_asian_network.isml/bbc_asian_network-audio=320000-264029021.ts"

17:59:53 Time: ...... play Audiofile /voice_time/18_00.mp3
17:59:53 AUDIO_info:  Song stopped correctly after 0 msec
17:59:53 AUDIO_info:  Reading file: "/voice_time/18_00.mp3"
17:59:53 Audiocodec:  MP3
17:59:53 AUDIO_info:  Content-Length: 144470
17:59:53 AUDIO_info:  ID3 framesSize: 55
17:59:53 AUDIO_info:  ID3 version: 2.4
17:59:53 AUDIO_info:  ID3 extended header
17:59:53 id3data: ..  UserDefinedText: SoftwareLavf57.20.100
17:59:53 AUDIO_info:  Audio-Length: 144195
17:59:53 AUDIO_info:  stream ready
18:00:01 AUDIO_info:  Song stopped correctly after 20 msec
18:00:01 AUDIO_info:  End of file "18_00.mp3"
18:00:01 end of file: 18_00.mp3

18:00:01 AUDIO_info:  Connect to new host: "https://poploungecafe.stream.laut.fm/poploungecafe?t302=2023-07-19_18-06-08&uuid=547c12cb-5951-43b4-b0d6-4929d66e8076"
18:00:01 AUDIO_info:  Song stopped correctly after 0 msec
18:00:03 AUDIO_info:  SSL has been established in 1690 ms, free Heap: 98416 bytes
18:00:04 bitRate:     128Kbit/s
18:00:04 icy-descr:   Luxury Panorama Lounge.
18:00:04 icy-url: ..  http://laut.fm/poploungecafe

```

schreibfaul1, 2023-07-20:
The last URL that was successfully connected is always saved. With the m3u8 format, this changes constantly. Therefore this function was blocked. I implemented a workaround, to do this, the two audio libraries must be reloaded with 'Full Clean'.

steve6375, 2023-07-20:
Thanks. That has fixed it. I couldnt work out what was wrong.

closed 2023-07-20 by steve6375