schreibfaul1/ESP32-MiniWebRadio #260: Problem changing IR Remote codes using Web UI

steve6375, 2023-07-21:
Not sure of the best way to solve this....

1. Say a user has a different controller (i.e. my mp3 controller which generates same codes but has different button legends).
2. User goes to IR code setup screen in web UI - all boxes are empty
3. User then presses MUTE key on his new mp3 IR controller which generate the code 0x47 for key 10 
[image]
4. Now user tests it by pressing MUTE key - but now the large number '3' appears on the TFT screen (MUTE has not worked because 0x47 occurred in the table at key 3 before the key10 key) but the user does not know that key 3 is also 0x47 because the field is blank for key 3.

Even if the user pressed LOAD first, it will still find key 3 first in table. So it appears to the user that the reassignment has not worked.

steve6375, 2023-07-21:
Maybe say 'Press LOAD to see existing button assignments and make sure all codes are unique'

steve6375, 2023-07-21:
P.S. My mp3 controller has 21 buttons. It would be nice to be able to assign some extra buttons
Maybe you could buy one (very cheap from Ali Express ;-) and try to support it.
They have buttons marked
Standby
Mode - for Radio <> Time
Mute
EQ
RPT
U/SD - could be used for Radio or Player
Play/Pause

Perhaps a long pressed key could generate a different unique key code?
e.g.   short press of # key generates 0x0d code, but long press of # generates  0x10d and button code table expanded to 512?

You can add table entries for extra functions:

Function
======
Standby -  fall sleep instantly
Set Sleep Timer
Alarm Set
Play/Pause - stops/starts radio or  pause/resume mp3 file
EQ - displays freq gain/bass/treble controls (while Radio or Mp3 file playing)
REPEAT  - repeats mp3 file play
Radio/MP3 Player select
STOP  - Stop mp3 file or Radio
MUTE - on/off toggle
UP
DOWN
VOL+
VOL-
|<< - prev station or mp3 file
>>| - next station or mp3 file

So MUTE would be 0x0d and SLEEP would be 0x10d (for long mute press).
So Standby button could be short press assigned to Standby function but long press assigned to Set Sleep timer.
Short press EQ - EQ screen, Long press - Alarm set screen
Long press RPT - Brightness
Long press MODE - Player screen

User could assign long press functions to the number buttons if not enough buttons on their controller.

Extra functions could then be added to web UI and user told to use short press or long press to set each function.
Do you think this will work?

schreibfaul1, 2023-07-22:
if values ​​are assigned more than once, they are marked in color
[image]

steve6375, 2023-07-22:
Nice!
But what about when page first loads and all fields are blank ??
Does it work then...

steve6375, 2023-07-24:
Still a 'problem' when page is first loaded.
First time the new user loads the web UI, the values are not populated.
If he then adds an existing code, it is not yellow and the user test does not work.
[image]

Robert131, 2023-10-17:
I am facing similar problems, in some fieds from the Browsers table the ir-codes were faulty transmissed to the EEPROM. This was clear when the uart debug-info showed the button-list with the functions saveIRbuttonsToNVS()
Reason is the Browser-Type, i used first the Firefox-Browser. With Chrome there were no errors transmitting the codes to NVS.

steve6375, 2023-10-17:
I found that you must have only one browser page open at any time , otherwise it gets  confused, so make sure all other web radio pages are closed and only one is running .

schreibfaul1, 2023-10-18:
The problem is actually multiple instances that can be opened in the browser. Let's see how I can prevent this in the future.

closed 2024-01-22 by github-actions[bot]