schreibfaul1/ESP32-MiniWebRadio #261: Display Pinout Thinker A1S

newbee90, 2023-07-24:
Hallo, nachdem ich erfolgreich dein erstes Radio nachgebaut habe möchte ich auch das andere bauen.
Ich habe ein 3,5" Display für einen Raspi
https://i.ebayimg.com/images/g/YjsAAOSwQoxhCPvz/s-l1600.jpg
so eines, aber ich werde aus deinem Plan nicht schlau, obwohl ich Elektriker bin und auch Fernseher reparieren kann.
Hilf mir mal bitte auf die Sprünge, Danke für deine Mühe

schreibfaul1, 2023-07-24:
Die RPI Displays haben keinen BL Pin, sonst unterscheiden sie sich nicht von den anderen.
[image]
Ich habe selbst das A1S Board in Betrieb. Viele herausgeführten Pins besitzen eine RC Kombination zum Entprellen der Tasten. Bei hohen Frequenzen schließen die Kondensatoren das Signal kurz. Daher muss die RC Kombination unwirksam gemacht werden. Ich habe das so gelöst:https://github.com/schreibfaul1/ESP32-MiniWebRadio/blob/master/additional_info/AI%20Thinker%20A1S%20Board.pdf
SMD Bauelemente auslöten ist nicht ganz einfach, aber es geht. 
Der verbaute DAC wird über I2C gesteuert. Die I2C Leitungen sind von außen nicht zugänglich, müssen aber trotzdem definiert werden (Pins 32 und 33).

newbee90, 2023-07-29:
auslöten ist easy, einfach eine Tropfen Lötzinn drauf und mit der Entlötpumpe den Widerstand abgesaugt, spitze Pinzette geht auch

closed 2024-01-21 by github-actions[bot]