schreibfaul1/ESP32-MiniWebRadio #262: Need some help.

Jefke1, 2023-07-26:
Best,
Everything works perfectly, except the tp of the display.
I have an ILI9341 with chip XPT 2046 and have to press on the left side of the small buttons to make it work. when I press in the middle I get another function. I then ordered 2 other screens ILI9486 and ILI 9488 both with chip XPT 2046. I have tried all configurations with these displays, the TP remains in mirror writing. How can I fix this problem?? Thank you in advance.

steve6375, 2023-07-26:
I use these settings in common.h - what do you use?
```
#define _SSID               "mySSID"                        // Your WiFi credentials here
#define _PW                 "myWiFiPassword"
#define DECODER             1                               // (0)VS1053 , (1)MAX98357A PCM5102A... (2)AC101 (3)ES8388 (4)WM8978
#define TFT_CONTROLLER      0                               // (0)ILI9341, (1)HX8347D, (2)ILI9486a, (3)ILI9486b, (4)ILI9488, (5)ST7796, (6)ST7796RPI
#define DISPLAY_INVERSION   0                               // (0) off (1) on
#define TFT_ROTATION        1                               // 1 or 3 (landscape)
#define TFT_FREQUENCY       40000000                        // 27000000, 40000000, 80000000
#define TP_VERSION          0                               // (0)ILI9341, (1)ILI9341RPI, (2)HX8347D, (3)ILI9486, (4)ILI9488, (5)ST7796, (3)ST7796RPI
#define TP_ROTATION         3                               // 1 or 3 (landscape)
#define AUDIOTASK_CORE      1                               // 0 or 1
#define AUDIOTASK_PRIO      2                               // 0 ... 24  Priority of the Task (0...configMAX_PRIORITIES -1)
#define SDMMC_FREQUENCY     40000000                        // 40000000, 2000000, 10000000, not every SD Card will run at 40MHz
#define FTP_USERNAME        "esp32"                         // user and pw in FTP Client
#define FTP_PASSWORD        "esp32"
#define CONN_TIMEOUT        500                             // unencrypted connection timeout in ms (http://...)
#define CONN_TIMEOUT_SSL    2000                            // encrypted connection timeout in ms (https://...)
```

Jefke1, 2023-07-26:
I used your settings, the result is that mute is now in the top right corner instead of the bottom left.
#define DECODER             1                               // (0)VS1053 , (1)MAX98357A PCM5102A... (2)AC101 (3)ES8388 (4)WM8978
#define TFT_CONTROLLER      0                               // (0)ILI9341, (1)HX8347D, (2)ILI9486a, (3)ILI9486b, (4)ILI9488, (5)ST7796, (6)ST7796RPI
#define DISPLAY_INVERSION   0                               // (0) off (1) on
#define TFT_ROTATION        1                               // 1 or 3 (landscape)
#define TFT_FREQUENCY       40000000                        // 27000000, 40000000, 80000000
#define TP_VERSION          0                               // (0)ILI9341, (1)ILI9341RPI, (2)HX8347D, (3)ILI9486, (4)ILI9488, (5)ST7796, (3)ST7796RPI
#define TP_ROTATION         3                               // 1 or 3 (landscape)
#define AUDIOTASK_CORE      1                               // 0 or 1
#define AUDIOTASK_PRIO      2                               // 0 ... 24  Priority of the Task (0...configMAX_PRIORITIES -1)
#define SDMMC_FREQUENCY     20000000                        // 40000000, 2000000, 10000000, not every SD Card will run at 40MHz
#define FTP_USERNAME        "esp32"                         // user and pw in FTP Client
#define FTP_PASSWORD        "esp32"
#define CONN_TIMEOUT        500                             // unencrypted connection timeout in ms (http://...)
#define CONN_TIMEOUT_SSL    2000                            // encrypted connection timeout in ms (https://...)

steve6375, 2023-07-26:
So the TP calibration is off slightly by about 1cm? Is that what you are saying?
Where did you buy the displays from e.g. ILI9341. 
Is 5V supply OK? Can you paste a picture of your setup?

Jefke1, 2023-07-26:
yes, slightly shifted. I bought all the displays on AliExpress. My setup is correct also the 5V is no problem.
I made my setup on a test board with all wires soldered, not on a breadboard with jumper wires.
To avoid bad connections.
If i change TFT_Rotation and TP_Rotation than i have to press the left side of the small buttons.

steve6375, 2023-07-26:
In main.cpp you can print out the x and y coordinates
[image]

Then you can adjust them by addition or subtraction
e.g.
x = x - 20;
y = y + 35;

steve6375, 2023-07-26:
Here is my terminal output for  first press = top of screen area
other presses all MUTE button presses

Top of screen press...
```
11:29:45 tp_pressed, state is: 1
11:29:45 Touchpoint  x=27, y=213

Mute button presses

11:29:46 tp_pressed, state is: 1
11:29:46 Touchpoint  x=21, y=212
11:29:48 tp_pressed, state is: 1
11:29:48 Touchpoint  x=23, y=208
11:29:50 tp_pressed, state is: 1
11:29:50 Touchpoint  x=36, y=208
11:29:52 tp_pressed, state is: 1
11:29:52 Touchpoint  x=15, y=209
11:29:54 tp_pressed, state is: 1
11:29:54 Touchpoint  x=9, y=210

```
winbutton is defined as
`    struct w_o {uint16_t x = 0;   uint16_t y = 180; uint16_t w =  40; uint16_t h = 40; } const _winButton;`

The button number is in btnNR
so the first button (MUTE) must be between 180 and 220 for y
and 0 and 40 for x

The 2nd button (VOL-) would be  x = 41-80
e.g.
11:40:15 tp_pressed, state is: 1
11:40:15 Touchpoint  x=62, y=211

```
        case RADIOico:
                        if(y <= _winTitle.y) { yPos = RADIOico_1; }
                         if((y >_winButton.y) && (y < _winButton.y + _winButton.h)) {
                                            yPos = RADIOico_2;
                                            btnNr = x / _winButton.w;

```

steve6375, 2023-07-26:
P.S. To check right side screen calibration...
If you go to the mp3 menu screen
The last button on extreme right is the RADIO button
This should give a value like
11:50:16 tp_pressed, state is: 6
11:50:16 Touchpoint  x=309, y=206

So Y should be 180-219  and x should be 280-319

Jefke1, 2023-07-26:
Steve,
Thanks for your time and help, I've modified it as you said. Works perfectly now. I'm not much of a programmer and I overlooked this piece of code.
Thanks again for all the help, I appreciate it.

steve6375, 2023-07-26:
Wolle (author) says TP calibration is not required for most TFT screens, so not sure why yours require tweaking...

steve6375, 2023-07-26:
P.S. It doesn't hurt to add two capacitors, each one between 5V and GND at TFT panel pins (i.e. in parrallel).
e.g.  20nF and 100uF/16V.

Also same on SD card if using a separate holder/adaptor.

Jefke1, 2023-07-26:
> 

I don't know how it is possible, in V1 everything worked normally.
I have already placed capacitors to avoid a voltage drop. Thanks again.

steve6375, 2023-07-26:
The buttons were much bigger in V1  so accuracy was not so critical.

Rich1975, 2023-07-29:
Had a similar problem with the touch and thought replacing the 2.8 TFT to a 3.2 TFT would help and then saw these comments, I changed my TP_ROTATION from 1 to 3 and that took care of the touch problem
Thank you

Jefke1, 2023-08-23:
Dear Wolle,

I ordered 2 x 3.5 inch tft displays from Aliexpress
1 = TN ILI9486 Touch xpt 2046
2 = IPS 9488 Touch xpt 2046
They were not marked ??
One of them works perfectly, the other has problems with the startup screen and logo's (these are not clear), the text of the artist and the song appear in red, everything else seems ok. I tested the screen with Bodmer TFT/espi lib everything works normally. How can I find out which display is working correctly and which is not.
[image]
[image]
Or how can I solve this problem ??
Thank you in advance.

schreibfaul1, 2023-08-23:
I suspect ILI9488 is the problem. There are too many displays from different manufacturers that are not 1:1 compatible (maybe their gamma values ​​are wrong?)
Two questions: if you lower the TFT_TREQUENCY to 20MHZ, is it still there? Which settings do you use with the Bodmer Lib so that the display works correctly?

Jefke1, 2023-08-24:
I have changed the TFT-Freq to 20 MHZ, result remains the same. 
For both displays I have to use TFT_Controller ILI9488 if using ILI9486(a) or (b) nothing appears on the display. 
The attached pictures are not from the SD_card but from Spiffs.
Setup settings from Bodmer Lib. attached.
[User_Setup.txt](https://github.com/schreibfaul1/ESP32-MiniWebRadio/files/12427364/User_Setup.txt)
[image]
[image]

Jefke1, 2023-08-24:
Setup made with jpg read from Sd_card. Lib. setup file incl.
[User_Setup.txt](https://github.com/schreibfaul1/ESP32-MiniWebRadio/files/12429852/User_Setup.txt)

https://github.com/schreibfaul1/ESP32-MiniWebRadio/assets/47332797/c2bdac53-97ad-4c22-8b58-f73d5f59e5a1

schreibfaul1, 2023-08-24:
I have no idea if the display will work here. It will be an ILI9486 controller that works with 18 bit colour depth. My library uses 16 bits for each colour. 18 bits are selected in the eSPI library. [image] for me it gives:
[image]
 You can try in the tft.cpp (path /lib/tftlib/src/tft.cpp) to change the value to `spi_TFT->write16(0x66);` If the 18 bits are only used internally it might work. Good luck!

Jefke1, 2023-08-24:
code modified, but nothing changed.
Thank you.

newbee90, 2023-08-27:
I have TN-9486 With Touch from Aliexpress and the same problem

closed 2024-01-21 by github-actions[bot]