schreibfaul1/ESP32-MiniWebRadio #264: IL9344 problem

BruceKenley, 2023-07-29:
I have a problem with my touch screen. Running in landscape mode with resistance type selected in common.h Everything works but buttons are working reverse from centre. Centre button works correctly but button on right activates button on left and button on right activates button on left. Can someone tell me how to change this?
Hope you can help.
Using ESP~WRover
iL9344 TFT
PCM5102
PAM8406

steve6375, 2023-07-29:
Presumably you tried both TP_ROTATION at 3 and 1?
The ILI9344 is not supported (see README)
Did you see #262 issue?
Maybe you could reverse the x value

valera39, 2023-07-29:
try different combinations of TP_VERSION, TFT_ROTATION, TP_ROTATION. I had a similar problem, the ILI9488 display worked correctly with the ILI9486 touchpad. Not all displays from China work correctly

AVRsteffen, 2023-07-29:
Yes, I agree. I have same problem.

closed 2024-01-21 by github-actions[bot]