schreibfaul1/ESP32-MiniWebRadio #266: Last button in first menu isn't draw correct

AVRsteffen, 2023-07-29:
This is play in slow motion!

https://github.com/schreibfaul1/ESP32-MiniWebRadio/assets/2552410/aa7d8dd3-9de8-42bc-8c1e-16a3e42f0e68

steve6375, 2023-07-29:
What speeds are reported by the terminal output on boot?
Can you paste in here?
Is cpu at 240mhz?

AVRsteffen, 2023-07-29:
[image]

ESP32 Chip: ESP32-D0WD-V3
Arduino Version: 2.0.11
ESP-IDF Version: 4.4.4
MiniWebRadio     Version 2.8.3b Jul 22/2023
CPU speed 240 MHz
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 8388608 bytes, speed 40 MHz
PSRAM total size: 8388608 bytes

00:00:00
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V2     *    
00:00:00        ***************************    
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9488
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 3762.6 MB by 3781.5 MB free
00:00:01 setup: ....  seek for stations.csv
00:00:01 setup: ....  stations.csv found
00:00:01 setup: ....  seek for WiFi networks
[  1638][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: mySSID

I am just pinned core in common.h to core --> 0
And I guess TFT runs faster..

AVRsteffen, 2023-07-29:
@steve6375 No, I am just compile all new fixes of  master respository.
Come on, I'm not the only one with this bug?

steve6375, 2023-07-29:
I have never seen this issue before.
So this bug only occurs in latest build?
AFAIK  no one else has reported it.

AVRsteffen, 2023-07-29:
But you can agree?

Arne65, 2023-07-29:
The last build works perfectly for me
Is communication with the SD card ok?
Set AUDIOTASK_CORE 1 // 0 or 1 to 0 that helped me when the buttons are a bit slow

AVRsteffen, 2023-07-29:
Yes, communication with  SD card is okay. Get not wrong data from card. Connections are very short and made on PCB.
Audio task core just set to "0". Since that change TFT runs faster. But this bug in drawing 5th button still in both core tasks and slow frequency of TFT too.

steve6375, 2023-07-29:
Did you try a Full Clean build?

Arne65, 2023-07-29:
Which display do you use?

Arne65, 2023-07-29:
I had similar problems with the ILI 9486

Arne65, 2023-07-29:
my common.h 
[image]

AVRsteffen, 2023-07-29:
I found the buggy file of "RADIOico1.jpg" on SD card and some other ones too. I will download new "btn" folder at SD. Than I guess it will work again.

Thanks all you guys for helping me to find bug

AVRsteffen, 2023-07-29:
@Arne65 I use a Chinese 3.5" TFT with ILI9488 controller. Now it works @70Mhz. Thanks @steve6375 for this hint.

closed 2023-07-29 by AVRsteffen