schreibfaul1/ESP32-MiniWebRadio #269: Bad sound when decode AAC SBR streams

kpbicka, 2023-08-01:
Good afternoon colleagues :)

I have assembled and tested 4 different Wi-Fi radios and this project has proved to be the most stable in my opinion.
However, there are several questions. Please answer me who can

1. Poor decoding of "hard" AAC SBR (Spectral Band Replication) streams. As an example, you can play the link https://stream01.pcradio.ru/Sandra-hi on your computer and through your radio (this project) and hear the difference - the upper frequencies are cut off. I using VS1053B decoder, but the same board decoded this stream well if connected to the original Ka-Radio32 firmware. 
In general, other AAC streams also sound bad - not enough high frequencies, although as I said - on the original karadio32 project through the VS1053B or external I2C DAC they sound high quality

2. For some reason, my console output ends here:

[ 446][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
ESP32 Chip: ESP32-D0WD-V3
               Arduino Version: 2.0.11
                                ESP-IDF Version: 4.4.4
                                            MiniWebRadio Version 2.8.3b Jul 22/2023
                                                         CPU speed 240 MHz
                                                                           ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)

FLASH size 4194304 bytes, speed 40 MHz
PSRAM total size: 8388608 bytes
                                                                           00:00:00
00:00:00 ***************************
00:00:00 * MiniWebRadio V2 *
00:00:00 ***************************
00:00:00
00:00:00 setup: .... Arduino is p

although when compiling i tried changing DCORE_DEBUG_LEVEL different values but the result is the same the last output is "Arduino is p"
I even tried AUDIOTASK_CORE 0 or 1 - but no luck

3. Request for a decorative character - the VU-meter substrate is too bright, almost the same brightness as the level indicators - I did not succeed in making it darker by editing the file /SD card/common/s/level_bar.bmp
4. The same problem when displaying the clock - empty segments are too bright on my ILI9341 display

schreibfaul1, 2023-08-01:
1) SBR is turned off by default for the ESP32. On the one hand, this is due to the memory consumption and, on the other hand, to the access speed of the memory. Due to the own Arduino compilation, the situation is a bit more relaxed here. Here you can switch on SBR in the aac_decoder.h.
[image]
I don't know karadio32.
2) I suspect a setting problem in the terminal. Try an external terminal, is the behavior different then?

3+4) These are prepared graphics on the SD that are used for many displays. You can use a graphics program to adjust the color, saturation and contrast to your liking.

kpbicka, 2023-08-01:
Thanks for fast and useful answer.

kpbicka, 2023-08-01:
Unfortually I can't find aac_decoder.h in this project  :(

schreibfaul1, 2023-08-01:
that's a bit hidden
.pio / libdeps / esp32 / ESP32-audioI2S-master / src / aac_decoder / aac_decoder.h 
[image]

kpbicka, 2023-08-01:
Моя благодарность Вам!

closed 2023-08-01 by kpbicka

steve6375, 2023-08-01:
The clock empty segments seem too bright on my 9341 too, but if viewed at 90 degrees the contrast is much better.
I would prefer it if the shadow unlit segments were a lot darker too.

schreibfaul1, 2023-08-01:
Yes, that's why SBR is disabled for ESP32. This is too big for SRAM in this project and PSRAM is too slow with 40MHz SPI

kpbicka, 2023-08-01:
> Yes, that's why SBR is disabled for ESP32. This is too big for SRAM in this project and PSRAM is too slow with 40MHz SPI

Do you think only ESP32-S3 with PSRAM is the best fit for your project?

schreibfaul1, 2023-08-01:
The ESP32-S3 has advantages and disadvantages. The biggest advantage is a faster PSRAM and thus more can be outsourced (SSL, mbedtls)
A serious disadvantage is currently the speed of the SD card. This should improve with Arduino V3 and IDF 5